perexg/satip-axe #1: Unicable only works with first tuner

timehlers, 2015-04-01:
Dear perex,

thanks again for your minisatip patch! Yesterday I tried the patched minisatip version on the IDL4k box. Today I realized that only the first tuner works on my unicable setup. The box needs to have only first tuner to be connected to the LNB and normal s2i.bin usually can use that signal on all tuners. My guess is, that one somehow has to tell the driver to use the signal from first input on all tuners, but I am not sure (and furthermore would not know how that would need to be set).

Best

Tim

perexg, 2015-04-01:
Unfortunately, I don't have any ioctl dump for the unicable setup (s2i.bin), so I cannot add anything related to the code.

timehlers, 2015-04-01:
Once I made an strace with a grep for open(/dev/ and ioctl on s2i.bin while tuning to "Das Erste HD":

http://www.vdr-portal.de/board18-vdr-hardware/board84-allgemein/p1201892-vdr-inkompatible-dvb-api-inkomp-treiber-auf-sat-ip-telestar-digibit-r1-idl4k-sh4-basierende-boxen/#post1201892

Maybe you understand more by looking at the output?

How did you find out the required changes for using the plain devices for minisatip? I tried to get a VDR running on the Box, using the devices directly, but I never got it working.

Best

perexg, 2015-04-01:
Unfortunately, without the full dump of ioctls (including the passed data). Maybe 'strace -s1024 -xx' might uncover something, but the Portisch's dumper seems as a best tool.

I used only information from the last Portisch commented ioctl dump.

timehlers, 2015-04-01:
Ok, I found the Thread with Portischs ioctrl dumper, wait, does your sentence "I can get stream from the first input with it." from http://www.vdr-portal.de/board18-vdr-hardware/board84-allgemein/p1240121-vdr-inkompatible-dvb-api-inkomp-treiber-auf-sat-ip-telestar-digibit-r1-idl4k-sh4-basierende-boxen/#post1240121

mean that the current implementation generally only work with the first tuner? Then my search would have gone in the wrong direction...

perexg, 2015-04-01:
I admit that I tested all four inputs right now (not before). The result is that first three tuners are working and the fourth does not work. It's standard DiseqC setup AA/AB with two satellite positions. So a little piece is missing for the standard "quad" settings.

perexg, 2015-04-01:
Looking to logs again and there are two things which are different:

- set VOLTAGE for all four tuners at start (which MIGHT make sense for unicable)
- 0x0A value for the FRONTEND_RESET ioctl (but this value is send when all four tuners are used)

timehlers, 2015-04-02:
I compiled a litte ioctl hook, and got this from:

printf("ioctl(%d, 0x%x, %p)\n", fd, request, next_ioctl);

Tuning to "Das Erste HD" on tuner 1:

# LD_PRELOAD=/mnt/mnt/lib/libwrap_ioctl.so.1.0 /root/s2i.bin
[00000.264465] SAT>IP Server 1.0 started
ioctl(27, 0x8004667f, 0x29b79440)
ioctl(27, 0x8004667f, 0x29b79440)
ioctl(27, 0x8915, 0x29b79440)
ioctl(27, 0x8936, 0x29b79440)
ioctl(27, 0x8916, 0x29b79440)
ioctl(27, 0x891c, 0x29b79440)
ioctl(27, 0x890c, 0x29b79440)
ioctl(27, 0x890b, 0x29b79440)
ioctl(27, 0x8915, 0x29b79440)
ioctl(27, 0x891b, 0x29b79440)
killall: minidlna: no process killed
ioctl(35, 0x80a86f3d, 0x29b79440)
ioctl(35, 0x6f5d, 0x29b79440)
ioctl(35, 0x6f43, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(38, 0x8915, 0x29b79440)
ioctl(40, 0x80a86f3d, 0x29b79440)
ioctl(40, 0x6f5d, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(35, 0x40016f61, 0x29b79440)
ioctl(35, 0x40076f3f, 0x29b79440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8927, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8927, 0x29801440)
ioctl(3, 0x8912, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8913, 0x29801440)
ioctl(3, 0x8915, 0x29801440)
ioctl(3, 0x891b, 0x29801440)
ioctl(35, 0x6f43, 0x29b79440)
ioctl(35, 0x40086f52, 0x29b79440)
ioctl(35, 0x6f43, 0x29b79440)
ioctl(40, 0x40016f61, 0x29b79440)
ioctl(35, 0x40076f3f, 0x29b79440)
ioctl(36, 0x40026f02, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(35, 0x6f43, 0x29b79440)
ioctl(40, 0x40086f52, 0x29b79440)
ioctl(35, 0x80286f4e, 0x29b79440)
ioctl(35, 0x80286f4e, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(40, 0x80286f4e, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80286f4e, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(36, 0x40026f02, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(36, 0x40026f02, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(36, 0x40026f02, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(36, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(41, 0x40026f01, 0x29b79440)
ioctl(41, 0x40026f02, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(35, 0x80026f47, 0x29b79440)
ioctl(35, 0x80026f48, 0x29b79440)
ioctl(35, 0x80046f46, 0x29b79440)
ioctl(35, 0x80386f60, 0x29b79440)
ioctl(40, 0x80026f47, 0x29b79440)
ioctl(40, 0x80026f48, 0x29b79440)
ioctl(40, 0x80046f46, 0x29b79440)
ioctl(40, 0x80386f60, 0x29b79440)
ioctl(36, 0x40026f02, 0x29b79440)

Do you know how to interprete these values?

perexg, 2015-04-02:
It's not sufficient (the indirect ioctl contents is not printed). Could you run this wrapper ?

https://github.com/perexg/satip-axe/blob/master/tools/syscall-dump.c

(put only a link to pastebin.com with log or so here...)

timehlers, 2015-04-02:
Ok, pastebin has a limit for free users. I put it as gitrepo on github:

https://github.com/timehlers/syscall-dump.das_erste.log

just clone the repo.

Best

perexg, 2015-04-03:
Thanks. But it seems that some calls are missing. I tried to improve the wrapper. Could you retry ?

9000h, 2015-04-03:
would it be worth a try to get the dddvb driver running, the Maxlinear chip is similar to the one from the DD MaxS8 card.

perexg, 2015-04-03:
Just a note, I added socket()/dup()/dup2 wrappers, too.

timehlers, 2015-04-03:
Hi perexg,

syscall-dump.das_erste2.log is in the git. I used nfs for the log and for the second log I also used the SYSCALL_DUMP_LOG variable. Before I just said:

LD_PRELOAD=/mnt/mnt/lib/syscall-dump.so /root/s2i.bin > /mnt/mnt/syscall-dump.das_erste.log 2>&1

Best

P.S. If NFS was too slow and you think I/O got droped (which should not be the case), I could redo it with TMPFS.

perexg, 2015-04-03:
@timehlers : I updated the minisatip.patch, could you test it with unicable ? It seems working for the quad config.

timehlers, 2015-04-03:
He perexg,

I'd love to do, put you did not checkin yet. The minisatip.patch under patches was updated 3 days ago...

Best

perexg, 2015-04-03:
Sorry, updated.

timehlers, 2015-04-03:
Dear perexg,

unfortunately this did not help. I still get only one tuner working. My setup is at follows:

/mnt/mnt/bin/minisatip -g -f -u 0:0-1210,1:1-1420,2:2-1680,3:3-2040

which is exactly the freqs of my unicable. I also tried -u 0:0-1210,0:1-1420,0:2-1680,0:3-2040, because I thought maybe it needs to be routed through tuner 0. But I always get only one tuner working (btw. I not sure if this is indeed tuner 0, I just asumed it, since there is the cable connected).

On the other end I have a VDR with vtuner setup for 2 concurrent tuner, itself giving via streamdev a stream to my raspberry pi. So if VDR starts on "Das Erste HD", my raspberry pi either receives all channels except the channels on transponder with "Das Erste HD", or only receives this transponder, depending which tuner is used for what.

BTW: LEDs both (for tuner 0 and 1) are turned on now.

How can I help to find the root problem?

And thanks for you effort!

Best

timehlers, 2015-04-03:
One correction! The LED is only turned on for the first tuner! I don't even get LEDs on for the other tuners.

timehlers, 2015-04-04:
I just thought about, that my log only contains a log with tuner 1. So you can't actually see how a second tuner would be activated and used, only how s2i.bin behaves at init. So I made another trace with 2 tuners active.

Maybe you can see there more?

Best

perexg, 2015-04-04:
Could you try the direct HTTP streaming using vlc and select the second, third and fourth tuner explicitly (it's the fe=<number> parameter - [1,]2,3,4). For Arte HD on 19.2E, the url is like this:

http://<your_box_ip>:8080/?fe=2&src=1&freq=11494&sr=22000&pol=h&msys=dvbs2&pids=0,5110,5111,5112,5113,5116

I'm a bit lost, how your clients allocates tuners. I'm testing this with tvheadend where the static allocation is used (tvh asks to tune selected tuner).

perexg, 2015-04-04:
Also, provide '-vvvvvvv' log from minisatip with description what you're trying to do and what's not working.

timehlers, 2015-04-04:
Hi perexg,

ok I streamed now from

tuner1 with vlc and minisatip -> minisatip-arte-tuner1.log -> OK
tuner2 with vlc and minisatip -> minisatip-arte-tuner2.log -> No Stream

call for minisatip was:
/mnt/mnt/bin/minisatip -f -u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -lllllll

Best

timehlers, 2015-04-04:
Regarding LED, I got a bit confused yesterday. Now with vlc I can tell you: LEDs always go on and off correctly for all 4 tuners. Sometimes, the ethernet LED goes off as well, when a tuner LEDs is switched off.

So in summary: tuner1 works as expected, tuner[2,3,4] does not work, but LED turns on and off when trying to use it.

Best

timehlers, 2015-04-04:
Another test: When changing the unicable freqs between tuner 1 and 2 like:

/mnt/mnt/bin/minisatip -f -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040

still tuner1 works. My guess is still, that the driver has to be told that the satellite cable is only connected on tuner1 and all 4 tuners have to get their signal from that input. How should the driver know if I have connected 1 or 4 cables.

timehlers, 2015-04-04:
I have now connected the unicable sat cable to tuner2 instead of tuner1, and now tuner2 works! But of course tuner1 does not work anymore.

And I have tested the cable on tuner4, too. Since you told me somewhere that tuner4 is not working on your setup. But also tuner4 works, when connected with a cable.

So my strong guess is, that we need to tell the driver, that only one cable is hooked up on tuner1.

Best

9000h, 2015-04-04:
yes thats the normal behavior with the maxlinear, online ohne  cable to tuner 1 with 4 freq.

timehlers, 2015-04-04:
Yes ok, but as proofed with current minisatip implementation every tuner wants to get his signal from own input. We (or perexg because I know too less about syscalls etc) need to find the ioctl to set all tuners in a mode to receive the signal from input of tuner1...

timehlers, 2015-04-04:
Hi, I made another try, just comparing the ioctl of s2i.bin between tuner1 and tuner2:

http://pastebin.com/Ca5cNtQW

The first differ block has exactly the ioctl, except the Hexcode is shifted from 0x7bead to 0x7bb8c. Which might be the different addresses of tuner1 and tuner2?

In the next block we also differ in the file descriptor 28 and 31, which may are the different device files for tuner1 and tuner2? And we have for file descriptor 3 different Values shifted from 0x7bb35 to 0x7bea9. Everthing seems to be the same until now.

But now some differences begin (not many). Does this point somehow into the right direction? Can you interprete that? Shall I do someting different?

Best

9000h, 2015-04-04:
maybe there is a general mode setting pessary in advance

1 unicable (1 cable)
2 standalone tuners (4 cables)
3 switch mode ( 4 cables VH VL HL VH)

perexg, 2015-04-04:
@timehlers : Could you try to connect cable to tuner1, start streaming from tuner1 (fe1) and then start streaming from tuner2 (fe2) simultaneously (using a different mux)? From log, the only different thing is that s2i.bin just starts both (1&2) tuners together (not only voltage).

timehlers, 2015-04-04:
@perexg: Just to clarify: you want to get the syscall-dump from s2i.bin, when I first start e.g Arte HD on tuner1 and then start tuning on tuner2 a different transponder, e.g. ZDF HD? This should be syscall-dump.das_erste+zdf.log already (using Das Erste HD and ZDF HD). But of course relatively close together and with VDR.

I can redo that with 2 vlc requests. Or do I understand you incorrectly?

On the other hand, just tuning a transponder on tuner[2,3,4] with cable on tuner1 must already reveal the difference, since this is not possible with minisatip at the moment.

Best

perexg, 2015-04-05:
@timehlers : My previous question was for minisatip.. The idea is that the first tuner must be active to get input on other tuners.

timehlers, 2015-04-05:
@perexg: Ok, I can try today very late, when I am back home. But in principle I have tested that already with VDR. As I said, my VDR has a vtuner+satip setup for 2 tuners. With your minisat (also with the newest patch-version) always one of the two tuners work (using both at the same time). After the tests with vlc revealed: it must be always that tuner having physically connected the satelite cable (normally tuner1). But of course I can recheck with vlc.

Best

timehlers, 2015-04-05:
Just checked with vlc. Only tuner1 works, a second, concurrent instance of vlc for tuner2 doesn't get a stream.

Best

perexg, 2015-04-05:
@timehlers : could you retry with the latest 'axe' branch from my minisatip fork: https://github.com/perexg/minisatip/tree/axe . I see that s2i send the unicable setup on tuner 1 also for tuner 2 . The last commit 21822f09c8b8febf4b9c3a37797e0db137dc2649  should handle it.

timehlers, 2015-04-05:
Dear perexg,

you are my personal hero! That's it. Now all 4 tuners work!

Wow!

Thanks!

perexg, 2015-04-06:
Closing now. I released second build of the satip-axe firmware with all minisatip fixes related to unicable.

closed 2015-04-06 by perexg