perexg/satip-axe #2: minisatip will not recognized by vdr/tvh or other satip client sw e.g. android/ios apps

mrarel, 2015-04-04:
hi perexg,
great job this first approach to enhance the firmware by minisatip ... it is not working so far with vdr plugin-satip or tvh, but i'm convinced there is not a lot whats missing...
Current problem:
A) after booting the gss.box with usb stick minisatip does not provide a desc.xml url to e.g. tvheadend or other server/client software. In tvh or vdr there are no tuners available...from minisatip.
B) login by telnet/ssh is working fine, but the enviroment for the shell e.g. for the /sbin path is not set. In /sbin there are several tools like minisatip etc.

by the way ... a nfs server and usb kernel driver would help too :-) e.g. for mounting additional space on usb port 2 or to setup shares....maybe a rootfilesystem on usb could be an option too ?

Sorry...maybe to many wishes at once ;-) 
if you need further support for testing, just drop a message....

Regards mrarel

perexg, 2015-04-05:
It seems that the tuners are not initialized properly. You should turn off the box and wait about 5-30 minutes and retry. I don't know why, but sometimes the fast on/off sequence causes these errors. You may use 'dmesg' to check for these errors - they're described in README or you may also use 'logread' - minisatip will also print tuner initialiazation errors.

perexg, 2015-04-06:
Release second build.. It contains mainly minisatip fixes.

perexg, 2015-04-15:
I believe that this issue is mostly eliminated now.

mrarel, 2015-04-19:
Yes i have tested your v7 firmware. There are only problems concerning continuity errors  when i stress all 4 tuners e.g. by tvh and 4 vlc clients.
never the less great work.

perexg, 2015-04-19:
@mrarel : Do you have this problem also with the original firmware ? I am experiencing this issue with it, too, so it really seems like an error in the driver (I hope that it's not a problem with the hardware design). The problem is also that this issue rises under different conditions. Anyway, I'll track it in issue #6 .

mrarel, 2015-04-20:
I will test with the same environment, which means i will stress 4 tuners under 1.16 gssbox firmware. I realy never tested this under original fw , so i ll be back with results. And yes, i hope it will not be a hardware dedign issue ....
regards mrarel

mrarel, 2015-04-20:
back again and tested....the original fw ! there are also "continuity errors" when you stress the 4 tuners of an gss.box with tvh (vlc as a frontend showing 4 different channels on different transponders ) :-(.
so i fear that you are right concerning the possibility that there is a "general" (maybe HW ?) issue with theses boxes/tuners...
all 4 tuners starts streaming WITHOUT continuity errors after a one ore more minutes at least one or two tuners are running into continuity errors...
that´s bad. I hope inverto will give you a feedback which could solve further problems...or at least the bring up an other fw with "corrections"...
regards
mrarel

dreamcat4, 2015-04-20:
@mrarel Hello! :)

Please can you test with 3 tuners instead? It would be nice to know at least if 2 tuners there are no continuity errors.

Also:

* Does it only get continuity error if all the 4 stream are in S2 (HD)?
* Did you test with 4xS2 hi-def (1080i) steams or 4xS1 standard def (576i) streams?

I am an IDL-400S owner too. It would be nice to know these things.
Kind Regards.

perexg, 2015-04-20:
Unfortunately, inverto does not respond. I use e-mails info@inverto.tv and stb.support@inverto.tv . Perhaps, if more people will bomb their support, they'll do something. I sent another e-mail there.

Actually, I'm learning more about the used chips - tuner STV6120 and demodulator STV0900. I have a partial datasheet for STV6120 - I can reduce the input gain (on amplifier) in STV6120 - the range is from 0dB to 16dB (inverto sets the 16dB of course). It seems that it helps a little but not much. The errors are before the STi7108 chip, so the minisatip or any other software in the user space cannot do much to improve the situation. For STV0900 - there is a Linux driver in the standard kernel, but appearently, this hardware uses the demodulator in a little different modes. And the full datasheet is available under NDA only. I'm trying to do some dumps of the status registers and compare them when it's working correctly and when not using a lot guessing. But it's a bit difficult without the full datasheet.

perexg, 2015-04-20:
Also another note: The another "weird" issue is that the fullmux tuning fails in some situations (a stream which provides only few kb/s is grabbed using the STi7108). I will also try to track it, perhaps, these two things are related.

CvH, 2015-04-20:
I belive @jollyjinx had contact with inverto back in the days to get the kernel sources, maybe he had a different contact

timehlers, 2015-04-20:
Otherwise tell us, what we should write (roughly). And we will SPAM them. :)

pbriesch, 2015-04-20:
I'live about 45 km away from Inverto. 

````
Inverto Digital Labs
18, Duchscherstrooss
Wecker, L-6868
Luxembourg
Tel. +352 264 367 1
Fax. +352 264 313 68
info@inverto.tv
````
If somebody prepares for me a list with the issues that need to be discussed, I could try to get Inverto's attention.

pbriesch, 2015-04-20:
A little bit off topic:
There is now a DVB-C box from Inverto available: http://www.inverto.tv/products/product.php?section=2&id=265

CvH, 2015-04-20:
As I called GSS they said something that the driver devs are at poland.

***ul. Poznanska 62/98
60-853 Poznan
Poland
Tel. +352 264 367 64
Fax. +352 264 313 68***

pbriesch, 2015-04-20:
+352 is the country code of Luxembourg. So all calls and faxes to Poland actually will end up in Luxembourg.

pbriesch, 2015-04-20:
By the way: Inverto's Headquater is about 6 km away from SES Astra's headquarter in Luxembourg.

perexg, 2015-04-21:
The STV6120 amplifier settings can be modified in build 8 which I released right now. The "chip" default value is 6dB.

perexg, 2015-04-22:
Good news: I have a little Inverto feedback - they asked for more details. Let see...

mrarel, 2015-04-22:
@dreamcat4 
I could test the scenario you´ve mentioned...e.g. with V8 of perexg firmware, where 6db is the "standard" and no longer 16db.

But I fear it does not "only" have something in common with the amount of tuners which are in use (also not related "firmware version"...stock FW or not..)
In the meantime I fear that it has more relation to "power" related issues and influence to the behavior of the tuners depending on how much "power" is needed e.g. to "switch" the LNB frequency /position, or something like that. Which leads to the problem that it could be something more "HW" related :-(.
 
I ´ll try to reproduce the "continuity" errors by testing : 
a) using all tuners on different transponders = Streaming at least 4 different channels to a client (vlc)
b) compare it to "streaming" ONLY 1 CHANNEL, but to 4 clients (= 4 "subscriptions" to only one tuner)

I think that there wont be "continuity" errors at scenario b ! this at least would proof that "tvheadend" is not the reason for it, but the amaount of tuners "in use" is for sure a problem....

I´ll give feedback as soon as I´ve tested both...

perexg, 2015-04-22:
@mrarel : v8 firmware can change the gain, but by default it is kept to 16dB like the original firmware.

Also, I changed the tuner initialization sequence in v8 again + I2C bus speed - so far, I've not seen any I2C init error with it using fast physical on/off and soft reboots with v8. So I think that I may hit the problem.

dreamcat4, 2015-04-22:
@perexg Great.

I have finally installed it today. After finding an old 256MB sd card (+ $1 usb micro SD card reader).  Thank you for this.

Quick question:

It says in the `dist/README` that you must remove the USB stick after flashing to prevent it re-flashing the internal every subsequent boot. However, can I not just delete or rename the file `idl4k.scr` and continue to leave the USB stick plugged in? It is not easy to get to the device, where we have put it near to the dish.

perexg, 2015-04-22:
@dreamcat4 : Yes, the renaming of idl4k.scr is sufficient. The USB stick should be mounted already in /media tree in the satip-axe firmware.

dreamcat4, 2015-04-22:
@perexg Thanks!

x23piracy, 2015-05-12:
any news to the problem? have you received more feedback from inverto?

Regards X23

perexg, 2015-05-12:
No, Inverto is silent. I sent a new query today.

x23piracy, 2015-05-14:
@perexg could you give me detailt what you wrote them?
I may can also write but i do it with my company email address, it's a big known company.

Maybe that speed up things.

Regards X23

perexg, 2015-05-14:
e-mail: stb.support@inverto.tv

I sent information about CC errors seen in the PTI when all 4 tuners are used - but it may occur randomly - I use 3 standard service subscription and one fullmux subscription (4 different muxes) to reproduce the CC errors in few minutes:

<pre>
# /sbin/axe-debug pti vdev 0
# /sbin/axe-debug pti vdev 1
# /sbin/axe-debug pti vdev 2
# /sbin/axe-debug pti vdev 3
</pre>

And also that fullmux subscriptions (pids=all) do not work for all tuners. From my experience, usually only two tuners work, others receive only few TS packets per second.

Helron, 2015-05-16:
I had and still have similar problems with my inverto IDL 400 box. This is what I received from their support when I sent the following email a couple of months ago:

---> inverto:
I bought an IDL 400s some weeks ago. I am sorry to say, but it has some problems. After a lot of testing and analyzis, I have concluded that tuner 3 and 4 does not handle HD channels.

Tuner:Channel
1: HD CHANNEL OK
2: HD CHANNEL OK
3: HD CHANNEL NOK (artifacts)
4: FREE

1: HD CHANNEL OK
2: HD CHANNEL OK
3: SD CHANNEL OK
4: FREE

1: HD CHANNEL OK
2: HD CHANNEL OK
3: SD CHANNEL OK
4: HD CHANNEL NOK (artifacts)

1: HD CHANNEL OK
2: HD CHANNEL OK
3: SD CHANNEL OK
4: SD CHANNEL OK

It seems that tuner 3 and 4 does not handle HD CHANNELS. Is this a known issue? I see a lot of other people having trouble with this device and similar devices (GSS.BOX).

<--- inverto: 
We tested this scenario in our lab and were not able to reproduce such a problem. Even 4 HD channels in parallel are running smooth.
 
Are you sure that the connection to your router is a 1 Gbit/s connection, if it’s only 100Mbit/s it might not be sufficient to stream 4 HD channels.
 
If you disable tuner 1 & 2 are you able to stream HD channel from tuner 3 & 4?
 
Best Regards/Mit freundlichen Grüßen,
 
Inverto Support,
 
Tel:    +352 264367 0
www.inverto.tv

---> inverto:
My network is 1Gbit/s.

The channels are all encrypted, and I am using dvbviewers recording service to stream the channels. If I disable 1 and 2, dvbviewer is not working. Only disable from the tail is working, not from the front. SAT>IP protocol problem between dvbviewer and IDL400s?

Is the fact that the channels are encrypted (decrypted using a softcam in dvbviewer) putting more load on the tuners/cpu of the IDL400s?

Are the tuners 4 separate tuners, or are they 2 dual tuners? Is 1 and 3 the same dual tuner, and 2 and 4 the other dual tuner? Then my observation makes sense. 

<--- inverto: 
we use 4 individual tuners, they are all separated. Only tuner 1&2 and 3&4 share the same demodulator.
My mistake about deactivating the tuners, the streams are looped internally and if you deactivate tuner 1 the loop will fail.  
 
If the channels are encrypted or not does not make any difference for the streaming.
 
Can you please check this:
1: SD CHANNEL ???
2: SD CHANNEL ???
3: HD CHANNEL ???
4: FREE
 
Best Regards/Mit freundlichen Grüßen,
 
Inverto Support,
 
Tel:    +352 264367 0
www.inverto.to

---> inverto:
I have done the checks you requested:

1: SD CHANNEL OK
2: SD CHANNEL OK
3: HD CHANNEL NOK
4: FREE

Then I stop channel 1 and 2, and it is still failing:

1: FREE
2: FREE
3: HD CHANNEL NOK
4: FREE

So, something is wrong with tuner 3 and 4, regarding HD. SD is ok. Then I suspect a broken demodulator.

Unfortunately I have voided my guarantee when trying to solve the problem. I read about a possible heat problem on the net, opened the case and kept it running with a cooling fan, while testing. Obviously it did not solve the problem. Anyway, the guarantee sticker is now broken.

<--- inverto:
In this case I also suspect a problem on the hardware side.
If you did not touch the hardware, your dealer should take this unit back under warranty. But finally it’s up to the dealer if he accepted this unit for return.
 
Best Regards/Mit freundlichen Grüßen,
 
Inverto Support,
 
Tel:    +352 264367 0
www.inverto.tv

---------------------------------------------------------------------
The dialog ended here, as I didn't bother trying to return the device. I am using the device with tuner 3&4 disabled, now using tvheadend as server. With only 2 tuners running I still see continuity counter errors on tvheadend, but not as frequent as before.  A restart of the box will not solve the problem. If I turn it off and leave it off for a few minutes, it solves the problem. It is not a heat problem....

CvH, 2015-05-16:
@Helron 
did you buy the box near after release ? If got very similar symptoms, maybe they fu up the whole hardware and fixed it with a new internal rev.

perexg, 2015-05-24:
Inverto does not reply as usual, so if you like - write an e-mail to stb.support@inverto.tv  that you have continuity issues (distorted audio/video) with their V1.16.0.120 firmware under load (simultaneous streaming from all four tuners) and/or after some time of box usage.

perexg, 2015-05-26:
OK, perhaps, another way to push Inverto more. Open letter - for the SAT>IP committee: https://etherpad.mozilla.org/3TmDljNc19

Also, if you like to extend / correct (grammar) the document, I'll appreciate it. Your signs are welcome, too.

CvH, 2015-05-26:
sorry no discuss area at etherpad

1 **"We (several geeks ..."**
- describe more who "we" are and what "we" do (dev at Tvh for ex.) at the moment it sound like "the angry mob" ;) , nobody know that "we" have a bit more knowledge as the regular consumer

2 **the problem description is too tech specific**
- at the "SAT>IP committee" are likely no engineers, better describe it plain simple

3 **"Signed by:"**
maybe more detail, see 1. (for ex)
Jaroslav Kysela, Czech Republic, responsible for Tvh Sat>IP implementation
Paul Briesch, Germany, consumer
...

pbriesch, 2015-05-26:
Hi guys,

what do we expect form this letter in which we want to have the idl4k hardware removed from the list of certified SAT>IP products?
a) Inverto fixing the problem
b) Inverto being more talkative to us
c) helping to become SAT>IP a success story

and here comes in something that could happen as well

d) pissing off Inverto
e) give some ammunition to some manager to eventually fire the engineer
f) blaming the lab that certifies the products not having done a good job 

It is really important to know who is getting the letter in his hands because it could be misunderstood. 
Furthermore, I have no clue how the other hardware on the market compares to Inverto's design.

pbriesch, 2015-05-26:
Since it helps turning off the box for 5 to 10 minutes I assume a thermal problem. I didn't personally notice yet TS packet loss, however, from time to time I had to power cycle the box to make it work again. After turning on the box it happened several times that Ethernet did not work.

x23piracy, 2015-05-26:
Hi,
i like to see a,b and c happen, nobody wants d but if that happens it's their own fault because the error is proofed afaik ;) e we cannot save the world! f why not if they doing it wrong?

Regards X23

pbriesch, 2015-05-26:
OK, but then we should not only say, if at all, take the idl4k hardware from the list. We should say directly what we want: a), b), c).

e) lets save the poor engineer who worked his butt off to make it work but was not given the time and tools to do it right by the management and is now not allowed to talk to us.

x23piracy, 2015-05-26:
I agree with making it clear that we don't want a war but a solution ;) e what is your suggestion to save him?

pbriesch, 2015-05-26:
I believe the engineer(s) did talk already to us a little bit.

 "We would like the management to realize the importance of fixing this issue to allow the smooth introduction and further growth of SAT>IP technology into the consumer market. As a knowledgeable and highly motivated group of software developers and promoters of SAT>IP we are willing to help Inverto to further improve the user experience of its products and therefore would like to continue the fruitful initial discussions with Inverto's engineering team which has unfortunately stopped suddenly." .

Something like this maybe.

x23piracy, 2015-05-26:
That sound great, why don't you add this to the letter? :)

pbriesch, 2015-05-26:
OK, I can put this in the letter if the all agree. Furthermore, I rather like to not talk about removing the hardware from the list. But I leave this call to perexg because he started this initiative.

dreamcat4, 2015-05-26:
> Since it helps turning off the box for 5 to 10 minutes I assume a thermal problem.

If the manufacturer can come out with some kind of non-liability-addmitance statement which permits me to add my own heatsink solution without voiding the 2-year warranty. Then that would be helpful for me as I have some spare heatsinks lying around and enough motivation to do that. Else if they bring out Rev B. hardware within < 2 years. Then if it turns out to be such a problem, can exchange whilst under the regular warranty. Those are the only practical solutions I know of.

pbriesch, 2015-05-26:
Maybe you can try to put a fan close to the box and see if there is a difference. Forced convection transfers heat much better than natural convection. Some airflow will go thru the slots of the housing for sure.

x23piracy, 2015-05-26:
btw. since i disabled one of the 4 tuners i never had to restart the box in the last two weeks (fw 8)

pbriesch, 2015-05-26:
On ebay.de you can find some "refurbished" boxes. No idea what exactly has been refurbished.

x23piracy, 2015-05-26:
links?

pbriesch, 2015-05-26:
Well, it says B-Ware:

http://www.ebay.de/itm/TELESTAR-DIGIBIT-R1-B-Ware-SAT-IP-Transmitter-SD-HD-LAN-WLAN-5310440BW-/321735794054?pt=LH_DefaultDomain_77&hash=item4ae8f29d86

http://www.ebay.de/itm/Telestar-Digibit-R1-B1-Set-Bundle-SAT-to-IP-Router-Client-Receiver-HDTV-DVB-S-/121659460492?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2047675.l2557&nma=true&si=Us5JZubsOO7tqEX3veyQYtRNqwk%253D&orig_cvip=true&rt=nc

perexg, 2015-05-27:
OK, next round, Removed the paragraph about the certification as suggested. Added few more notes to description, presented myself as tvheadend coder. My idea is to send this e-mail to inverto support (all known e-mails) to the SAT>IP contact e-mail and perhaps to Grundig / Telestar e-mail contacts.

pbriesch, 2015-05-27:
It would be good to address our letter to real persons. Otherwise the email will just be deleted by a intern or secretary. Please check this out: https://www.linkedin.com/vsearch/p?f_CC=133063&trk=rr_connectedness

dreamcat4, 2015-05-27:
You know, I just tried zapping between prev/next channels in showtime. And after about 6-7 sequential changes the picture got scrambled. I reboot the box and then it's back working. That is with just 1 tuner in use.

Problem is: sometimes that will happen when I am not around to reboot it. For scheduled recordings, and I am not home. Then the program won't record anything valid whatsoever in the DVR (tvheadend) - bad picture. So that is a real issue (missed programs), is not reliable. And if you go on holiday to schedule a series or a soap. Then maybe you get none at all, since are not there to correct the problem next day (for next episodes). It's rather a shame.

It could be scheduled easily a task to reboot every morning at 05:58 hours. However it will interrupt anything recording at that time (probably just adverts). However that still does not prevent the failed recording on primetime shows previous evening.

I mean, what if I miss Game of Thrones? And they did not repeat it again? I doubt any Inverto executives cannot empathize with that. Missing all of the beheadings, torture etc. How could they abide that? I would imagine they could not cope with that, and have already rectified the situation. Sitting happily at home with their own private advanced prototype v2 units and all problems fixed. Beheading, torturing the engineers that crossed them.

perexg, 2015-05-27:
@pbriesch : could you list the real persons in the pad ? I'm not in linkedin so I don't see the page.

pbriesch, 2015-05-27:
I checked my LinkedIn page and found that there are some people at Inverto which are directly connected to people I am directly connected to. This might be advantageous for our purpose.

x23piracy, 2015-05-27:
Have you talked to them?

pbriesch, 2015-05-27:
No, not yet. Will do tomorrow.

Helron, 2015-05-27:
@CvH 
I received my box mid february this year (www.satking.de).

@pbriesch 
I don't think the problem is thermal related. I voided my warranty by opening the case and installed a 12cm usb powered fan. Unfortunately it did not solve the problems.

pbriesch, 2015-05-27:
@CvH
Do you have photos of the mainboard you could share?

x23piracy, 2015-05-27:
I had contact to a telestar technician some weeks ago, i could add his email address to the letter.

pbriesch, 2015-05-27:
What did he say?

perexg, 2015-05-27:
I also don't think that it's thermal issue. I have heatsink on all chips (really all chips) with FAN and I can reproduce a part of the issue programatically (fullmux on all four tuners does not work - to make things more "magic" - the fullmux work on different tuners after each cold/warn reboot). So it looks like a "weak" initialization somewhere.

x23piracy, 2015-05-27:
I talked with him about a different thing but i found a really interesting comment from him in the email, in german he said:

Hallo,

ich habe gerade nochmal nachgeschaut.
Bei den M3 u Listen ist die Tunerzuweisung so nicht vorgesehen.
Es geht nur bei M3 UListen über Multicaststreaming oder beim RTSP Protokoll.

Da Sie ja mit RTSP arbeiten kontrollieren Sie bitte nochmal nach folgendem Schema:

(eckigen Klammern sind Platzhalter):

rtsp://[DigibitR1Adresse]/?src=[DiSEqC Befehl]&fe=[Tuner]&freq=[Satellitenfrequenz]&sr=[Symbolrate]&pol=[Polarisation]&msys=[Demodulations-Norm]&pids=[PAT,PMT,V-PID,A-PID]

Ein Beispiel:

rtsp://192.168.178.51/?src=1&fe=1&freq=10744&sr=22000&pol=h&msys=dvbs&pids=0,400,401,402

In diesem Beispiel hat der Digibit R 1 die IP-Adresse 192.168.10.51,

stellt DiSEqC-Position A (src=1),
empfängt auf dem 1. Tuner,

die Frequenz 10744MHz (freq=10744),

eine Symbolrate von 22Msys (sr=22000),

die Norm DVB-S (msys=dvbs),

eine horizontale Polarisation (pol=h) und die benötigten PIDs - PAT-PID, PMT-PID, Video-PID und Audio-PID - (pids=0,400,401,402)ein. Wenn die Eingabe fe=[Tuner] fehlt, erfolgt die Auswahl des Tuners automatisch.

Mit diesem Beispiel müsste Arte in SD über z.B. den VLC Player abgespielt werden.

Hier würde ich jetzt mal in die  fe=1 setzen, für Tuner 1,
oder fe=2 für Tuner 2.

Dann kann man ja sehen, ob Tuner 2 arbeitet.

Aber Vorsicht:
Wenn keine Antennenleitung, sprich kein Signal, am Tuner 2 ansteht, dann wird der Stream automatisch abgebrochen.

Was noch sein könnte:
Der Digibit R 1 neigt bei Pegeln über 80 dB an den Tunereingängen zum Übersteuern. Das kann dazu führen, dass der Empfang nicht mehr gewährleistet ist.
Das kann passieren, bei sehr kurzen Kabelwegen und großen Sat Spiegeln bzw. LNBs mit hohem Ausgangspegeln.

Bitte antworten Sie bei Bedarf nochmal über das Service Ticket, dann bekommt das auch mein Kollege.
Vielen Dank

Mit freundlichen Grüßen

[64 more replies]

closed 2015-08-06 by perexg