perexg/satip-axe #4: Artifacts in Streams

darkover70, 2015-04-10:
In Buidl 5 i get Artifacts in Live.-Streams.....

See post in : http://www.vdr-portal.de/index.php?page=Thread&postID=1241558#post1241558

perexg, 2015-04-11:
Resolved using modified /etc/sysctl.conf - see the forum link above for more details.

net.core.rmem_max=33554432
net.core.wmem_max=33554432
net.core.netdev_max_backlog=2000
net.ipv4.tcp_rmem=4096 65536 65536
net.ipv4.tcp_wmem=4096 65536 65536

closed 2015-04-11 by perexg

timehlers, 2015-04-12:
Unfortunatelly I have to reopen this (I'm not sure if this works), but using a raspberry pi (100 MBit/s Ethernet) as satip client. It is impossible to get a clean stream with:

#define ADAPTER_BUFFER 70*7*DVB_FRAME

Changing it back to

#define ADAPTER_BUFFER 7*7*DVB_FRAME

everything is working again. I am really surprised reducing the buffer solving the problem. But I was searching for the couse of my stream-problem now for hours and finally found this as the solution...

Best

Tim

perexg, 2015-04-14:
The problem is that rpi has slow ethernet card and also, there's 100Mbit connection only. When the UDP data are send in chunks using gigabit source, I'm sure that they're lost in the switch 1000->100 or in the slow USB ethernet adapter. I implemented/fixed the '-b' option, so you may use '-b 7' to set the DVR buffer to the old value to reduce chunking.

timehlers, 2015-04-14:
I'm not so sure, why the switch shall lose packet, but anyhow there is still a problem (or two :)):

1. even with the defaut (7*7*DVB_FRAME) is get quite seldom errors in stream,with original s2i.bin there are none. But this can be found later.

2. The problem with your option is, that you want KBytes! But DVB_FRAME ist 188, so 7*7*DVB_FRAME=9212, which is 8.99609375 KBytes. I tried 9 (9216). This already gives no viewable stream. Already one byte more than X*188 is unviewable.

So I can't set 9212, since ony integers are allowed as option. Could you change the option to bytes then?

Best