perexg/satip-axe #5: continuity errors when using tvheadend as server, oscam does not start by default

mrarel, 2015-04-13:
Hi perexg,
great work and a lot of progress with version -6 of your firmware !!

- initializing of tuners is much better than before. 
  I don´t need to let the device "powerless" for more than 3 minutes to avoid that the usb-fw will be loaded AND the tuners are initialized correct, thats great ! 

- All 4 tuners are working, 
...after some time there are "continuity errors" e.g. visible when I connect a tvhedend server and starting to stream more than 2 channels (on different transponders = 3 tuners are working).

- Oscam does not start by default...
no issue, but I´have the general question whether you are planning to decode the channels by the "SATIP Server" ? 
I fear this would´nt  work, because the cpu of the satip-server (GSSBOX and clone) is really to "weak", or am I wrong ?
in this case oscam is more or less a waste of cpu, or is it an simple alternative for an dedicated oscam server...?

We/you still would need a client (or Tvheadend  server, VDR or something like this  "in the middle) to decode the signal, after getting the keys from oscam....so I don´t understand why running oscam on this device ? if the device will never be able to descramble and stream the descrambled signal to the homenetwork (XBMC or other client...) 

I think the currently best alternative is to run a "banana pi" "on top" to the SATIP device, to run oscam/tvheadend on it ...and distribute the streams by tvheadend (= descrambled) to the house by TCPIP. 

regards
arel

CvH, 2015-04-13:
The soc is really to slow to decrypt on the fly, therefore the oscam server is only that you don't need to have a separate device (router, rpi ..) to connect your card reader and run oscam to read your keys (waste of energy and money).

perexg, 2015-04-13:
The oscam is started by default, but you may have an old config file. Use /etc/config.default as a template for /etc/sysconfig/config . OSCAMD="yes" , The hw has not power to descramble contents. The oscam server is provided only to extend the useability.

The rest of your questions - unfortunately, I must use the kernel drivers provided by Inverto for tuners. I believe they have also some bugs (seems timing related). I'm trying to get a contact in Inverto Digital Labs to ask for a fix. But I hasn't been successful yet.

mrarel, 2015-04-13:
Thanks for your fast feedback. Hope that Inverto will provide a fix for the tuner issues...
In the meantime i'm happy with the gss.box original firmware ....and running a really impressive stable banana pi with tvheadend as tv server for my small  openelec (amlogic !)  Selfmade  "tvh receiver/clients".  Descrambling, recording on a 2.5 hd, while streaming 4 hd streams is working fine .
the banana pi does not need more than 7w in total (incl. Hd and gb network running/streaming).

by the way...does minisatip in general connect to oscam (by default with dvbapi 5) and could stream a descrambled signal ?

regards mrarel

perexg, 2015-04-15:
Closing. CC errors are tracked in #6 .

And yes, new minisatip has initial code for descramling using oscam/networked dvbapi.

closed 2015-04-15 by perexg

fedecarnales, 2015-06-26:
The STi7108 chip supposedly has hardware descrambling (DVB-CSA2 and DVB-CSA3) on up to 6 live TS inputs. I wonder if it would be possible to use it on satip-axe to do the descrambling on the box itself.

Reference:
http://www.st.com/st-web-ui/static/active/en/resource/technical/document/data_brief/CD00256520.pdf
http://www.eeherald.com/section/new-products/nps201101213.html

perexg, 2015-06-26:
Yes and no. Inverto excluded the STAPI kernel module for the security processor. So we don't have a driver for it - otherwise oscam/stapi v5 code exists.

fedecarnales, 2015-06-26:
Are the STAPI kernel modules available at ftp://ftp.stlinux.com/pub/stlinux/2.4/stapi/ ? Or is that something entirely different?

perexg, 2015-06-26:
It's only kernel. The STAPI modules are available only through NDA.

fedecarnales, 2015-06-26:
Are these the kernel modules? http://yun.baidu.com/wap/link?shareid=1695939041&uk=658736778&third=1&dir=%2F708n%E5%90%AF%E5%8A%A8%2Fprog_s101%2Fmodules&page=1&

perexg, 2015-06-26:
Yep, but these modules must match the used kernel.

mrarel, 2015-06-26:
its getting closer...but not closer to the general problem with "continuity
errors2...which is sad ...

pbriesch, 2015-06-26:
@perexg 

Let me please ask you some naive questions:
1. Could the modules match the kernel by accident?
2. Can they be hacked to make them match (hex editor, ? )

CvH, 2015-06-26:
Would't it be enogh to have a fitting stsubt_core.ko, or did you need the source ?
For STx7111 (Spark HD receiver) are stsubt_core.ko avaible, isn't this close enough ? Or is there some protection that you can't bypass an module which is not compiled directly for this hardware (never needed this kind of kernel hacking) ?

perexg, 2015-06-26:
I don't know the stapi modules internals. As I wrote - it's under NDA. I guess if Inverto unselected the secure processor code, it might imply the other internal interfaces (APIs).