perexg/satip-axe #7: NTPD dont work

darkover70, 2015-05-01:
It seems that in /etc the file "ntp.conf" is missing.
The Daemon can´t start without this File.

I use the Build 8.

perexg, 2015-05-01:
It's busybox's ntpd which accepts -p argument. By default "-p 0.pool.ntp.org" should be passed to the ntpd daemon. Could you run /etc/init.d/ntpd script from ssh if it works. Perhaps, the daemon should be run after the network is initialized otherwise the daemon will stop.

darkover70, 2015-05-01:
Ok.
After run manually /etc/init.d/ntpd script over ssh it works.
But how can do it automatic ?
I get Network over DHCP.

x23piracy, 2015-05-04:
Hi,

try to shuffle the inittab:
(maybe ntpd as last?)

# Boot
::sysinit:/etc/init.d/rcS
# Console
ttyAS0::askfirst:/bin/sh
# Network
::once:/etc/init.d/network
# Time Network Protocol
::once:/etc/init.d/ntpd
# NFSD
::once:/etc/init.d/nfsd
# minisatip
::once:/etc/init.d/satip
# telnetd
::once:/etc/init.d/telnetd
# dropbear
::once:/etc/init.d/dropbear
# inetd
::once:/etc/init.d/inetd
# oscamd
::once:/etc/init.d/oscamd
# Others
::ctrlaltdel:/sbin/reboot
::shutdown:/sbin/swapoff -a
::shutdown:/bin/umount -a -r
::restart:/sbin/init

CvH, 2015-05-04:
also not working for me, need to hit ***/etc/init.d/ntpd*** to get a proper date

dreamcat4, 2015-05-20:
Right... I don't just need ntp only. But also to set the timezone. That is needed for running cron scripts at the correct time.

It seems `/etc/init.d` is not saved over reboots? So maybe that is a good idea too to have 2nd user remembered folder loaded after the read-only one?

Else any other solution that works well enough would be fine, thank you.

perexg, 2015-05-24:
I made build 9 (beta) - https://github.com/perexg/satip-axe/tree/beta/dist . Please test before I move it to master.

dreamcat4, 2015-05-24:
Hello. I can see that `ntpd` is working now, automatically straight after the firmware upgrade. Not sure how to set the timezone though. It is on `UTC` at the moment.

perexg, 2015-05-24:
Fixed in final build 9.

closed 2015-05-24 by perexg