perexg/satip-axe #8: No Issue more Request > TVHeadend as Backend?

x23piracy, 2015-05-04:
Hi,

is it possible that you include tvh?
Sorry i found no other place for a Feature Request.

Regards X23

mrarel, 2015-05-25:
after some research there were someone at the (more or less) german speaking forum called VDR Portal
http://www.vdr-portal.de/board19-verschiedenes/board52-andere-hardware/126321-initial-release-of-the-satip-axe-firmware-is-available-minisatip-for-digibit-r1-dsi-400-idl400s/index5.html?s=00489f9b9d2c8516242ac32d7c45d5942aa70a52

which mentioned to have tvh running on the box.....but i doubt (especially if you try to descramble a stream with oscam while streaming it to other satip-clients like openelec...) that tvh or at least vdr will be the way to go on this device.

maybe I´m wrong...

perexg fw "9" is stable here ...on a DSI400 (GRUNDIG) if there would be a tvh server on this stblinux device I would test it too....especially the ability in connection with oscam would be of my interest...

regards
arel

perexg, 2015-05-25:
@mrarel : the reporter (x23piracy) is running tvheadend on his box (I helped him with the cross-compiling); anyway, I don't plan to include tvh to my builds...

x23piracy, 2015-05-25:
Yeah only me :)

mrarel, 2015-05-25:
@x23piracy 
....that doesnt mean you have to be the only one for the rest of your life ;-)
maybe you can share the binary...or the knowledge how to cross compile...and i will test the whole thing on my gssbox 
greets mrarel

x23piracy, 2015-06-07:
Sure i can share that but it does not make really much sense.
Can you give me an email address?

Regards X23

mrarel, 2015-06-07:
Ich tippe wir können auch deutsch reden ;-)

tvh macht also keinen Sinn, da es nciht wirklich "läuft" (descrambling etc.
?)...oder weil Du keinen Anwendungsfall dafür hast ?
Bei mir sollten 2 - 3 kodi clients per tvh-plugin darauf zugreifen...da die
box aber bei mehr als 2 Tunern "in use" aber anfngt sich auf zu hängen
macht das keinen sinn...korrekt ?
grüße
Arel

x23piracy, 2015-08-12:
the tuner problem does not exist anymore, any chance that the lower cpu usage the firmware needs that there is maybe enough power for descrambling now?

mrarel, 2015-08-13:
As long as perexg would have the (not official ??) STAPI part from INVERTO there would be a chance to include HW descrambling which means that the box MAYBE could handle it.
This means TVH have to be compiled with HW descrambling code based on this INVERTO INTERNAL Libs/CODE...then I would see a chance. 
Without it...it stays as it is..a real cheap AND NOW STABLE (THANKS PERXG !!!) box for SATIP !

regards
mrarel

perexg, 2015-08-13:
@mrarel : The oscamd can handle stapi v5 descrambling like for the linux dvbapi. So, the descrambling part is completely independent from tvh (the app like tvh will receive descrambled MPEG-TS packets). The only issue is that the kernel part of stapi is missing in the idl4k modules, so oscam cannot set the keys in the secure coprocessor.

CvH, 2015-08-13:
@perexg
is there a chance that you can ***wish*** the inclusion from inverto ?
I don't know if there is some licence witch prohibit this or maybe they licensed to less for this platform.

x23piracy, 2015-08-13:
Lets ask for it, maybe a 50:50 chance :)

perexg, 2015-08-13:
I asked approx. two weeks ago - no response.

pbriesch, 2015-08-13:
Probably the chances are very low: Is there some other box on the market that uses the same SoC and the needed (and matching) kernel part, which could be "transplanted" to satip-axe?

godoll, 2015-09-05:
Though perexg already mentioned not going to include tvh I nevertheless would be grateful if it would be added (if it could be doable).

[even as a n00b I was able to compile a tvh backend on my spare SheevaPlug a few years ago, so I may be able to add it on my own.

However as I'm lazy I'm currently mis-using an OpenELEC 5.95.4, which 'comes' with a tvh backend, on a RaspPiB (shutting down kodi), originally just with a TeVii S660 DVB-S2 connected via USB. In the meantime it also gets the feeds from a Digibit R1 and the option to merge the Pi's function with the Digitbit sounds desirable ...]

HazCod, 2018-09-28:
Is there any news on this? Currently looking into DVB-S2 -> sat>ip -> tvheadend -> Plex.

perexg, 2018-09-28:
No further plans for this.

CvH, 2018-09-28:
ofc not really the the right person to ask, but may VDR maybe work ?
without decryption not that useful I guess :/