perexg/satip-axe #9: Feature request: nano text editor

dreamcat4, 2015-05-19:
Hello!
Sorry to ask about this:

Can we use buildroot to install `nano`? Is that possible?

But having `nano` is really useful for novices, who don't know the `vi` keystrokes. For exiting configuration files (eg for the boot up etc). It's important not to mess them up by accident.

In my case - i'd like to be able to set my `EDITOR=nano` for editing the crontab file `crontab -e`. Simple things.

dreamcat4, 2015-05-19:
Hi again,
I've just tried `crontab -e`. But unfortunately there is an error message:

```sh
~ # crontab -e
crontab: can't change directory to '/var/spool/cron/crontabs': No such file or directory
```

Not sure how to get around that yet. Is cron supposed to be working already on the satip-axe v8 firmware ?

dreamcat4, 2015-05-20:
OK. I can see cron is working if it is started manually:

```sh
mkdir -p /var/spool/cron/crontabs
crontab -e # or echo "$cron_line" >> /var/spool/cron/crontabs/root
crond
```
It just needs to be restarted after every reboot, And for a persistent directory for crontabs (or re-copied), if `/var/spool/cron/crontabs` will be forgotted.

dreamcat4, 2015-05-20:
Hmm. I create seperate issue for nano as had put 2 different things here.

But also: `cron` to be useful (for me), it also depends upon a working `ntpd` to set the system time correctly

https://github.com/perexg/satip-axe/issues/7

perexg, 2015-05-24:
I made build 9 (beta) - https://github.com/perexg/satip-axe/tree/beta/dist . Please test before I move it to master. This build has crond support in.

dreamcat4, 2015-05-24:
AFAIKT it's not persistent, but presumably we can now use `customd start` to start it up from 3rd party script every reboot.

```sh
/media/SD_256MB # crontab -e
crontab: can't change directory to '/var/spool/cron/crontabs': No such file or directory
/media/SD_256MB # mkdir -p '/var/spool/cron/crontabs'
/media/SD_256MB # EDITOR=nano crontab -e
/media/SD_256MB # reboot
/media/SD_256MB # Connection to digibit closed by remote host.
Connection to digibit closed.
id@emachines-e520:/dev$ ssh root@digibit
id@emachines-e520:/dev$ ssh root@192.168.1.22

Welcome to SATIP-AXE server 201505241045-9

~ # crontab -l
crontab: can't change directory to '/var/spool/cron/crontabs': No such file or directory
~ # 
~ # ps -a | grep cron
 1408 root       0:00 grep cron
```

dreamcat4, 2015-05-24:
Actually i can see now you added a start script for `crond`. However it isn't making the folder `/var/spool/cron` for some reason. And `crond` isn't starting up. I tried to run the start script manually, but it didn't seem to work.

```sh
/etc/init.d # ls -lsa /var/spool/
total 0
     0 drwxr-xr-x    2 root     root            40 Jan  1 12:00 .
     0 drwxr-xr-x    6 root     root           120 Jan  1 12:00 ..
/etc/init.d # cat /etc/init.d/crond 
#!/bin/sh

test -r /etc/sysconfig/config && . /etc/sysconfig/config

if test "$CROND" != "yes"; then
  exit 0
fi

mkdir -p /var/spool/cron
mkdir /etc/sysconfig/crontabs
ln -sf /etc/sysconfig/crontabs /var/spool/cron/crontabs

exec /usr/sbin/crond -f -S -c /etc/sysconfig/crontabs
/etc/init.d # /etc/init.d/crond 
/etc/init.d # ls -las /var/spool
total 0
     0 drwxr-xr-x    2 root     root            40 Jan  1 12:00 .
     0 drwxr-xr-x    6 root     root           120 Jan  1 12:00 ..
/etc/init.d # 
```

dreamcat4, 2015-05-24:
Ah right... maybe i need to set `CROND=true` in the `/etc/sysconfig/config` settings file.

closed 2015-05-24 by perexg