perexg/satip-axe #10: Feature request: nano text editor

dreamcat4, 2015-05-20:
Hello!
Sorry to ask about this:

Can we use buildroot to install nano? Is that possible?

But having nano is really useful for novices, who don't know the vi keystrokes. For exiting configuration files (eg for the boot up etc). It's important not to mess them up by accident.

In my case - i'd like to be able to set my EDITOR=nano for editing the crontab file crontab -e. Simple things. I just checked on my DD-WRT router and it has nano - similar memory / flash sizes.

x23piracy, 2015-05-20:
Hi,

i don't see nano as important because you are able to work with vi if you know this:

Command Mode : ESC
i : input
x : remove sign at cursor position
dd : remove complete line
/ string : search for string
:w : write
:wq : write, quit
:q! : quit, no save

Regards X23

9000h, 2015-05-20:
full ack

x23piracy, 2015-05-20:
Hi,

vi is the most essential editor under ux os so deal with it and you are independent ;)

Regards X23

dreamcat4, 2015-05-20:
Jesus. Well if you are going to think so backwards ways, personally (for me) it feels a lot safer to not trust vi for anything and just do things like:

```sh
echo "line1
line2" >> file

sed -i 's/string/replacement/'
# and so on
```

CvH, 2015-05-21:
VI has zero usability, you need to google how you can do such basic things like exit. 
I know this editor follows the linux mantra, If it can be done easy - complicate it as much as possible, but is this helpful? If you need to change for example the ip at the confs you need to learn vi instead of just using nano. Easy tasks, easy tools !?

perexg, 2015-05-21:
I'll include nano in the next release. BTW: Nice flame, but I also found vi as a non-intuitive editor for _very_ old *nix gurus, although I can edit the config stuff in it. I have EDITOR=joe on my systems (as I worked with Borland tools in past and joe follows the basic old wordperfect key mapping).

The only reason why vi is in satip-axe is that this editor is inside the busybox.

pbriesch, 2015-05-21:
Hi,
in case the TV program will be really boring, I suggest to have pacman4console included in the next release ;-)
https://www.youtube.com/watch?v=GTabAPSCesE

perexg, 2015-05-24:
I made build 9 (beta) - https://github.com/perexg/satip-axe/tree/beta/dist . Please test before I move it to master.

dreamcat4, 2015-05-24:
Yes. `nano` works well. Thank you!

closed 2015-05-24 by dreamcat4