perexg/satip-axe #11: ftp: it only downloads - no ftp uploads

dreamcat4, 2015-05-24:
tested on v9 with filezilla. Ftp cmd `STOR` - response: `500 Unknown command`

Reading works fine, just not the writing.

```sh
Status:	Starting upload of /sync/Downloads/dvb-tuner-si2158-a20-01.fw
Command:	CWD /SD_256MB
Response:	250 Operation successful
Command:	TYPE I
Response:	200 Operation successful
Command:	PASV
Response:	227 PASV ok (192,168,1,22,199,184)
Command:	STOR dvb-tuner-si2158-a20-01.fw
Response:	500 Unknown command
Error:	Critical file transfer error
```

If you can improve this before `v9` release, then would also have a reason to try out your new `upgrade-fw` command. A question about it though:

```sh
~ # upgrade-fw --help
Usage: /sbin/upgrade-fw FIRMWARE
```

Does not warn whether the firmware file must be put on the internal flash, USB or it doesn't mind. So does it mind about that? You could briefly mention the fact in your `dist/README.md` release notes.

E.g. I am assuming that the entire flash file copied to RAM before flashing, and is not being flashed upon a reboot?

perexg, 2015-05-24:
Look to dist/README file... The help is there for both issues:

1) copy /etc/inetd.conf to /etc/sysconfig/inetd.conf and add -w (write) option to ftpd
2) upgrade process is described in dist/README - https://github.com/perexg/satip-axe/tree/master/dist

closed 2015-05-24 by perexg

dreamcat4, 2015-05-25:
@perexg I appreciate all these impreovements. Including the `localtime` which I have just tried and works perfectly.

FYI: The instruction about ftp being read only, and needing to set the `ftp -w` Is not clearly documented in the README. Everything else - great!