perexg/satip-axe #12: Peristent Timezone

dreamcat4, 2015-05-24:
For busybox, it uses `ulibc`, which needs a `TZ` variable. It seems to be set in the file `/etc/TZ`.

Here is a real example from another busybox-based system (home router):

```sh
tomato:/tmp/etc# cat /etc/TZ
GMT0BST,M3.5.0/2,M10.5.0/2
```

More infos about busybox timezone can be found here:

http://comments.gmane.org/gmane.linux.busybox/20587

http://www.uclibc.org/FAQ.html#timezones

https://stackoverflow.com/questions/18588293/glibc-to-uclibc-timezone-mapping-file

dreamcat4, 2015-05-24:
I guess it may be solved either making `/etc/TZ` a permanment symlink to `/etc/sysconf/TZ`. And have users write their timezone to it's own `TZ` file:

```sh
ln -sf /etc/sysconfig/TZ /etc/TZ
echo "GMT0BST,M3.5.0/2,M10.5.0/2" > /etc/sysconfig/TZ
```

Or else have the system take the user's config value from a new `TZ=` variable within the config file. And make a script set the timezone from there at some point during system startup.

perexg, 2015-05-24:
It does not work this way. AXE firmware uses glibc. There are no timezone files in the firmware, but I added the localtime file copy operation to final build 9 - just place this file to /etc/sysconfig/localtime (using scp or any other method).

closed 2015-05-24 by perexg