perexg/satip-axe #14: Error message: unaligned program counter: 00e0 [#1]

pbriesch, 2015-05-25:
Hi,

I have attached an easymouse 2 reader via USB to /dev/ttyUSBlower. After setting it up in the webif and restarting OSCAM, OSCAM crashes and dmesg shows following:
`````
.
.
.
JFFS2 notice: (458) check_node_data: wrong data CRC in data node at 0x02a63da0: read 0xfa8c9874, calculated 0xde907551.
USB Serial support registered for cp210x
usbcore: registered new interface driver cp210x
cp210x: v0.09:Silicon Labs CP210x RS232 serial adaptor driver
USB Serial support registered for pl2303
usbcore: registered new interface driver pl2303
pl2303: Prolific PL2303 USB to serial adaptor driver
USB Serial support registered for FTDI USB Serial Device
ftdi_sio 2-1:1.0: FTDI USB Serial Device converter detected
usb 2-1: Detected FT232RL
usb 2-1: Number of endpoints 2
usb 2-1: Endpoint 1 MaxPacketSize 64
usb 2-1: Endpoint 2 MaxPacketSize 64
usb 2-1: Setting MaxPacketSize 64
usb 2-1: FTDI USB Serial Device converter now attached to ttyUSB0
ftdi_sio 4-1:1.0: FTDI USB Serial Device converter detected
usb 4-1: Detected FT232RL
usb 4-1: Number of endpoints 2
usb 4-1: Endpoint 1 MaxPacketSize 64
usb 4-1: Endpoint 2 MaxPacketSize 64
usb 4-1: Setting MaxPacketSize 64
usb 4-1: FTDI USB Serial Device converter now attached to ttyUSB1
usbcore: registered new interface driver ftdi_sio
ftdi_sio: v1.5.0:USB FTDI Serial Converters Driver
stmmac: Rx Checksum Offload Engine supported
	TX Checksum insertion supported
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
PHY: 1:01 - Link is Up - 100/Full
unaligned program counter: 00e0 [#1]
last sysfs file: /sys/bus/usb/drivers/ftdi_sio/uevent
Modules linked in: ftdi_sio pl2303 cp210x axe_dmxts(P) axe_dmx(P) axe_fp(P) axe_fe(P) i2c_mangle axe_i2c(P) stapi_ioctl(P) stapi_core(P)

Pid : 887, Comm: 		wr08-Reader-0
CPU : 0        		Tainted: P            (2.6.32.42_stm24_0208-idl4k_7108 #14)

PC is at 0xffffffe9
PR is at 0xffffffe9
PC  : ffffffe9 SP  : 81d75ed8 SR  : 40008001 TEA : c11e5ed0
R0  : 00000006 R1  : 00000000 R2  : 00000000 R3  : 00000000
R4  : 9e8000c0 R5  : 812525a0 R6  : 823b0d28 R7  : 00000006
R8  : 8100e196 R9  : 00000001 R10 : ffffffea R11 : 823c19e0
R12 : 2974de5c R13 : 823d5c00 R14 : 0051e4d8
MACH: 0000005c MACL: 5b509940 GBR : 2974e938 PR  : ffffffe9

Call trace:
 [<8108955e>] vfs_ioctl+0x1e/0x80
 [<81067b20>] handle_mm_fault+0x0/0x6c0
 [<81089d02>] do_vfs_ioctl+0x62/0x640
 [<81091d5e>] alloc_fd+0x5e/0x120
 [<8107aa56>] do_sys_open+0x96/0x10c
 [<8108a30e>] sys_ioctl+0x2e/0x60
 [<81007920>] syscall_call+0xa/0xe
 [<8108a2e0>] sys_ioctl+0x0/0x60

Process: wr08-Reader-0 (pid: 887, stack limit = 81d74001)
Stack: (0x81d75ed8 to 0x81d76000)
5ec0:                                                       81d83ee4 00000000
5ee0: 81d58580 8108955e 00000000 00000594 80045415 823c19e0 80045415 2974de5c
5f00: 81067b20 81089d02 00000005 2974de5c 00000101 00000001 00000000 81d32d88
5f20: 81d93e44 00000000 00000071 81d75f4c 81d75f38 ffffb7a3 81091d5e 00000005
5f40: 00000903 00000000 823c19e0 81d93ba0 00000002 00000000 00000000 8107aa56
5f60: 823c19e0 00000000 8108a30e 0051e4d8 00000000 00000594 80045415 2974de5c
5f80: 00000005 823c19e0 0051e4d8 00000001 81007920 00000000 00000071 00000100
5fa0: 8108a2e0 296693e0 00000000 fffffff4 00000036 00000005 80045415 2974de5c
5fc0: 2974e938 0051e1b8 00000000 2974de84 004029c4 2974de5c 00000000 0051e4d8
5fe0: 2974de58 296693e4 0041139e 00000000 2974e938 00000000 00000000 0000004c
---[ end trace 29624e837f865871 ]---

x23piracy, 2015-05-25:
Hi,

what about the oscam log? lets have a look at your reader conf in oscam.servers?

Regards X23

pbriesch, 2015-05-25:
I copied the config from a working installation and just changed the device.  Still need to hunt for the oscam log.

````
[reader]
label                         = reader
description                   = reader
enable                        = 1
protocol                      = mouse
device                        = /dev/ttyUSBlower
key                           =
password                      =
pincode                       = none
gbox_max_distance             = 2
gbox_max_ecm_send             = 3
gbox_reshare                  = 0
cccam_reshare                 = 5
readnano                      =
services                      =
lb_whitelist_services         =
reconnectdelay                = 60000
resetcycle                    = 0
keepalive                     = 0
smargopatch                   = 0
autospeed                     = 0
sc8in1_dtrrts_patch           = 0
fallback                      = 0
fallback_percaid              =
localcards                    =
cacheex                       = 0
cacheex_maxhop                = 0
cacheex_ecm_filter            =
cacheex_allow_request         = 0
cacheex_drop_csp              = 0
cacheex_allow_filter          = 0
caid                          = 0100
atr                           =
boxid                         =
boxkey                        =
rsakey                        =
ins7e                         =
ins7e11                       =
ins2e06                       =
fix07                         = 0
fix9993                       = 0
readtiers                     = 1
force_irdeto                  = 0
needsemmfirst                 = 0
ecmnotfoundlimit              = 0
ecmwhitelist                  =
ecmheaderwhitelist            =
detect                        = cd
nagra_read                    = 0
mhz                           = 368
cardmhz                       = 368
ident                         = 0100:000068
chid                          =
class                         =
aeskeys                       =
group                         = 1,2,3,4
emmcache                      = 1,1,2,0
blockemm-unknown              = 0
blockemm-u                    = 0
blockemm-s                    = 0
blockemm-g                    = 0
saveemm-unknown               = 0
saveemm-u                     = 0
saveemm-s                     = 0
saveemm-g                     = 0
blockemm-bylen                =
lb_weight                     = 100
lb_force_fallback             = 0
savenano                      =
blocknano                     =
dropbadcws                    = 0
disablecrccws                 = 0
use_gpio                      = 0
cccreshare                    = -1
ccchop                        = 0
use_ssl                       = 0
deprecated                    = 0
audisabled                    = 0
auprovid                      = 000068
ndsversion                    = 0
ratelimitecm                  = 0
ratelimittime                 = 0
ecmunique                     = 0
srvidholdtime                 = 0
cooldown                      = 0,0
read_old_classes              = 0

pbriesch, 2015-05-25:
Here the log that shows that the reader is not recognized.
````
015/01/01 12:00:02        0 s >> OSCam <<  cardserver started, version 1.20-unstable_svn, build r10663 (sh4-linux)
2015/01/01 12:00:02        0 s     (main) System name    = Linux
2015/01/01 12:00:02        0 s     (main) Host name      = satip-axe
2015/01/01 12:00:02        0 s     (main) Release        = 2.6.32.42_stm24_0208-idl4k_7108
2015/01/01 12:00:02        0 s     (main) Version        = #14 PREEMPT Sun May 24 20:57:46 CEST 2015
2015/01/01 12:00:02        0 s     (main) Machine        = sh4
2015/01/01 12:00:02        0 s     (main) The current system time is smaller than the build date (May 22 2015). Waiting up to 60 seconds for time to correct
2015/05/25 19:45:49        0 s     (main) creating pidfile /tmp/.oscam/oscam.pid with pid 830
2015/05/25 19:45:49        0 s   (config) services reloaded: 0 services freed, 7 services loaded, rejected 0
2015/05/25 19:45:49        0 s   (config) userdb reloaded: 9 accounts loaded, 0 expired, 0 disabled
2015/05/25 19:45:49        0 s     (main) signal handling initialized
2015/05/25 19:45:49        0 s   (config) 764 service-id's loaded in 13 ms
2015/05/25 19:45:49        0 s   (config) 3 tier-id's loaded
2015/05/25 19:45:49        0 s   (config) 942 provid's loaded
2015/05/25 19:45:49        0 s   (config) 1 entries read from oscam.ratelimit
2015/05/25 19:45:49        0 s   (reader) NC+ [mouse] creating thread for device /dev/ttyUSBlower
2015/05/25 19:45:49        0 s     (stat) loadbalancer: file /tmp/.oscam/stat not found
2015/05/25 19:45:49        0 s     (main) waiting for local card init
2015/05/25 19:45:50 29800468 h    (webif) webif: decompressed 160180 bytes back into 384088 bytes
2015/05/25 19:45:50 29800468 h    (webif) HTTP Server running. ip=0.0.0.0 port=8006

CvH, 2015-05-25:
I got the same problem (Easymouse 2 USB Premium), the oscam conf should be okay (there is no voodoo) but if you try to connect to the ttyUSBlower the system getting slow and crash. Device getting detected but you can't access it.

perexg, 2015-08-03:
There is an kernel issue (limitation) for ST7108 for USB. Appearently, the data cannot be fetched from or to stack directly. The Duckbox project gave me a hint and I created a workaround. The updated/fixed firmware with latest Inverto driver is here for testing:

http://s000.tinyupload.com/index.php?file_id=29378920033560522468

Please, report, if it works for you.

connaisseur, 2015-08-03:
@perexg : Your /etc/sysconfig/config file misses a matching quote char (") at line 57. It's the MINISATIP_OPTS

If not closed boot "fails". If fixed boot succeeds.

Took me a serial console and voiding my warranty. :-)

connaisseur, 2015-08-03:
@perexg : Disregard my last comment. Was a bad config /mnt/data/satip-axe. I've deleted it and started from scratch again.

Sorry. My bad!

perexg, 2015-08-06:
Fixed in build 10.

closed 2015-08-06 by perexg