perexg/satip-axe #15: Multiple unicable inputs?

fedecarnales, 2015-06-22:
Hi,

First of all, sorry if this is the wrong place to post a question like this, I couldn't find any other place to post it.

I want to buy an Inverto IDL 400S to run satip-axe on, but want to make sure it would work with my planned setup first.

What I plan to do is have two dishes, each pointing to a different satellite and each with an Unicable LNB. Is it possible to set tuner 1 and 2 to use LNB A, and tuner 3 and 4 to use LNB 2, or even better, use Unicable splitters and Diseqc switches to have both LNBs available on all four tuners?

Thanks in advance and congrats on all the fantastic work you've done so far on the project!

Fede

perexg, 2015-06-22:
Only basic unicable setup is supported, but I think that you can combine two unicable LNBs on one cable - just configure them to use different SCR ID and frequencies.

Anyway, you can modify the minisatip source code to support more complex installations.

fedecarnales, 2015-06-22:
Thank you very much for the speedy response! Do you know if the IDL4K hardware is actually capable of supporting such setup? I noticed that for the manual only lists input 1 as Unicable option, the other are listed as Universal/Quattro setup. I assume the other tuners are looped through internally to tuner 1.

fedecarnales, 2015-06-22:
Also, would the first scenario (SCR LNB A in input 1 and 2, SCR LNB B in input 3 and 4) work with the current minisatip/satip-axe? So for example, if I set tuner 3 to a certain Unicable freq, will it take that frequency signal from the cable in tuner 3 or does it always use the cable in tuner 1 when Unicable option is used? Thanks again.

9000h, 2015-06-22:
I believe the box with the MaxLinear Chip has a special mode for SCR with a single cable input on port1 similar to the "digital devices max s8" card.

perexg, 2015-06-22:
The original firmware support unicable only on input 1. The satip-axe firmware can choose unicable input, but only one unicable input is supported now. Other inputs can be standard DiseqC ones.. The slave tuners (sharing input) can be configured for both unicable / plain diseqc. Also the minisatip / satip-axe can configure the slot and frequency and position for unicable, so you can already create a complex wiring using unicable only and one input (cable).

fedecarnales, 2015-06-22:
Ok, so it should technically be possible to add multiple unicable inputs to satip-axe? If so, I'll buy the box and take a crack at modifying the code to support more than one unicable inputs. Thank you!

perexg, 2015-06-23:
Yes, you can use any input as source to any tuner/demodulator in this hardware. So, it's possible when the code is hacked to have multiple unicable inputs (cables),

fedecarnales, 2015-06-23:
Great, thanks!

x23piracy, 2015-09-01:
Hi,

look at this (its german): http://www.kodinerds.net/index.php/Thread/45366-Alternative-Firmware-f%C3%BCr-Telestar-Digibit-R1-Grundig-GSS-BOX-und-Inverto-IDL-400/?postID=247015#post247015

perexg, 2015-10-03:
I don't see what's wrong in README. Of course, user should know the meaning of values in -u argument . "minisatip -h" helps.

perexg, 2016-02-21:
Latest build 11 in issue #45 supports the multiple unicable coaxial inputs (multiple unicable groups) now.

closed 2016-02-21 by perexg