perexg/satip-axe #16: a warning in the readme for prospective buyers?

pvdputte, 2015-06-25:
Hi,

While reading up on SAT>IP receivers a couple of weeks ago, I stumbled upon this project and I was highly enthusiastic about the current and future possibilities. I ordered by hardware (digibit r1) one day later, expecting a device that actually works using the stock firmware (1.16), and hoping for your firmware to make it even more fabulous. (Your firmware is very nice, by the way.)

Just one problem: as you know, the device is actually crap for many reasons: heat issues, stream continuity errors, audio and video artefacts as a result, making the product utterly useless.

I should have known that 4 tuners and open-source firmware at that price was too good to be true :-)

Therefore, perhaps a small warning in the README that it's not quite recommended to buy an idl4k device for 'production use' until Inverto gets its act together. If they ever will :-/

P.S. I've tried fw 1.16, 1.17, satip-axe, sysctl buffer tuning, VDR, TVH, other computers, other switches, everything :-) I'm out of ideas and hope I can get my money back instead of trying home-made cooling solutions. It's a shame.

perexg, 2015-06-25:
It does not seem like a heating issue as you can see in #2 .

pvdputte, 2015-06-25:
Possibly not, which is why I won't attempt better cooling (voiding the warranty). Nevertheless, the device does not work properly when it leaves the factory. FTA channels work *reasonably* well if you don't push it, but when watching encrypted streams it's artefacts every couple of seconds, maybe because decryption suffers more heavily from missing data.

It's really a pity because SAT>IP in itself should be awesome.

Anyway, feel free to close this if you just don't feel like pissing off Inverto because of a warning to potential users :-)

9000h, 2015-06-25:
I cant tell if it's hardware or software related but probably the hardware is ok.
I did also give up with the Digibit R1 and a Panasonic TV as client I was not able to get it stable, now I have minisatip on a small pc and all issues are gone.

mrarel, 2015-06-26:
@pvdputte 
I ca tell you that my GssBox (which is simply a clone of the Digit R1) works more or less stable, artefacts (even on scrambled channels) are NO ISSUE !

My evinronment is a GB lan where a Banana Pi works as a tvheadend server for some openelec clients (over all 3). 

The real issue is what perexg and some others detected....if you use more than one tuner at the same time you are getting in trouble and getting "continuity errors, artefacts and so on...

My current solution is a workaround...just using 2 tuners with tvheadend !

As long as you (e.g. by setting up tvheadend zo only use the 2 tuners which have the best signal rate...) it is currently working rock stable !

the disadvanage is never the less the "loss" of 2 tuners in such a setup......

If INverto will help to cleanup the problem this devices are the  cheapest choice available in europe market to serve SATIP TV to more than 4 clients ...for only ~130 Euro !!

If you add a banana pi for less than 40 euro you also could stream descrambled channels to your house.with HD channel switching around 1s. I never had a tv set with an internal tuner which was so fast...by the way.

Hope to see the problems solved soon...

regards
arel.

9000h, 2015-06-26:
sorry to repeat myself, 2 official certified SAT>IP devices (Digibit R1 and a Panasonic tx-55asx579) with only one tuner do not work stable here over some hours, so for me it's clear after some software upgrades and suffering issues over several month last year at least the Digibit R1 has software issues which I do not have with the same TV same network setup but a different SAT>IP server.

pvdputte, 2015-06-26:
@mrarel Hmm, there's an idea. Actually, I bought the box it as a replacement for a single broken DVB-card, but my expectations kept rising thinking about the possibilities :-)

I could keep using the remaining cards, plus the satip box with one tuner and see if that's workable until Inverto gets it fixed. I'll let it run over the weekend as the primary tuner and see if that's stable.

Maybe I have better luck than 9000h. Then I might take the risk of going past the return period, hoping that Inverto gets it fixed some day.

9000h, 2015-06-26:
I do not expect any futher firmware update from Inverto exept there is major security bug.

perexg, 2015-06-28:
@9000h: You're wrong. Quoting Inverto words: "We do have an effort these days to improve the streaming mechanisms of the product. We should have our results by end of July."

9000h, 2015-06-28:
nice to hear, as I do not have to rely on the Digibit R1 I hope they will surprise us an the invest is saved.

x23piracy, 2015-06-28:
5 weeks to go ;)

pvdputte, 2015-06-29:
Well, apparently my box is stable with 1 tuner. Start using 2 tuners and I'm right back swimming in artefacts. Perhaps I should hold on to it then. At worst I'll be stuck with a 25% capable product.

closed 2015-06-29 by pvdputte

pvdputte, 2015-07-10:
FYI, it did not take many days until I got artefacts even with just one tuner active. So... I sent it back and ordered a Digital Devices Cine S2 Duo (dual tuner card, revision 6.5) and tried not to think about SAT>IP anymore.

But guess what? With the new card, I get artefacts on the very same paytv channels as with the Digibit. Less artefacts, but still at times totally unwatchable :s 

With my TeVii DVB-S cards in the same server still *no* problem on those channels (and I switched the cables for testing). The most logical explanation is that components on the new card are more sensitive to some source of HF interference... Probably causing more issues with some transponder frequencies (around 12.1GHz), the ones where my paytv subscription resides.

The first suspect is the NVIDIA GT620 which is right next to the DVB card, but Digital Devices support also suspects the power supply. The components on the TeVii are probably less sensitive (which has other drawbacks but at least I get watchable channels).

Why am I mentioning this here? Because all this time my Digibit was sitting right on top of my PVR server, right on top of the videocard and the power supply. Perhaps it's susceptible to similar interference sources? As I sent it back, I cannot test anymore whether moving it to the other side of the room would have made a difference. :-(

But maybe this can information can help somebody else who has his device right next to this PVR box.

pvdputte, 2015-07-10:
P.S. I clearly have a lot less artefacts after disabling Xorg on my PVR/home theater system and watching the same channels over the network on another computer, so it seems at least the video card has something to do with it in my case.

CvH, 2015-07-10:
@pvdputte OpenElec (current 6.0Beta2) should be your reference to rule out that you installed outdated packages and incompatible versions at your htpc. ***A misconfiguration is not ruled out !***

BTW The TeVii's are mostly crap (also the support), not surprising that they are less sensitive.
BTW the  DD Cine S2 are not crucial "better" then other cards, mainly expensive - if this card is not working others won't do it better if it is really the HF interference. 
small sidenote the DVBSky S952 V3 does the same for 100€ less and in kernel drivers (4.0+) ! (got 2 of this cards as Tvh / sat>ip server)

pvdputte, 2015-07-10:
@CvH thanks for the DVBSky tip. As I understand, DVBSky is the home-brand of dvbshop and these used to be Digital Devices OEMs, but the latest versions are from Chinese sources.

I'm happy for you that your experience with them is great, I'll keep that in mind when I need another card. But after the Digibit fiasco (*still* waiting for a reply from Telestar support...) I was afraid of going the "best-bang-for-the-buck" route once more and just wanted to pay $$$ for a card that works and has decent support, instead of having to pay for shipping defective stuff back and forth (40 euros out the window for my Digibit).

Well... except that the card does not work yet. But at least they reply to my e-mail within the hour :-) That's worth something :-D

pvdputte, 2015-07-20:
Grrrr... I just checked my supposedly broken TeVii card again in a different PC and there it works wonderfully... Which means I've spent $$$ and countless hours troubleshooting my Digibit and Cine S2 for nothing, it's just some other component in my original VDR box which has gone haywire, causing all the stream artefacts.

It would be a bit silly to send my Cine S2 back and re-order a Digibit so I'll leave it at that. Good luck to all SAT>IP owners :-)