perexg/satip-axe #18: No picture with build 9/Unicable/TV Headend/Raspberry 2

hermn, 2015-06-30:
Using your firmware with Digisat I can see much better Signal stength in TV Headend. But opening a channel in Kodi using Tvheadend just gives me back some frozen artefacts of the picture. As nobody else reports this problem, it might have to do something with my specific setup.

I'm using unicable connected to the first port of my Digisat. I'm booting your firmware from USB and I added to /etc/sysconfig/config a new line: MINISATIP_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040". To make it easier for the weak processor I tried MINISATIP_OPTS="-u 0:1-1420,1:0-1210" as well. I do not need more than two tuners.

Tvheadend (server) runs on a Raspberry 2 together with Tvheadend Client/Kodi. I added to /etc/sysctl.conf "net.ipv4.conf.default.rp_filter = 1
net.ipv4.conf.all.rp_filter = 1
net.core.rmem_max=33554432
net.core.wmem_max=33554432
net.core.netdev_max_backlog=2000
net.ipv4.tcp_rmem=4096 65536 65536
net.ipv4.tcp_wmem=4096 65536 65536"
because someone reported these parameters to helb getting rid of artefacts.

In TVheadend (version 3.9.1879) minisatip is shown correctly as additional TV adapter. Switching back to the original firmware, everything works fine. I would like to use your frimware anyway and I'm one of the guys hoping you will add TVheadend one day to the firmware, so I can switch off the Raspberry completely without missing the option to record day and night.

Using DVB Viewer your firmware works perfectly. I have no idea how to test with VLC, what most people seem to use, because using your firmware Digibit does not occure as Univesal Plug & Play Component anymore and the m3u list is not accessible anymore.

Any idea what I could try next? Is there any documentaion apart from the README in the download section?

perexg, 2015-07-01:
It seems that RPI does not receive all packets in your lan. Try 'MINISATIP_OPTS="-b 9"' .
You need to provide m3u yourself - '-p url' for minisatip - see help.

hermn, 2015-07-01:
Thank you, perexg!

You are right! It seems to be a simple network problem. I didn't have this possibility in mind, because everything works perfectly running the original firmware. I made a new test configuration now, connecting all boxes directly to a GigaBit Switch. And yes, there are still some artefacts and continuity errors, but the Live stream is running and is of good quality.

'MINISATIP_OPTS="-b 9"' leads to no picture at all. Tvheadend shows the subscription as idle and logs an error:
2015-07-01 21:38:10.000 subscription: 0002: No transponder available for subscription "192.168.178.21 [ VLC media player ]" to channel "ZDF HD".

I tried MINISATIP_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 9" as well, but same result.

It seems, there's more bandwith needed using the alternative firmware? Maybe this an additional argument for integrating TVheadend? Bandwith would only be shared by the clients, but not additionally between Digisat and TVheadend. ;-)

I would like to study the "help" that you mentioned. Where can I find it?

perexg, 2015-08-06:
The problem with -b 9 parameter was fixed in build 10.

closed 2015-08-06 by perexg