perexg/satip-axe #19: Power Management - Auto-Suspend / WOL

janeczku, 2015-07-01:
The GSS DSI 400 has a horrible power consumption footprint. Even in standby (no tuners active) it draws around 10 (!) Watt. It's a joke that STi7108 is being advertised with its "market leading energy efficiency": http://www.arm.com/files/downloads/ST_ARM_Seminar_ppt.pdf
Allegedly, the SoC has a "programmable power management API". Obviously this is not implemented (right) in the current Axe drivers.
      
To tame this power hungry beast, i propose that we implement a auto-suspend functionality. Essentially a timer that checks the status of the tuner FEs and triggers a poweroff/halt/suspend when all frontends have been idle for at least x minutes. Of course Wake On Lan needs to be working for this, too.
    
Since i haven't yet flashed the new firmware:
1. Does STi7108 support ACPI ? What does `cat /proc/acpi/wakeup` show?
2. Could we package `pm-utils` and `ethtool` in the next release to experiment with WoL?

/cc @perexg

CvH, 2015-07-01:
Supports Wake-on: ug
Wake-on: g
kernel is unknown if it supports suspend,
If we could get pm-utils build in would be great to test.
It is easy to fine tune https://github.com/dinacel/autoshutdown/tree/develop/src to reach a good working state (an default off inclusion would be nice).

But better wait till the "new+better" driver is there, because suspend is only another workaround for bad drivers - if inverto does "their" job it isn't necessary.

perexg, 2015-07-01:
ACPI is PC specific.... The issue is that you need to fight with the proprietary STAPI interface, but it's better to let Inverto resolve the driver issue int the first phase.

janeczku, 2015-07-01:
I think the power draw in idle state is due to the hardware design. I am not sure a revised driver will solve this nor are there any indications that Inverto has acknowledged this problem and is working to fix it.      
I also disagree with @CvH. Auto-suspend would not be a workaround - it's a feature. It means nothing other than implementing the "Standby" power state that you commonly find in any consumer embedded devices (think of your TV, Stereo or PC monitor). 
Also, Suspend power state is explicitly documented for the STLinux platform and according to the driver specs for the ST Synopsys GMAC Ethernet controller "Wake up on Lan feature through Magic and Unicast frames are supported".
http://www.stlinux.com/advanced/power-management/features/standby    
https://www.kernel.org/doc/Documentation/networking/stmmac.txt

CvH, 2015-07-02:
@janeczku
FYI **If I remember correctly** the box takes 3-5W after power on, "after" the first use you are about 10W because the/some tuners never power down again.

janeczku, 2015-07-02:
@CvH thats interesting info indeed. Would flashing satip-axe fix this behavior?

CvH, 2015-07-03:
@janeczku no due to the "broken" driver (all closed source) - hope i remember correctly and don't spread nonsense ;D

janeczku, 2015-07-03:
@CvH makes sense. This box is a curious thing. GSS seems to have already sunsetted their DSI 400. I wonder if Inverto actually engineered the hardware platform or if it is rather just a reference design by STM :-)