perexg/satip-axe #20: Oscam with CCAM Protocol

armaghanshahzad, 2015-07-26:
I have tried every posibility and need some guidence:

I have SAT>IP from Inverto IDL 400S and I am not able to decrypt any channel, My Configuration is:

oscam.conf:
[global]
logfile                       = /tmp/oscam.log
maxlogsize                    = 2048

[cache]
[dvbapi]
enabled = 1
au = 1
pmt_mode = 1
user = armaghan
boxtype = pc-nodmx
----------------------------------------------------------------------------------------------------------
oscam.user:

[account]
user                          = armaghan
monlevel                      = 4
keepalive                     = 1
au                            = 1
group                         = 1
cccignorereshare              = 0
cccstealth                    = 1
-----------------------------------------------------------------------------------------------------------------------
oscam.server:

[reader]
label                         = share1
protocol                      = cccam
device                        = host,port
user                          = userXXXX
password                      = passwordXXX
inactivitytimeout             = 30
group                         = 1
caid                          =
emmcache                      = 1,3,2
cccversion                    = 2.0.11
cccmaxhops                    = 3
cccwantemu                    = 1
ccckeepalive                  = 1
--------------------------------------------------------------------------------------------------

With above configuration, I can connect to CCCAM server 
[image]
 
[image]

[image]

--------------------------------------------------------------------------------------------------------------------------------------

Log:

2015/07/26 15:32:32        0 s >> OSCam <<  cardserver started, version 1.20-unstable_svn, build r10663 (sh4-linux)
2015/07/26 15:32:32        0 s     (main) System name    = Linux
2015/07/26 15:32:32        0 s     (main) Host name      = satip-axe
2015/07/26 15:32:32        0 s     (main) Release        = 2.6.32.42_stm24_0208-idl4k_7108
2015/07/26 15:32:32        0 s     (main) Version        = #14 PREEMPT Sun May 24 20:57:46 CEST 2015
2015/07/26 15:32:32        0 s     (main) Machine        = sh4
2015/07/26 15:32:32        0 s     (main) creating pidfile /tmp/.oscam/oscam.pid with pid 1684
2015/07/26 15:32:32        0 s   (config) userdb reloaded: 3 accounts loaded, 0 expired, 0 disabled
2015/07/26 15:32:32        0 s     (main) signal handling initialized
2015/07/26 15:32:32        0 s   (config) 842 service-id's loaded in 7 ms
2015/07/26 15:32:32        0 s   (reader) share1 [cccam] creating thread for device sc.cohosting.co
2015/07/26 15:32:32        0 s     (stat) loadbalancer: file /tmp/.oscam/stat not found
2015/07/26 15:32:32        0 s     (main) waiting for local card init
2015/07/26 15:32:32   520018 h    (webif) webif: decompressed 160180 bytes back into 384088 bytes
2015/07/26 15:32:32   520018 h    (webif) HTTP Server running. ip=0.0.0.0 port=8888
2015/07/26 15:32:32   51E228 p   (reader) sc.cohosting.co: resolved ip=.111.1.111.111
2015/07/26 15:32:32   51E228 p   (reader) share1 [cccam] connecting to ip:port
2015/07/26 15:32:32   51E228 p   (reader) share1 [cccam] proxy initialized, server sc.cohosting.co:34300
2015/07/26 15:32:33        0 s     (main) init for all local cards done
2015/07/26 15:32:33        0 s (emmcache) loaded 0 emmcache records from /tmp/.oscam/oscam.emmcache in 1 ms
2015/07/26 15:32:33        0 s (emmcache) loaded 0 emmstat records from /tmp/.oscam/oscam.emmstat in 0 ms
2015/07/26 15:32:33        0 s (anticasc) anti cascading disabled
2015/07/26 15:32:33   557EC0 c   (client) plain dvbapi-client granted (armaghan, au=auto (1 reader))
2015/07/26 15:32:33   557EC0 c   (dvbapi) dvbapi channelcache loaded from /etc/sysconfig/oscam/oscam.ccache
2015/07/26 15:32:33   557EC0 c   (dvbapi) Using /tmp/camd.socket listen socket, API forced to DVBAPIv3 (0), userconfig boxtype: 12
--------------------------------------------'

I am using client for Windows.

[image]

FTA OK but when I change to pay channel black screen.

I have spent a lot of time but no luck, nothing appears in Oscam log when I change to pay channel.

Please help.

Regards
Armaghan

pbriesch, 2015-07-26:
Please look at the README: https://github.com/manio/vdr-plugin-dvbapi
pmt_mode = **4**

pbriesch, 2015-07-26:
@armaghanshahzad 
Please note that at this time the Inverto box is missing a Linux kernel driver to support decryption by its hardware. The CPU of the box is to slow to descramble a channel. So even if your setup is fine you will not get a descrambled stream, maybe a still picture.

perexg, 2015-08-22:
It's dup of issue #17  .

closed 2015-08-22 by perexg