perexg/satip-axe #21: Unicable setup - one coax to all four tuners --> no data received by clients.

connaisseur, 2015-08-04:
I need assistance on setting up my MINISATIP_OPTS-String, as I got a little confused.

Currently I have:

MINISATIP_OPTS="-X 0 -u 0:0-1280,1:1-1382,2:2-1484,3:3-1586"

all four tuners a hooked up behind the Coax #0 connector (labeled "1"). But I do not receive any channels when scanning, neither on tvheadend (head), nor on a Panasonic SAT>IP-Client TV.

Do I need something else? E.g.a "-L"-Setting?

perexg, 2015-08-04:
The config looks good. Could you check the verbose debug output?

<pre>
touch /tmp/nosatip
killall -9 minisatip
minisatip -f -lllllllllll -X 0 -u 0:0-1280,1:1-1382,2:2-1484,3:3-1586
</pre>

connaisseur, 2015-08-04:
This was the result:
````
~ # minisatip -f -lllllllllll -X 0 -u 0:0-1280,1:1-1382,2:2-1484,3:3-1586
[04/08 18:26:49.000]: Setting unicable adapter 0 slot 0 freq 1280
[04/08 18:26:49.000]: Setting unicable adapter 1 slot 1 freq 1382
[04/08 18:26:49.000]: Setting unicable adapter 2 slot 2 freq 1484
[04/08 18:26:49.000]: Setting unicable adapter 3 slot 3 freq 1586
[04/08 18:26:49.001]: Starting minisatip version 0.1.84, compiled with AXE DVB API, DVR buffer 18424 bytes
[04/08 18:26:49.002]: New UDP socket 3 bound to 0.0.0.0:1900
[04/08 18:26:49.002]: setting multicast for 239.255.255.250
[04/08 18:26:49.004]: New UDP socket 4 bound to 239.255.255.250:1900
[04/08 18:26:49.004]: sockets_add: handle 3 (type 0) returning socket index 0 [0.0.0.0:0]
[04/08 18:26:49.004]: sockets_add: handle 4 (type 0) returning socket index 1 [0.0.0.0:0]
[04/08 18:26:49.004]: sockets_add: handle 5 (type 2) returning socket index 2 [0.0.0.0:0]
[04/08 18:26:49.005]: sockets_add: handle 6 (type 2) returning socket index 3 [0.0.0.0:0]
[04/08 18:26:49.005]: starting init_hw 0
[04/08 18:26:49.006]: AXE - cannot open /dev/axe/frontend-0: 512
~ # 
````

BTW: It's the version you've announced on Aug, 3rd, approx. 19h00 CET.

perexg, 2015-08-04:
It seems that the frontend initialization failed. Try to turn off the box for five minutes. Do you see these lines in the 'dmesg' output ?

<pre>
       [i2c] I2C bus error /errno -121/!!
        [i2c] I2C bus error /errno -121/!!
        [i2c] I2C bus error /errno -121/!!
        [pll6120] init error!
        [fta_fe] FE Initialization done, res -1
</pre>

perexg, 2015-08-22:
???

connaisseur, 2015-10-02:
As I've moved away from this HW platform, I close this issue.

closed 2015-10-02 by connaisseur