perexg/satip-axe #22: Ethernet does not work

pbriesch, 2015-08-10:
Hi,

I cannot connect to the box via Ethernet. The network light stays dark after the Knight Rider intro. The box does not get an IP address. However, the lights on the Ethernet socket blink as there were some traffic.

Any suggestions?

Paul

perexg, 2015-08-10:
Try idl4k.usbresetusb script to reset the old configuration stored in flash which may cause the boot failure. Just keep the .fw file and replace idl4k.scr with idl4k.cfgresetusb . But boot only once with this boot script otherwise the configuration will be renamed each time.

pbriesch, 2015-08-12:
Resetting the configuration did not work. However, I could fix the problem:
In the Webif of the original firmware I had to change the IP setting back from fixed to dhcp. Then it worked. There is some other user in the vdr-portal who reported the same issue: http://www.vdr-portal.de/board19-verschiedenes/board52-andere-hardware/p1250510-initial-release-of-the-satip-axe-firmware-is-available-minisatip-for-digibit-r1-dsi-400-idl400s/#post1250510

perexg, 2015-08-12:
It seems like a side effect of something, because the satip-axe firmware has own network config which should be independent from the original firmware (which stores this config to EEPROM). I'll try to look into this issue. Thanks.

perexg, 2015-08-12:
Unfortunately, I cannot reproduce this on my box (the booting process works fine with static network settings in the original firmare). I cannot do much without a dump from the serial console.

pbriesch, 2015-08-12:
Maybe you could mention this potential problem in the README recommending to set DHCP on in the original firmware.

perexg, 2015-08-12:
Done.

closed 2015-08-12 by pbriesch