perexg/satip-axe #23: droppt and skippt frames with dvbviewer recording service

bylderup, 2015-08-12:
I´m using dvbviewer and recording service 1.31.0.0. The beta Version satip-axe-201507310907-10 with new drivers ran rockstable - 4 Fullhd PayTV Streams without errors. But i have massive dropped and skipped frames after some minutes with the new firmware on all channels - with only one frontend active. Any hints to solve this problem ?

perexg, 2015-08-13:
Try the final 10 build.

x23piracy, 2015-08-20:
@bylderup is it solved? Please respond i think @perexg likes to close solved issues.

bylderup, 2015-08-20:
Sorry not solved  i am ill and can not test again before next week

perexg, 2015-08-22:
Perhaps you may try to lower the network write buffer using '-b 9' parameter for minisatip (see /etc/sysconfig/config - MINISATIP_OPTS) which means 9*1024 = 9216. The default value is 14 (14336 bytes). Or try to tune the UDP buffers in your system (but it's probably ok, because if I understand the report correctly, the streaming of more channels work fine).

bylderup, 2015-09-01:
Thanks for your support. I think you can close this issue. It seems that the dropped and skipped frames are not a problem from your Firmware. I think its a problem of kodi 15.1 / dvbviewer addon / FireTV / dvb recording service.  I have tested now with different other frontends - no problems. And I have learned : do not change versions of Software or firmware at the same time on different devices..   ;-)

perexg, 2015-09-01:
Thanks for the feedback. Closing.

closed 2015-09-01 by perexg