perexg/satip-axe #25: unable tp return to factory firmware (inverto idl 400)

ovidiu31, 2015-08-29:
I followed the below steps : (and not succeded to go back to factory firmware getting 404 error)
Recovery of the original firmware:

  - copy idl4k.recovery file as idl4k.scr file to the root directory on the USB stick
  - plug the USB stick to the upper USB port in box
  - turn on the box
  - box is accessible through http (web browser) - original firmware

perexg, 2015-08-30:
Make sure that you use a right USB stick (some sticks are not detected by uboot - mostly old - slow ones) and the stick should have vfat partition - the sticks without partition are not detected too.

dreamcat4, 2015-08-30:
But also make sure it's boot sector is of the traditional PC - MBR partition type. GUID formatted one's (such as apple partition table) may not be recognized.

x23piracy, 2015-09-01:
Hi,

i had to try 3 sticks until one worked, so don't give up that box does not like any usb stick.

Regards

ovidiu31, 2015-09-02:
still not succeded.

404 Not Found

The requested URL was not found

perexg, 2015-09-02:
Just to be sure - does the upper USB slot work for you ? If you put a stick there with the satip-axe firmware, do you see the stick mounted ?

If yes, could you try to run the USB version of the satip-axe firmware and tell me, if you see the LED blinking (from left-to-right and right-to-left in a loop (Knight Rider effect)) during the loading ?

ovidiu31, 2015-09-02:
usb working and stick is mounted as i am able to see files via network with winscp

perexg, 2015-09-02:
OK. Try to boot the USB satip-axe firmware ...

ovidiu31, 2015-09-03:
It worked ...booted from usb stick i had the Kit car knight rider effect

ovidiu31, 2015-09-03:
but when i try recovery steps to the original firmware still no go.

perexg, 2015-09-03:
Could you try this idl4k.scr script ?

http://s000.tinyupload.com/index.php?file_id=72912870796244270298

ovidiu31, 2015-09-03:
tested with new script ... knight rider effect is there ... but wen accessing webui ... still the satip-axe firmware and not the factory one

perexg, 2015-09-03:
there should not be the effect with this uboot script... use clean stick..

perexg, 2015-09-03:
Also try this script: http://s000.tinyupload.com/?file_id=02975968489752019172

ovidiu31, 2015-09-03:
on the memory stick is only idl4k.scr file nothing else ... first scrpipt ... 404 second script  404 also when i try to access webui

perexg, 2015-09-03:
There was one missing step with the original idl4k.recovery script - you need to turn off / on the box to boot to the original firmware. Fixed version is here:

https://github.com/perexg/satip-axe/tree/master/dist

Anyway, the last script I provided you, does exactly same, so I wonder why it does not work for you...

ovidiu31, 2015-09-04:
ill test again today with the missing step

ovidiu31, 2015-09-05:
nope ... still no go

perexg, 2015-09-05:
OK. I cannot help you without the serial console. It's an uboot issue. The script https://github.com/perexg/satip-axe/blob/master/dist/idl4k.recovery works on my hardware.

ovidiu31, 2015-09-06:
could be because i have flashed on emmc from original firmware to satip-axe?

perexg, 2015-09-06:
No, the original firmware is on different mtd partition than satip-axe firmware. I really don't understand, why the uboot scripts do not work for you. I tested them several times on my boxes.

Polarity issue: Do you have universal LNB or unicable ?

ovidiu31, 2015-09-07:
universal quad  LNB

perexg, 2015-09-07:
What's your minisatip configuration ? But it should work without any issues.

ovidiu31, 2015-09-07:
When i reach home i will power it on and provide you remote access to play around and see for yourself how is performing.

perexg, 2015-09-07:
I cannot do much remotely, just show me the 'ps ax | grep minisatip' then run minisatip in foreground with verbose log from console:

<pre>
touch /tmp/nosatip
killall -9 minisatip
minisatip -f -vvvvvvvvv
</pre>

And provide this log. Also, if you have a voltmeter, you may check if there are 18V/13V changes on the inputs (F-connectors). The voltage changes polarization.

ovidiu31, 2015-09-07:
~ # minisatip -f -vvvvvvvvv
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
minisatip: invalid option -- 'v'
[07/09 16:24:33.000]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
[07/09 16:24:33.025]: Initializing with 4 devices

perexg, 2015-09-07:
Ok, replace v with l

<pre>
minisatip -f -lllllllllll
</pre>

ovidiu31, 2015-09-07:
~ # minisatip -f -lllllllllll
[07/09 16:49:55.000]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
[07/09 16:49:55.002]: New UDP socket 3 bound to 0.0.0.0:1900
[07/09 16:49:55.002]: setting multicast for 239.255.255.250
[07/09 16:49:55.002]: New UDP socket 4 bound to 239.255.255.250:1900
[07/09 16:49:55.003]: sockets_add: handle 3 (type 0) returning socket index 0 [0.0.0.0:0]
[07/09 16:49:55.003]: sockets_add: handle 4 (type 0) returning socket index 1 [0.0.0.0:0]
[07/09 16:49:55.003]: sockets_add: handle 5 (type 2) returning socket index 2 [0.0.0.0:0]
[07/09 16:49:55.003]: sockets_add: handle 6 (type 2) returning socket index 3 [0.0.0.0:0]
[07/09 16:49:55.003]: starting init_hw 0
[07/09 16:49:55.010]: trying to open [0] adapter 0 and frontend 0
[07/09 16:49:55.016]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x434460
[07/09 16:49:55.016]: opened DVB adapter 0 fe:8 dvr:9
[07/09 16:49:55.016]: deleting pids on adapter 0, sid -1, pids=NULL
[07/09 16:49:55.016]: Delivery System DVB-S/DVB-S2 (AXE)
[07/09 16:49:55.016]: sockets_add: handle 9 (type 5) returning socket index 4 [0.0.0.0:0]
[07/09 16:49:55.016]: done opening adapter 0 fe_sys 6
[07/09 16:49:55.016]: trying to open [1] adapter 1 and frontend 0
[07/09 16:49:55.021]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x438c68
[07/09 16:49:55.021]: opened DVB adapter 1 fe:10 dvr:11
[07/09 16:49:55.021]: deleting pids on adapter 1, sid -1, pids=NULL
[07/09 16:49:55.021]: Delivery System DVB-S/DVB-S2 (AXE)
[07/09 16:49:55.021]: sockets_add: handle 11 (type 5) returning socket index 5 [0.0.0.0:0]
[07/09 16:49:55.021]: done opening adapter 1 fe_sys 6
[07/09 16:49:55.021]: trying to open [2] adapter 2 and frontend 0
[07/09 16:49:55.025]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x43d470
[07/09 16:49:55.026]: opened DVB adapter 2 fe:12 dvr:13
[07/09 16:49:55.026]: deleting pids on adapter 2, sid -1, pids=NULL
[07/09 16:49:55.026]: Delivery System DVB-S/DVB-S2 (AXE)
[07/09 16:49:55.026]: sockets_add: handle 13 (type 5) returning socket index 6 [0.0.0.0:0]
[07/09 16:49:55.026]: done opening adapter 2 fe_sys 6
[07/09 16:49:55.026]: trying to open [3] adapter 3 and frontend 0
[07/09 16:49:55.030]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x441c78
[07/09 16:49:55.030]: opened DVB adapter 3 fe:14 dvr:15
[07/09 16:49:55.031]: deleting pids on adapter 3, sid -1, pids=NULL
[07/09 16:49:55.031]: Delivery System DVB-S/DVB-S2 (AXE)
[07/09 16:49:55.031]: sockets_add: handle 15 (type 5) returning socket index 7 [0.0.0.0:0]
[07/09 16:49:55.031]: done opening adapter 3 fe_sys 6
[07/09 16:49:55.031]: done init_hw 1
[07/09 16:49:55.031]: Initializing with 4 devices
[07/09 16:49:56.233]: ssdp_discovery: bootid: 3 deviceid: 1 http: 192.168.1.6:8080
[07/09 16:49:56.233]: Discovery packet 1:
NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.1.6:8080/desc.xml
NT: upnp:rootdevice
NTS: ssdp:alive
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385::upnp:rootdevice
BOOTID.UPNP.ORG: 3
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1

[07/09 16:49:56.233]: Discovery packet 2:
NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.1.6:8080/desc.xml
NT: uuid:11223344-9999-0000-b7ae-000f6f030385
NTS: ssdp:alive
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385
BOOTID.UPNP.ORG: 3
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1

[07/09 16:49:56.234]: Discovery packet 3:
NOTIFY * HTTP/1.1
HOST: 239.255.255.250:1900
CACHE-CONTROL: max-age=1800
LOCATION: http://192.168.1.6:8080/desc.xml
NT: urn:ses-com:device:SatIPServer:1
NTS: ssdp:alive
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385::urn:ses-com:device:SatIPServer:1
BOOTID.UPNP.ORG: 3
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1

[07/09 16:49:56.235]: Dropping packet from the same UUID as mine (from 192.168.1.6:1900)

perexg, 2015-09-07:
Tune to the non-working transponder and provide the log as I asked. Also, you may try to measure the voltage on the box output (appropriate connector - the first one) if the driver sets/changes the voltage correctly.

ovidiu31, 2015-09-07:
[07/09 16:55:44.16292]: describe_adapter: sid 0, aid 0 => ver=1.0;src=1;tuner=1,81,0,0,12545,h,qpsk,off,0.35,dvbs,22000,56;pids=0,18,511,512,97,33
Message repeated 1 times
[07/09 16:55:44.16653]: Received SSDP packet from 192.168.1.5:2388 -> handle 3
[07/09 16:55:44.16653]: M-SEARCH * HTTP/1.1
MX: 5
ST: upnp:rootdevice
MAN: "ssdp:discover"
User-Agent: Linux/3.0.13 UPnP/1.0 LGE_DLNA_SDK/1.6.0 [TV][LG]49UB830V-ZG/04.11.18 DLNADOC/1.50
DLNADeviceName.lge.com: %5bTV%5d%5bLG%5d49UB830V-ZG
Connection: close
Host: 239.255.255.250:1900

[07/09 16:55:44.16653]: ssdp_reply fd: 3 -> 192.168.1.5:2388, bootid: 5 deviceid: 1 http: 192.168.1.6:8080
[07/09 16:55:44.16653]: HTTP/1.1 200 OK
CACHE-CONTROL: max-age=1800
DATE: Mon, Sep 7 16:55:44 2015 GMT
EXT:
LOCATION: http://192.168.1.6:8080/desc.xml
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
ST: urn:ses-com:device:SatIPServer:1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385::urn:ses-com:device:SatIPServer:1
BOOTID.UPNP.ORG: 5
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1

[07/09 16:55:44.16654]: Received SSDP packet from 192.168.1.5:2388 -> handle 4
[07/09 16:55:44.16654]: M-SEARCH * HTTP/1.1
MX: 5
ST: upnp:rootdevice
MAN: "ssdp:discover"
User-Agent: Linux/3.0.13 UPnP/1.0 LGE_DLNA_SDK/1.6.0 [TV][LG]49UB830V-ZG/04.11.18 DLNADOC/1.50
DLNADeviceName.lge.com: %5bTV%5d%5bLG%5d49UB830V-ZG
Connection: close
Host: 239.255.255.250:1900

[07/09 16:55:44.16655]: ssdp_reply fd: 3 -> 192.168.1.5:2388, bootid: 5 deviceid: 1 http: 192.168.1.6:8080
[07/09 16:55:44.16655]: HTTP/1.1 200 OK
CACHE-CONTROL: max-age=1800
DATE: Mon, Sep 7 16:55:44 2015 GMT
EXT:
LOCATION: http://192.168.1.6:8080/desc.xml
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
ST: urn:ses-com:device:SatIPServer:1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385::urn:ses-com:device:SatIPServer:1
BOOTID.UPNP.ORG: 5
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1

[07/09 16:55:44.16655]: Received SSDP packet from 192.168.1.5:2388 -> handle 3
[07/09 16:55:44.16655]: M-SEARCH * HTTP/1.1
MX: 5
ST: upnp:rootdevice
MAN: "ssdp:discover"
User-Agent: Linux/3.0.13 UPnP/1.0 LGE_DLNA_SDK/1.6.0 [TV][LG]49UB830V-ZG/04.11.18 DLNADOC/1.50
DLNADeviceName.lge.com: %5bTV%5d%5bLG%5d49UB830V-ZG
Connection: close
Host: 239.255.255.250:1900

[07/09 16:55:44.16656]: ssdp_reply fd: 3 -> 192.168.1.5:2388, bootid: 5 deviceid: 1 http: 192.168.1.6:8080
[07/09 16:55:44.16656]: HTTP/1.1 200 OK
CACHE-CONTROL: max-age=1800
DATE: Mon, Sep 7 16:55:44 2015 GMT
EXT:
LOCATION: http://192.168.1.6:8080/desc.xml
SERVER: Linux/1.0 UPnP/1.1 minisatip/0.1.84-axe1
ST: urn:ses-com:device:SatIPServer:1
USN: uuid:11223344-9999-0000-b7ae-000f6f030385::urn:ses-com:device:SatIPServer:1
BOOTID.UPNP.ORG: 5
CONFIGID.UPNP.ORG: 0
DEVICEID.SES.COM: 1
[... 95 lines cut ...]

[07/09 16:55:46.18650]: read RTSP (from handle 17 sock_id 8, len: 0):
TEARDOWN rtsp://192.168.1.6:554/stream=1 RTSP/1.0
CSeq: 7
Session: 1804289383

[07/09 16:55:46.18650]: closing stream 0
[07/09 16:55:46.18650]: sockets_del: 9 -> handle 19, sid 0
[07/09 16:55:46.18650]: sockets_del: 9 Last open socket is at index 8 current_handle 19
[07/09 16:55:46.18650]: closed adapter 0 for stream 0 m:-1 s:0
[07/09 16:55:46.18650]: deleting pids on adapter 0, sid 0, pids=NULL
[07/09 16:55:46.18650]: closing adapter 0  -> fe:8 dvr:9
[07/09 16:55:46.18650]: deleting pids on adapter 0, sid -1, pids=NULL
[07/09 16:55:46.18651]: AXE standby
[07/09 16:55:46.18656]: sockets_del: 4 -> handle 9, sid 0
[07/09 16:55:46.18656]: closing DVR socket -1 pos 4 aid 0
[07/09 16:55:46.18658]: sockets_del: 4 Last open socket is at index 8 current_handle 9
[07/09 16:55:46.18658]: done closing adapter 0
[07/09 16:55:46.18658]: closed stream 0
[07/09 16:55:46.18658]: minisatip.c:506 get_sid returns NULL for s_id = -1
[07/09 16:55:46.18658]: reply -> 17 (192.168.1.7:50734) CL:0 :
RTSP/1.0 200 OK
Date: Mon, Sep 7 16:55:46 2015 GMT
Cseq: 7
Session: 1804289383

perexg, 2015-09-07:
Incomplete. Try redirection to a file and upload it somewhere. Also, which client ? Have you tried vlc ?

ovidiu31, 2015-09-07:
using the certified dvbviewer windows app.
http://pastebin.com/wFwVhehi

perexg, 2015-09-07:
Could you try vlc with this url ?

<pre>
rtsp://gssbox1:554?fe=1&freq=12188&sr=27500&pol=h&msys=dvbs&pids=0,44,163,104,106'
</pre>

Use your IP adress instead gssbox1 . You may also test all four inputs - just replace 'fe=1' with 'fe=2' up to 4.

ovidiu31, 2015-09-08:
i will have to delay few days this test as i will be at IBC Amsterdam

ovidiu31, 2015-09-28:
unable to make it work it keep giving me errors

~ # minisatip -f -lllllllllll
[28/09 10:28:32.000]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
[28/09 10:28:32.000]: New UDP socket 3 bound to 0.0.0.0:1900
[28/09 10:28:32.000]: setting multicast for 239.255.255.250
[28/09 10:28:32.001]: New UDP socket 4 bound to 239.255.255.250:1900
[28/09 10:28:32.002]: tcp_listen: failed: bind(): Address already in use
[28/09 10:28:32.002]: RTSP: Could not listen on port 554
~ #

perexg, 2015-10-03:
OK. So it appears like an dvbviewer incompatibility or misconfiguration. I cannot do much without logs.

ovidiu31, 2015-10-04:
i ordered an serial cable to try to go back to the factory fw .... probably in 2-3 days  will have the cable.
yes dvbviewer lite for satip seems to be a little bit bugged

ovidiu31, 2015-10-11:
received the cable and connected via ttl to the inverto box ... what steps do i need to follow to restore back to original firmware?

perexg, 2015-10-11:
Follow: https://github.com/perexg/satip-axe/blob/master/README.md to get the idl4k> prompt.
Type 'print' to dump the configuration. I'll send you instructions based on this config.

ovidiu31, 2015-10-11:
here is the print 

idl4k> print
bootdelay=1
baudrate=115200
board=idl4k
monitor_file_name=u-boot.bin
load_addr=0x84000000
autostart=yes
silent=1
resetenv=eeprom write $load_addr 0x00080000 1000
updatefw=usb start;set filesize;fatload usb 0 $load_addr idl4k.scr;if test "${filesize}" != ""; then autoscr $load_addr; fi
bootargs=
mtdids=nand0=gen_nand.1
mtdparts=mtdparts=gen_nand.1:128k(env-nand0),4M(system-nand0),32M(fw1-nand0),32M(fw2-nand0),-(data-nand0)
filesize=BD
bootcmd=run updatefw;nboot.jffs2 $load_addr fw1-nand0
ethaddr=00:0F:6F:03:03:85
stdin=serial
stdout=serial
stderr=serial

Environment size: 560/16380 bytes

perexg, 2015-10-11:
Get firmware from this URL:

http://www.gss.de/typo3conf/ext/syt_gss_produktdatenbank/res/download.php?uid=152&puid=540&src=detail

Put both files to the root dir of the USB stick and reflash original firmware. The uboot config is fine.

Alternatively, you may reflash 'fw1-nand0' partition using these commands:

<pre>
set filesize
fatload usb 0 $load_addr idl4k.bin
nand erase fw1-nand0
nand write.jffs2 $load_addr fw1-nand0 $filesize
</pre>

If  any of above command fail, do not continue.

ovidiu31, 2015-10-11:
this is how was going

U-Boot 1.3.1-47 (Sep 14 2012 - 18:39:46) - stm24_0056-B0_80-i                                                                                                                                                 dl4k-rev47

DRAM:  256 MiB
NAND:  128 MiB
SPI:   2 MiB
In:    serial
Out:   serial
Err:   serial
Hit any key to stop autoboot:  0
(Re)start USB...
USB:   scanning bus for devices... 2 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s                                                                                                                                                 ) found
reading idl4k.scr

264 bytes read
## Executing script at 84000000
reading idl4k.bin
.............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 .............................................................                                                                                                                                                 ............................... [... cut ...]
Automatic boot of image at addr 0x84000000 ...
## Booting image at 84000000 ...
   Image Name:   Linux-2.6.32.42_stm24_0208-idl4k
   Image Type:   SuperH Linux Kernel Image (gzip compressed)
   Data Size:    19194047 Bytes =  18.3 MiB
   Load Address: 81000000
   Entry Point:  81001000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel  hw=16001000000000000000000000000000 sn=1S81C11301000601 uuid=0C565CF306D9424FA94EAB9B3FA4D535 revBoot=stm24_0056-B0_80-idl4k-rev47 - 0x00000000 - 0 ...

ovidiu31, 2015-10-11:
all is ok back to normal

Thank you very much @perexg 

This issue was solved.

Perexg having a small question related to tvheadend ... whats the parameter from configure file to compile tvheadend with hardware CI support???

perexg, 2015-10-11:
OK.  The CI support is available with --enable-dvben50221 . You must have libdvben50221 installed in your system.

closed 2015-10-11 by perexg