perexg/satip-axe #27: UDP/RTP Multicast Streaming

newspaperman, 2015-09-03:
Hi perexg,

do you think it's possible to do multicast streaming with some tools like mumudvb or dvblast?
Is it possible to get these tools running in your firmware?

perexg, 2015-10-09:
Does anyone know about a simple tool which will resent the HTTP MPEG-TS stream to UDP/RTP (multicast) ? I can include it to the distro.

The above tools must be patched to get them work (the idl4k DVB driver uses incompatible API to linuxdvb) and I'm not involved to do that.

Jalle19, 2015-10-11:
@perexg I think you can hack together a wrapper that uses curl and socat (see http://www.dest-unreach.org/socat/doc/socat-multicast.html)

newspaperman, 2015-10-11:
@perexg: incompatible api sounds complicated. Can you point me to an application that was already modified t match the idl4k driver?

Jalle19's suggestion should work, I think netcat (nc) works as well.
I wouldn't use a bash wrapper piping from curl to socat/netcat.
A better way would be using libcurl simple api and using udp socket's in C.
If you like I could write a wrapper and post it here.

perexg, 2015-10-12:
Only my modified minisatip actually shows the diferences. Anyway, minisatip has already support for the RTSP multicast transport. So you may subscribe the multicast RTP stream using vlc (or any subscriber with the LIVE555 RTSP header to avoid subscription timeouts).

perexg, 2016-02-29:
Resolved in issue #45 (python script).

closed 2016-02-29 by perexg