perexg/satip-axe #28: RPi2 stream errors

CvH, 2015-09-17:
Hi,
I'am using AXE v10 with MINISATIP_OPTS="-Q -b (tested 1-20)" and was not able to get an error free stream at my RPi2 (OpenElec). 
The least problems I got with VDR where I'am able to watch with 1 big stream error / 10-30sec (changing -b made no real difference).

With Tvh is much more difficult, I got it working (-b 6) that it is viewable (big errors every 30-60sec) for 20-30min, after that time I get enormous errors in stream (every ~5-10sec). Also I can force errors if I hit reload at complex sites like github at browser at my pc which are connected to the same switch. Tvh only works if I activate 1 tuner, as soon I activate 2 or more tuners (even with idle scan off) I get immediately errors (1 error/sec). My other htpc's and my pc with DVBViewer/VLC ... working without an problem at my pc (see network conf ⬇).

**I thought this is an exclusive RPi2 network problem (google), but yesterday I tested the "official" beta fw (idl4k-1.0.0.146.bin) where everything works great.**
Even with 4 activated tuners at Tvh and activate idle scan I get 0 errors for hours.

As soon as I boot to AXE again I get stream errors again.

Obviously this should be an AXE (minisatip?) problem.
Any idea how to fix that ? (if any logs are needed just ask)

About my network:
[image]

perexg, 2015-09-18:
As I wrote somewhere, the problem is packet loss in the network between 1G -> 100M connection. Perhaps, the original firmware does a packet delaying which is not very effective. I would suggest to set 100M network speed through ethtool in the satip firmware.

I don't plan to write any own packet scheduler for satip, but I agree that it might be useful to use TCP/HTTP protocol instead UDP/RTP for the data transfers where TCP should ensure that data are retransmitted . The idea is that RTSP gives a ticket value for the new TCP/HTTP connection, but it's not defined in the RTSP standard and it will be specific feature. Unfortunately, I don't have any errors on my full gigabit network, so these plans have very low priority.

CvH, 2015-09-18:
At the moment the connections are handled as rtsp/udp right? To switch to rtsp/tcp should be enough due to package re-send?

The minisatip readme says *"Supported protocols are rtsp (both tcp and udp)"* but I can't find any switch to force it nor to change it at all (client/server).

perexg, 2015-09-18:
Yes, minisatip supports embedded data through RTSP connection (TCP), but it's not much useful for SAT>IP - imagine that you have big data queue and you need to send a quick stop/change command. The processing might be delayed. Also, TVH does not support this transfer method yet.

CvH, 2015-09-18:
Force 100mbit is working 
>ethtool -s eth0 speed 100 duplex full autoneg on

error free stream **but** if I create some traffic (100mb/s) to my NAS (see network chart) I get a complete broken stream. This wasn't the case with 1gbit O_o 
This shouldn't have any impact at all or did I completely misunderstand how a switch is working ?!

perexg, 2015-09-19:
There is a limited buffer and also the "dumb" switches use a simple queues, so some packets might be droped  without explicit port priorities.

Anyway, I created this: https://tvheadend.org/issues/3092

Although I don't like the embedded RTP data in RTSP connection, this protocol should be supported by tvh, too.

CvH, 2015-09-20:
Thank you for your effort, it try *soon™* a managed switch maybe this helps.

perexg, 2015-10-07:
The RTSP/TCP support is in... If you have any problems, create a new issue in the tvh bugtracker.

CvH, 2015-10-07:
testing asap, the minisatip version from axe is already supporting this feature ?

CvH, 2015-10-08:
Tested a bit and can report that this fix this problem completely.
Even with 1gbit network (at the AXE box) it works perfectly without an single error for >3h. (Tvh stats say 0/0/0)
This wasn't possible with force to 100mbit + buffer adjustments and smart switch at udp!

Only activate the rtsp/tcp option at Tvh and your done, no need for any modification at the axe box and this without any disadvantage from my point.

Thank you :)

closed 2015-10-08 by CvH

perexg, 2015-10-09:
Note: If you get RTSP error -22 for this TCP mode, then it's fixed in v4.1-581-g935f983 .