perexg/satip-axe #29: 2 out of 4 positions fail to tune

clandmeter, 2015-09-20:
Ive just tried to flash satip-axe to my gss box, but not all my positions seems to work.
In my normal setup i have 2 usb tuners which work on all 4 positions A1,A2,A3,HB but when i disable them and start to use satip, only position 1 and 2 will work, 3 and 4 dont. After a few tests i did get a picture on position 3 (Astra 3) but after trying again it fails again, after that i was never able to get a picture on the other positions.
Am i missing something here, some setting that is different for satip-axe? My gss box did work before on original firmware.

-carlo

CvH, 2015-09-20:
Sounds like you forgot **MINISATIP_OPTS="-Q"** at settings if you use quattro LNB.

clandmeter, 2015-09-20:
I'm not using a quadro lnb. I use regular twin lnb's with two 4p switches.

clandmeter, 2015-09-20:
this is my config

```diff
~ # diff /etc/config.default /etc/sysconfig/config
--- /etc/config.default
+++ /etc/sysconfig/config
@@ -55,7 +55,7 @@
 # minisatip
 #
 MINISATIP="yes"
-MINISATIP_OPTS=""
+MINISATIP_OPTS="-e 0"

 #
 # inetd (httpd,ftpd)
```

perexg, 2015-09-20:
Only diseqc 1.0 switches are supported (A/A, A/B, B/A and B/B positions). The uncommited diseqc messages are not supported in the minisatip version used in satip-axe yet. What's version of your diseqc switch ?

clandmeter, 2015-09-24:
Thanks for the clarification on irc. I am currently using it without switches which works fine. When i run into this issue again ill reopen it.

closed 2015-09-24 by clandmeter