perexg/satip-axe #31: Upgrade minisatip version

m4rcs, 2015-10-04:
Hi, 

I was trying to get minisatip working with oscam. During debugging I noticed that -o parameter gets totally ignored. This is due to the old version of minisatip (commit: 54df9348e7bd7e6075f54f1b93ec4ad36429abe0). 

Please update minisatip to the latest version.

Greets Marc

9000h, 2015-10-04:
I had asked also long before #24

perexg, 2015-10-05:
Upgrading to recent minisatip requires some work (many patches and fixes specific to the axe boxes are there). I concentrate to tvheadend at the moment. Also, someone tested the sh4 processor speed regarding DES and the speeds are very low. I suggest to not do the descrambling on this box until we can enable the hardware support (yes, hw can do descrambling in hardware), but we don't have proper STAPI drivers from Inverto ( issue #4 ).

perexg, 2016-02-16:
Descrambling speed for dvbcsa library on sh4 CPU used in idl4k:

<pre>
# /tmp/benchdec 
* CSA decryption bench *
 benchmarking with 256 TS packets
 benchmarking with 512 TS packets
 benchmarking with 1024 TS packets
 benchmarking with 2048 TS packets
 benchmarking with 4096 TS packets
 benchmarking with 8192 TS packets
 benchmarking with 16384 TS packets
 benchmarking with 32768 TS packets
 - 65280 packets proceded, 3.8 Mbits/s
</pre>

m4rcs, 2016-02-16:
Is this good or bad?

perexg, 2016-02-16:
One SD channel which uses a standard bandwidth.. Useless.. I put this value just for a reference.

Anyway, I have new minisatip working (without descrambling).

perexg, 2016-02-16:
Resolved in #45 .

closed 2016-02-16 by perexg