perexg/satip-axe #33: Old (from previous tune) MPEG-TS packet issue

perexg, 2015-10-09:
It seems that driver passes old data (from previous tuning) to the minisatip under some circumstances which confuses clients (TVHeadend) - mostly noticable in the network scan procedure. It seems that dropping first 30 MPEG-TS packets resolves this issue. I'll add an option to minisatip to skip the defined amount of first MPEG-TS packets.

prilly, 2015-10-09:
Thank you. so its not possible to flush buffer upon retune instead of skip 30?

PsychoRS, 2015-10-26:
I solved the issue setting the "Skip TS Packets" to 200 under the TV Adapter configuration on tvheadend. Maybe 200 is too high, but I just tried that amount, worked nice and didn't change anymore :D

prilly, 2015-10-26:
perexg is geting good result with =30, i found that a little to smal so i use a value =40

yigido, 2015-11-23:
I tried to set "Skip TS Packets" but it dosn't work for me. The first mux was found and OK but for the rest i get "scan no data, failed". I tested it with stock beta and last satip-axe.

perexg, 2015-11-23:
@yigido : Then your issue is different. This workaround does not resolve "no data" issues.

yigido, 2015-11-23:
Can you help me to fix this issue?

perexg, 2015-11-23:
@yigido : Create another issue and describe your setup in detail. You have probably something misconfigured. Or ping me on IRC - #hts / perexg .

perexg, 2016-02-12:
I did some more hacking and here's new package (beta for 11 release):

http://s000.tinyupload.com/?file_id=78443875428972064634

List of changes:

https://github.com/perexg/satip-axe/blob/master/dist/README (upcoming)

dreamcat4, 2016-02-14:
Thanks for this improvement (packet skipping) @perexg.

Just to clarify something minor was wondering about: Does this solve the 'scan no data failed' msgs during network tuning? Was this issue present / absent on the Inverto OFW?

Those are mostly rhetorical questions / not to worry too much.

perexg, 2016-02-14:
It's same functionality as TVH's 'Skip TS packets'. Nothing else. I'm not aware of any tuning issues with tvh / satip-axe (except with the network driver locks - issue #43 ).

dreamcat4, 2016-02-14:
ok. Thanks @perexg.

perexg, 2016-02-16:
Resolved in #45 .

closed 2016-02-16 by perexg