perexg/satip-axe #34: box stopped working

CvH, 2015-10-19:
Box worked for 1 month without any problem, yesterday it freezes.
Put it 1h away from power and get this log:
http://pastebin.com/95gger2w

>jffs2_scan_eraseblock(): Node at 0x03146ffc {0x1985, 0xe001, 0xe0021985) has invalid CRC 0x00000044 (calculated 0x515918d5)

and

>[i2c] I2C bus error /errno -121/!!
>[demod900-2] i2c bus error, unable to read ChipID

looks a bit strange to me, any ideas ?

perexg, 2015-10-19:
The jffs2 might be have inconsistencies sometimes, but it should be harmless. Check the configuration file, if it's corrupted.

The second problem is bigger - the demodulator chips don't communicate with the main CPU. Try to reboot several times (with/without poweroff for 1-5 minutes) and try to eventually boot to the original firmware.

BTW: I have both boxes open with fans which blow to demodulators and tuners, because I don't think that the temperature conditions are good for this hw using the metal cover.

CvH, 2015-10-19:
Boot to official beta fw works after first boot. Waited 1 min then rebooted to axe -> works.

I got the box at the attic, there is at the moment ~10°C but I`ve stressed the tuners a bit for ~3-4h maybe a overheat. Could you share a picture how you cool your box ?

CvH, 2015-10-26:
After 7 days got the same problem (besides the fs corruption) http://pastebin.com/kTD1RwPU

>[demod900-2] i2c bus error, unable to read ChipID
>[dmx] ... init 0
>[i2c] I2C bus error /errno -121/!!

Only 1 tuner at work and the temperatures are really ok (1min after crash checked).
The problem start (~1week later) after I activated the TCP feature at tvh - maybe pure accident ?
I'm back to official beta fw to see If it happens here again.

CvH, 2015-11-01:
surprise, after 7 days the same problem again even with official beta but it only fails to stream - box is responsive and a reboot via webui was possible -> working again
Is there some kind of cron job at the driver blob which happen after 7 days (a bit too lucky that this happens every 7 days)? 

Full log from the box, maybe it helps http://pastebin.com/5fD3Tjhh (big!)

closed 2016-03-02 by CvH