perexg/satip-axe #35: Disable LNB voltage (13-18V)

PsychoRS, 2015-11-18:
Hi,

I wanna put two of this boxes at home, for having 8 tuners. I only need 1 band (L) and 1 polarity (V) for the channels I need.

I have only a single LNB and changing it to a Twin LNB and having another 100 meters of cable from the roof of the building to my home is not an option.

Is there any way of disabling the 13-18v polarity selection in minisatip for the Inverto? Polarity won't be a problem, I will configure in tvheadend the second Inverto Box as low priority so the box will only be used when the first one  4 tuners are used, so the LNB will be receiving the voltage from the first box.

So, I need to set an Inverto Multibox to NOT send the 13/18V signal to my LNB. Is this possible?

PsychoRS, 2015-11-18:
Wait, I found something that may be will work...

If I change my LNB to a Inverto Unicable one... can I use the single cable drop I have, and put the Unicable Spliter before the two boxes, and then use the 8 tuners with no issues of polarity/band?

perexg, 2015-11-18:
You should have an unicable LNB for 8 receivers. Then you can just connect the two boxes using a simple splitter with power pass-through.

But perhaps a cheaper solution might be to buy a multiswitch for one satellite and 8 outputs and connect only one L/V input. In this way, the other inputs will be without signal, but the multiswitch will assure that the LNB is switched permanently to L/V combination.

PsychoRS, 2015-11-18:
Or, maybe, as I only need the L/V combination... can just do the trick with a simple 2-way splitter with power pass.. as I have minisatip configured with -L 0:0,0:1,0:2,0:3 so I only need to use the cable on input 1.

Don't mind me, just doesn't know that the LNB wouldn't ever receive "double" the voltage using two boxes... So I can just connect the two boxes to my LNB using a splitter and all the 8 tuners (2x4 multiboxes) will receive L/V without problems.

Thanks perexg, you can close this if you want to.

closed 2016-02-12 by perexg