perexg/satip-axe #36: Network scan not work

yigido, 2015-11-24:
When I try a network scan, the first Mux is successfully found, but for the rest I get an error message "scan no data, failed".
I use a digibit r1 with the last beta (I tried also the satip-axe) and tvheadend from git with following configurations:

[image]

[image]

[image]

The digibit shows this errors in the log:
> [686][355717.590][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[687][355736.543][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[688][355746.550][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[689][355756.548][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[690][355766.551][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[691][355776.550][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]
[692][355786.546][s2i-sock-util][0x13b34b8][listen:ERROR:utils/s2iutilsock.cpp:346][: [error: -1 errno:95 meaning:[Operation not supported]]]

[tvh_log.txt](https://github.com/perexg/satip-axe/files/42816/tvh_log.txt)

perexg, 2015-11-24:
Provide '--trace satip,mpegts' for tvh...

yigido, 2015-11-24:
[tvh_trace_satip_mpegts.txt](https://github.com/perexg/satip-axe/files/42870/tvh_trace_satip_mpegts.txt)

perexg, 2015-11-24:
I'm afraid. There should be [TRACE] lines in the log. See: https://tvheadend.org/projects/tvheadend/wiki/Traces

yigido, 2015-11-24:
Sorry my fault...I have updated the log from the previous post. 
Now seem the first 4muxes working...

perexg, 2015-11-24:
There's an issue on your ethernet network. I see many messages like:

 RTP discontinuity (31519 != 31520)

It means that some UDP packets were lost on the ethernet segment. Usually, the issue is 1Gbit -> 100Mbit conversion or some ethernet switches with small buffers.

If you can try again satip-axe firmware, you may use TCP RTSP connection mode..

yigido, 2015-11-24:
OK i will test it... Thx perexg for help

yigido, 2015-11-25:
So I tested it, but get the same error "scan no data, failed". I use the "RTSP/TCP" option and set "Skip TS packets" to 30. The log shows no discontinuity problems.

[tvh_minisatip.txt](https://github.com/perexg/satip-axe/files/43711/tvh_minisatip.txt)

perexg, 2015-11-25:
OK, I see the problem now. Some PIDs are not subscribed by TVH. Which version do you use ? You've not fetched tags, so iI cannot determine it from logs.

yigido, 2015-11-25:
The git from yesterday, but I don't know the exact version number. Where can I find it?

perexg, 2015-11-25:
<pre>
git pull --tags origin master
git describe
</pre>

yigido, 2015-11-25:
Ok I get the commit "851aebb5aa5f68272cf2ed5af5c83de38e78fb44"

perexg, 2015-11-25:
Could you provide same log with the patch bellow?

<pre>
diff --git a/src/input/mpegts/satip/satip_frontend.c b/src/input/mpegts/satip/satip_frontend.c
index d6d0a68..5dfdedd 100644
--- a/src/input/mpegts/satip/satip_frontend.c
+++ b/src/input/mpegts/satip/satip_frontend.c
@@ -61,6 +61,7 @@ satip_frontend_signal_cb( void *aux )
   signal_status_t        sigstat;
   service_t             *svc;
 
+  tvhtrace("satip", "signal_cb : mmi = %p, sf_tables = %d", mmi, lfe->sf_tables);
   if (mmi == NULL)
     return;
   if (!lfe->sf_tables) {
</pre>

yigido, 2015-11-26:
[tvh_patched.txt](https://github.com/perexg/satip-axe/files/44848/tvh_patched.txt)

perexg, 2015-12-16:
Fixed in v4.1-1233-gc50d70e . Closing.

closed 2015-12-16 by perexg