perexg/satip-axe #37: Minor fixes to Makefile

flyingfridge, 2015-11-26:
Had to do minor changes to Makefile to get satip-axe built:
- The dropbear releases get moved into the 'releases' subdirectory when they get old
- when nano is built without the --disable-utf8 configure option it wants to link against libncursesw, but STLinux 2.4 only supplies libncurses

diff --git a/Makefile b/Makefile
--- a/Makefile
+++ b/Makefile
@@ -297,7 +299,7 @@ busybox: apps/$(BUSYBOX)/busybox
 #
 
 apps/$(DROPBEAR)/configure:
-       $(call WGET,https://matt.ucc.asn.au/dropbear/$(DROPBEAR).tar.bz2,apps/$(DROPBEAR).tar.bz2)
+       $(call WGET,https://matt.ucc.asn.au/dropbear/releases/$(DROPBEAR).tar.bz2,apps/$(DROPBEAR).tar.bz2)
        tar -C apps -xjf apps/$(DROPBEAR).tar.bz2
 
 apps/$(DROPBEAR)/dropbear: apps/$(DROPBEAR)/configure
@@ -456,6 +458,7 @@ apps/$(NANO)/nano: apps/$(NANO)/configure
          CFLAGS="-O2" \
        ./configure \
          --host=sh4-linux \
+         --disable-utf8 \
          --prefix=/
        make -C apps/$(NANO)

flyingfridge, 2015-11-26:
```
diff --git a/Makefile b/Makefile
--- a/Makefile
+++ b/Makefile
@@ -297,7 +299,7 @@ busybox: apps/$(BUSYBOX)/busybox
 #
 
 apps/$(DROPBEAR)/configure:
-       $(call WGET,https://matt.ucc.asn.au/dropbear/$(DROPBEAR).tar.bz2,apps/$(DROPBEAR).tar.bz2)
+       $(call WGET,https://matt.ucc.asn.au/dropbear/releases/$(DROPBEAR).tar.bz2,apps/$(DROPBEAR).tar.bz2)
        tar -C apps -xjf apps/$(DROPBEAR).tar.bz2
 
 apps/$(DROPBEAR)/dropbear: apps/$(DROPBEAR)/configure
@@ -456,6 +458,7 @@ apps/$(NANO)/nano: apps/$(NANO)/configure
          CFLAGS="-O2" \
        ./configure \
          --host=sh4-linux \
+         --disable-utf8 \
          --prefix=/
        make -C apps/$(NANO)
```

perexg, 2016-02-01:
Applied. Thanks.

closed 2016-02-01 by perexg