perexg/satip-axe #39: No streaming possible. Channel is tuned. Only black screen

crazy888, 2015-12-16:
ok i tried hard to get it work ........ but no sucess.

what is working:
i can stream tv with stock firmware without any problems.

what i have done:
- upgrade firmware from satip-axe/firmware/ with sucess.
- flash satip-axe/dist/satip-axe-201508061051-10.fw with sucess.

axe is running. i can log into with ssh.

minisatip with running with 
1311 607 root S N 1772 0.4 0 0.0 minisatip -f -g -u 0:0-1748,1:1-1864,2:2-1980,3:3-2096

Unicalbe frequencies are correct.

But when i start streaming a channel i have only a black window.
I see tuner1 led is on.

Log say:
[spoiler]

~ # logread
Jan 1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.23.2
Jan 1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan 1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan 1 12:00:02 (none) daemon.info init: starting pid 604, tty '': '/etc/init.d/network'
Jan 1 12:00:02 (none) daemon.info init: starting pid 605, tty '': '/etc/init.d/ntpd'
Jan 1 12:00:02 (none) daemon.info init: starting pid 606, tty '': '/etc/init.d/nfsd'
Jan 1 12:00:02 (none) daemon.info init: starting pid 607, tty '': '/etc/init.d/satip'
Jan 1 12:00:02 (none) daemon.info init: starting pid 608, tty '': '/etc/init.d/telnetd'
Jan 1 12:00:02 (none) daemon.info init: starting pid 609, tty '': '/etc/init.d/dropbear'
Jan 1 12:00:02 (none) daemon.info init: starting pid 610, tty '': '/etc/init.d/inetd'
Jan 1 12:00:02 (none) daemon.info init: starting pid 612, tty '': '/etc/init.d/crond'
Jan 1 12:00:02 (none) daemon.info init: starting pid 613, tty '': '/etc/init.d/oscamd'
Jan 1 12:00:02 (none) daemon.info udhcpc[604]: udhcpc (v1.23.2) started
Jan 1 12:00:02 (none) daemon.info init: starting pid 616, tty '': '/etc/init.d/customd'
Jan 1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_rsa_host_key
Jan 1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan 1 12:00:02 (none) local0.notice root: DHCP deconfig
Jan 1 12:00:02 (none) cron.info crond[612]: crond (busybox 1.23.2) started, log level 8
Jan 1 12:00:02 (none) authpriv.info dropbear[609]: Not backgrounding
Jan 1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan 1 12:00:02 (none) daemon.warn mountd[606]: Kernel does not have pseudo root support.
Jan 1 12:00:02 (none) daemon.warn mountd[606]: NFS v4 mounts will be disabled unless fsid=0
Jan 1 12:00:02 (none) daemon.warn mountd[606]: is specfied in /etc/exports file.
Jan 1 12:00:02 (none) daemon.notice mountd[606]: Version 1.3.2 starting
Jan 1 12:00:02 (none) daemon.info udhcpc[604]: Sending discover...
Jan 1 12:00:05 (none) daemon.info udhcpc[604]: Sending discover...
Jan 1 12:00:08 (none) daemon.info udhcpc[604]: Sending discover...
Jan 1 12:00:09 (none) daemon.info udhcpc[604]: Sending select for 192.168.1.69...
Jan 1 12:00:09 (none) daemon.info udhcpc[604]: Lease of 192.168.1.69 obtained, lease time 864000
Jan 1 12:00:09 (none) local0.notice root: DHCP bound ('192.168.1.69' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '192.168.1.1') replacing ''
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 5 slot 4 freq 1748
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 6 slot 5 freq 1864
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 7 slot 6 freq 1980
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: New UDP socket 4 bound to 0.0.0.0:1900
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: setting multicast for 239.255.255.250
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: New UDP socket 5 bound to 239.255.255.250:1900
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: sockets_add: handle 4 (type 0) returning socket index 0 [0.0.0.0:0]
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: sockets_add: handle 5 (type 0) returning socket index 1 [0.0.0.0:0]
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: sockets_add: handle 6 (type 2) returning socket index 2 [0.0.0.0:0]
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: sockets_add: handle 7 (type 2) returning socket index 3 [0.0.0.0:0]
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: starting init_hw 0
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: trying to open [0] adapter 0 and frontend 0
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x4361a0
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: opened DVB adapter 0 fe:9 dvr:10
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: deleting pids on adapter 0, sid -1, pids=NULL
Jan 1 12:00:09 (none) daemon.notice minisatip[1307]: Delivery System DVB-S/DVB-S2 (AXE)
[... 313 lines cut ...]
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: clearing filters on PID 5100 ADAPTER 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: clearing filters on PID 5101 ADAPTER 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: clearing filters on PID 5102 ADAPTER 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: clearing filters on PID 5103 ADAPTER 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: clearing filters on PID 5106 ADAPTER 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: closing adapter 0 -> fe:10 dvr:11
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: deleting pids on adapter 0, sid -1, pids=NULL
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: AXE standby
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: sockets_del: 5 -> handle 11, sid 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: closing DVR socket -1 pos 5 aid 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: sockets_del: 5 Last open socket is at index 8 current_handle 11
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: done closing adapter 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: closed stream 0
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: minisatip.c:506 get_sid returns NULL for s_id = -1
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: reply -> 9 (10.8.0.14:55079) CL:0 : RTSP/1.0 200 OK^M Date: Wed, Dec 16 08:28:49 2015 GMT^M Cseq: 6^M Session: 1804289383^M ^M
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: select_and_execute[4]: Close on socket 9 (sid:-1) from 10.8.0.14:55079 - type rtsp failure 0:Success
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: sockets_del: 4 -> handle 9, sid -1
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: minisatip.c:866 get_sid returns NULL for s_id = -1
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: Closing stream -1
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: sockets_del: 4 Last open socket is at index 8 current_handle 9
Dec 16 08:28:49 (none) daemon.notice minisatip[1304]: Delete socket 4 done: sid -1
~ #
[/spoiler]

What did i miss?

perexg, 2015-12-16:
This is wrong setup:

<pre>
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 5 slot 4 freq 1748
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 6 slot 5 freq 1864
Jan 1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 7 slot 6 freq 1980
</pre>

The correct lines should be:

<pre>
[16/12 09:26:07.000]: Setting unicable adapter 0 slot 0 freq 1748
[16/12 09:26:07.000]: Setting unicable adapter 1 slot 1 freq 1864
[16/12 09:26:07.001]: Setting unicable adapter 2 slot 2 freq 1980
[16/12 09:26:07.001]: Setting unicable adapter 3 slot 3 freq 2096
</pre>

Could you try this from cmdline?

<pre>
touch /tmp/nosatip
minisatip -ll -d 38 -f -u 0:0-1748,1:1-1864,2:2-1980,3:3-2096
</pre>

crazy888, 2015-12-16:
Ok here you are:

login as: root
root@192.168.1.69's password:

Welcome to SATIP-AXE server 201508061051-10

~ # logread
Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.23.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 12:00:01 (none) daemon.info init: starting pid 604, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 605, tty '': '/etc/init.d/ntpd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 606, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 607, tty '': '/etc/init.d/satip'
Jan  1 12:00:01 (none) daemon.info init: starting pid 608, tty '': '/etc/init.d/telnetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 609, tty '': '/etc/init.d/dropbear'
Jan  1 12:00:01 (none) daemon.info init: starting pid 610, tty '': '/etc/init.d/inetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 612, tty '': '/etc/init.d/crond'
Jan  1 12:00:01 (none) daemon.info init: starting pid 613, tty '': '/etc/init.d/oscamd'
Jan  1 12:00:01 (none) daemon.info udhcpc[604]: udhcpc (v1.23.2) started
Jan  1 12:00:01 (none) daemon.info init: starting pid 617, tty '': '/etc/init.d/customd'
Jan  1 12:00:02 (none) local0.notice root: DHCP deconfig
Jan  1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_rsa_host_key
Jan  1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan  1 12:00:02 (none) cron.info crond[612]: crond (busybox 1.23.2) started, log level 8
Jan  1 12:00:02 (none) authpriv.info dropbear[609]: Not backgrounding
Jan  1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan  1 12:00:02 (none) daemon.warn mountd[606]: Kernel does not have pseudo root support.
Jan  1 12:00:02 (none) daemon.warn mountd[606]: NFS v4 mounts will be disabled unless fsid=0
Jan  1 12:00:02 (none) daemon.warn mountd[606]: is specfied in /etc/exports file.
Jan  1 12:00:02 (none) daemon.notice mountd[606]: Version 1.3.2 starting
Jan  1 12:00:02 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:05 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:08 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:09 (none) daemon.info udhcpc[604]: Sending select for 192.168.1.69...
Jan  1 12:00:09 (none) daemon.info udhcpc[604]: Lease of 192.168.1.69 obtained, lease time 864000
Jan  1 12:00:09 (none) local0.notice root: DHCP bound ('192.168.1.69' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '192.168.1.1') replacing ''
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 0 slot 0 freq 1748
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 1 slot 1 freq 1864
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 2 slot 2 freq 1980
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 3 slot 3 freq 2096
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: New UDP socket 4 bound to 0.0.0.0:1900
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: setting multicast for 239.255.255.250
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: New UDP socket 5 bound to 239.255.255.250:1900
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 4 (type 0) returning socket index 0 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 5 (type 0) returning socket index 1 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 6 (type 2) returning socket index 2 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 7 (type 2) returning socket index 3 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: starting init_hw 0
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: trying to open [0] adapter 0 and frontend 0
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x4361a0
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: opened DVB adapter 0 fe:9 dvr:10
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: deleting pids on adapter 0, sid -1, pids=NULL
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: Delivery System DVB-S/DVB-S2 (AXE)
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 10 (type 5) returning socket index 4 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: done opening adapter 0 fe_sys 6
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: trying to open [1] adapter 1 and frontend 0
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x43a9a8
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: opened DVB adapter 1 fe:11 dvr:12
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: deleting pids on adapter 1, sid -1, pids=NULL
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: Delivery System DVB-S/DVB-S2 (AXE)
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: sockets_add: handle 12 (type 5) returning socket index 5 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1306]: done opening adapter 1 fe_sys 6
[... cut ...]
Dec 16 09:42:23 (none) daemon.notice minisatip[1306]: minisatip.c:866 get_sid returns NULL for s_id = -1
Dec 16 09:42:23 (none) daemon.notice minisatip[1306]: Closing stream -1
Dec 16 09:42:23 (none) daemon.notice minisatip[1306]: sockets_del: 4 Last open socket is at index 8 current_handle 9
Dec 16 09:42:23 (none) daemon.notice minisatip[1306]: Delete socket 4 done: sid -1
~ #

minisatip running with:
 1306   607 root     S N   1772  0.4   0  0.1 minisatip -f -g -ll -d 38 -f -u 0:0-1748,1:1-1864,2:2-1980,3:3-2096

perexg, 2015-12-16:
I'm not sure why, but VLC closes the connection almost immediately in this case. Try HTTP streaming or another SAT>IP client..

crazy888, 2015-12-16:
Using http stream bring up following status.txt

StatusVersion=1
Subscriptions_1=1
DiseqC_1=1
Freq_1=11493000
Pol_1=h
Pids_1=0,5100,5101,5102,5103,5106
Stream_1_1=0.0.0.0:0
Streaming_1_1=4267700
LastRTSP_1_1=86038
TimeoutRTSP_1_1=30000

I also can see that LastRTSP is counting up. It looks like the box is streaming but there is no video in VLC.
When pushing the stop button in vlc the stream is also stopped at the box status is back to normal. 

I tried this an different machines and also on the elegato Sat>ip android app.
(Please keep in mind that the stock firmware does stream it very well)

login as: root
root@192.168.1.69's password:

Welcome to SATIP-AXE server 201508061051-10

~ # logread
Jan  1 12:00:00 (none) syslog.info syslogd started: BusyBox v1.23.2
Jan  1 12:00:00 (none) local0.notice root: AXE drivers load start
Jan  1 12:00:01 (none) local0.notice root: AXE drivers load finish
Jan  1 12:00:01 (none) daemon.info init: starting pid 604, tty '': '/etc/init.d/network'
Jan  1 12:00:01 (none) daemon.info init: starting pid 605, tty '': '/etc/init.d/ntpd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 606, tty '': '/etc/init.d/nfsd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 607, tty '': '/etc/init.d/satip'
Jan  1 12:00:01 (none) daemon.info init: starting pid 608, tty '': '/etc/init.d/telnetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 609, tty '': '/etc/init.d/dropbear'
Jan  1 12:00:01 (none) daemon.info init: starting pid 610, tty '': '/etc/init.d/inetd'
Jan  1 12:00:01 (none) daemon.info init: starting pid 611, tty '': '/etc/init.d/crond'
Jan  1 12:00:01 (none) daemon.info init: starting pid 613, tty '': '/etc/init.d/oscamd'
Jan  1 12:00:01 (none) daemon.info udhcpc[604]: udhcpc (v1.23.2) started
Jan  1 12:00:01 (none) daemon.info init: starting pid 615, tty '': '/etc/init.d/customd'
Jan  1 12:00:02 (none) local0.notice root: DHCP deconfig
Jan  1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_rsa_host_key
Jan  1 12:00:02 (none) authpriv.warn dropbear[609]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Jan  1 12:00:02 (none) authpriv.info dropbear[609]: Not backgrounding
Jan  1 12:00:02 (none) cron.info crond[611]: crond (busybox 1.23.2) started, log level 8
Jan  1 12:00:02 (none) local0.notice root: Setting STV6210 gain to 8
Jan  1 12:00:02 (none) daemon.warn mountd[606]: Kernel does not have pseudo root support.
Jan  1 12:00:02 (none) daemon.warn mountd[606]: NFS v4 mounts will be disabled unless fsid=0
Jan  1 12:00:02 (none) daemon.warn mountd[606]: is specfied in /etc/exports file.
Jan  1 12:00:02 (none) daemon.notice mountd[606]: Version 1.3.2 starting
Jan  1 12:00:02 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:05 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:08 (none) daemon.info udhcpc[604]: Sending discover...
Jan  1 12:00:09 (none) daemon.info udhcpc[604]: Sending select for 192.168.1.69...
Jan  1 12:00:09 (none) daemon.info udhcpc[604]: Lease of 192.168.1.69 obtained, lease time 864000
Jan  1 12:00:09 (none) local0.notice root: DHCP bound ('192.168.1.69' 'broadcast 192.168.1.255' 'netmask 255.255.255.0' | '192.168.1.1' '192.168.1.1') replacing ''
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 0 slot 0 freq 1748
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 1 slot 1 freq 1864
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 2 slot 2 freq 1980
Jan  1 12:00:09 (none) user.notice minisatip: Setting unicable adapter 3 slot 3 freq 2096
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: Starting minisatip version 0.1.84-axe1, compiled with AXE DVB API, DVR buffer 18424 bytes
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: New UDP socket 4 bound to 0.0.0.0:1900
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: setting multicast for 239.255.255.250
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: New UDP socket 5 bound to 239.255.255.250:1900
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: sockets_add: handle 4 (type 0) returning socket index 0 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: sockets_add: handle 5 (type 0) returning socket index 1 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: sockets_add: handle 6 (type 2) returning socket index 2 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: sockets_add: handle 7 (type 2) returning socket index 3 [0.0.0.0:0]
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: starting init_hw 0
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: trying to open [0] adapter 0 and frontend 0
Jan  1 12:00:09 (none) daemon.notice minisatip[1288]: adapter.c:172 allocation_wrapper malloc allocated 18434 bytes at 0x4361a0
[... 120 lines cut ...]
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: get free adapter -1 - a[0] => e:1 m:-1 sid_cnt:0 f:0 pol=0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: Got adapter 0 on socket 4
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: set adapter 0 for stream 0 m:0 s:1
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: setting DVB parameters for adapter 0 - master_sid 0 sid 0 old f:0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: deleting pids on adapter 0, sid -1, pids=NULL
[... cut ...]
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: setting filter on PID 5101 for ADAPTER 0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: setting filter on PID 5102 for ADAPTER 0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: setting filter on PID 5103 for ADAPTER 0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: setting filter on PID 5106 for ADAPTER 0
Dec 16 10:05:58 (none) daemon.notice minisatip[1288]: reply -> 9 (10.8.0.14:51700) CL:0 : HTTP/1.0 200 OK^M Date: Wed, Dec 16 10:05:58 2015 GMT^M ^M ^M
~ #

perexg, 2015-12-16:
Could you save the HTTP stream using wget or curl ?

I don't have unicable, but some users reported that it should work.

crazy888, 2015-12-16:
trying to save with wget for windows not possible.

See http://pastebin.com/GEZ9NJCF :)

perexg, 2015-12-16:
Put the whole http url to "" .

crazy888, 2015-12-16:
http://pastebin.com/n44CKw6v

It is saving but without any bandwidth. The file size stays at 1kb.

Btw: Yes unicalbe setup has been confirmed as working. So it should normaly work.
As the stock firmware is streaming over the same unicalbe setup as minisatip should do, i think the unicable hardware is working.

crazy888, 2015-12-17:
Feels like being a never sloved problem?
Anything else you need to help me? Logs?
Anything else i should do?

I realy want that box to work for me .............

I spend a lot of hours with testing now.
Always the same........... Stock firmware is streaming absolutly fine........ Minisatip doesn´t.............

CvH, 2015-12-17:
fyi
it looks like unicable EN50494 and EN50607/JESS is not support at the minisatip version that is included at this fw. I suspect the support was added late April 2015 (this version is from Mar).

crazy888, 2015-12-17:
seriously ??? 

So it´s just that simple ???

crazy888, 2015-12-18:
Ok, it´s working with normal lnb input.
I just connected my multiswitch and streaming is running.

So it´s realy a problem with that unicable .........
Is it possible to update minisatip?

CvH, 2015-12-18:
https://github.com/perexg/satip-axe/blob/master/patches/minisatip-axe.patch
needs to be adjusted to the new version :) otherwise no problem!

crazy888, 2015-12-19:
Hopefully perexg will find some time do make a new release.

perexg, 2015-12-19:
I don't see any unicable related changes in the latest minisatip except for PIN in jess setup. Does your LNB require PIN authentication ?

CvH, 2015-12-19:
referring to https://github.com/catalinii/minisatip/issues/38
the commit messages aren't helping much to figure out what is added or not :)

perexg, 2015-12-19:
Maybe it's missing mdelay(15 or 25) after 13V setup which I added already to the minisatip in the satip-axe firmware..

crazy888, 2015-12-29:
Anything else I can do?

mcasselo, 2016-01-27:
Had the same issue, stock firmware was streaming fine, minisatip was not working. So I played around and finally found the solution: 

I'm using Technisat Technirouter 5 / 1x8 with two clients, the first 4 frequencies are used by first client, second 4 frequencies used by Digibit R1. So i had to change this line in config:

MINISATIP_OPTS="-X 0 -u 0:4-1748,1:5-1864,2:6-1980,3:7-2096"

This works fine for me. It tooks a very long time to find out this soulution. The second number after ":" has to be the same number as counted the frequency at the Technirouter.

perexg, 2016-01-27:
Guys, the minisatip help explains that:

<pre>
		-u unicable_string: defines the unicable adapters (A) and their slot (S) and frequency (F):
			The format is: A1:S1-F1[,A2:S2-F2[,...]] 
			When * character is used before frequency, force 13V for setup
</pre>

The SLOT is SLOT for your unicable equipment (you should consult manual to obtain the right value).

closed 2016-02-12 by perexg