perexg/satip-axe #41: Network Issue Inverto Box

Quarcs, 2016-01-05:
Hello,

I got some problems with my Inverto Box:
I boot up from an usb drive. I have not flashed the device. 
The LAN Port is not stable. Dmesg shows me, that the port goes up and down

PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 1000/Full

That is why I don't get a stable connection to it. I lose a lot of packages at a ping request. Do you have an idea how to fix the issue? 

Thanks a lot!

perexg, 2016-01-05:
Wrong gigabit connection ? You may try to force 1000M with ethtool (which is in the satip-axe firmware), but it would be better to resolve the culprit.

pbriesch, 2016-01-05:
It could help to force the connection to 100 M on the switch if it is a manageable switch.

Quarcs, 2016-01-05:
@perexg 
Tried:
ethtool -s eth0 speed 1000 duplex full

Config now:

ethtool eth0
Settings for eth0:
	Supported ports: [ TP AUI BNC MII FIBRE ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  1000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 1
	Transceiver: external
	Auto-negotiation: on
	Supports Wake-on: ug
	Wake-on: g
	Current message level: 0x0000003f (63)
			       drv probe link timer ifdown ifup
	Link detected: yes

==> Still not stable
==> Will try another TP-Cable (Cat 5e)

@pbriesch: It is an unmanaged switch

9000h, 2016-01-05:
@Quarcs if you have it on a FritzBox try to disable the Green/ECO mode for this port

pbriesch, 2016-01-05:
Try to force **100** M with ethtool.

Quarcs, 2016-01-05:
@9000h Good idea, I got a FritzBox, but Green/ECO mode is already disabled :(

@pbriesch I will try :)

dreamcat4, 2016-01-05:
Or bad gigabit cable. Which can work for some device and not others @ 1000 MBits speed.

Quarcs, 2016-01-05:
Changed cable from cat5e to cat6. No improvement.

@pbriesch have forced to 100M. It is working at the moment. Hope it will stay like this... Thanks a lot :-) 

Will let this ticket open for a while and will close it soon if there are no issues for a while

Quarcs, 2016-01-05:
same problems  again. Has only helped a few minutes :(

Quarcs, 2016-01-05:
That's my full dmesg. Perhaps it helps...

Linux version 2.6.32.42_stm24_0208-idl4k_7108 (perex@stlinux.perex-int.cz) (gcc version 4.5.3 20111123 (STMicroelectronics/Linux Base 4.5.3-99) (GCC) ) #43 PREEMPT Thu Aug 6 10:51:35 CEST 2015
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
On node 0 totalpages: 126976
free_area_init_node: node 0, pgdat 81459dbc, node_mem_map 8291e000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 125984 pages, LIFO batch:31
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line:  console= hw=16001000000000000000000000000000 sn=1S81C11301000121 uuid=F9E168D1428C4048B48509BADE8FC7C0 revBoot=stm24_0056-B0_80-idl4k-rev47 console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72d02000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 381184k/507904k available (3365k kernel code, 1091k data, 20132k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[STM]: [PM]: HoM support registered
[STM]: [PM]: [HoM]: Early console @ 0xfd733000
Installing knfsd (copyright (C) 1996 okir@monad.swb.de).
[... 253 lines cut ...]
PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 1000/Full
PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 100/Full
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0

perexg, 2016-02-12:
Could you try the new satip-axe firmware with the updated ethernet driver?

http://s000.tinyupload.com/?file_id=78443875428972064634

List of changes:

https://github.com/perexg/satip-axe/blob/master/dist/README (upcoming)

Please, give me feedback.

Quarcs, 2016-02-20:
It was also not working well with the original firmware. So I send it back to the manufacturer. I will try it, when I get the device back. I will write you directly. Thanks for your help @perexg  :-)

perexg, 2016-02-29:
Closing. Re-open if you have issues with new box.

closed 2016-02-29 by perexg

fedecarnales, 2016-11-27:
I'm having the same problem, the network keeps disconnecting. Dmesg shows repeated instances of:

PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 1000/Full

I tried forcing it to 100m with ethtool but I lose the connection altogether until I reboot, but after rebooting it's back to 1000m.

I also have connection problems with the original firmware. Might be a faulty unit but I can't send it back. Is there any way to get a USB Ethernet adapter to work so as to bypass the internal ethernet chip? I tried an adapter I have (with Realtek RTL8153 chipset) and it didn't seem to recognize it.

Thanks!

ant-thomas, 2017-12-17:
Sorry to reopen this, I seem to be suffering from the same issue.
I've been using my Digibit R1 without issue on satip-axe for over 2 years, up until about a month ago and it keeps suffering from the issue
```PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 1000/Full```

Could the box have become faulty?
I'm using the same cable as from day 1.

ant-thomas, 2017-12-17:
It could well have just been the cable, which is odd since it hasn't been moved for 2 years!
Hopefully the change has fixed it.

SerialVelocity, 2019-06-09:
I'm also hitting the same issue with a bunch of link down/up in my logs:
```
PHY: 1:01 - Link is Down
PHY: 1:01 - Link is Up - 1000/Full
```

@ant-thomas @fedecarnales Is your device behind a switch? If I bypass the switch I don't get this issue.

The latest official Digisat R1 firmware (V1.24.0.156) doesn't seem to have this problem but previous versions did.