perexg/satip-axe #42: Network access problem

aleceuta, 2016-01-21:
Hi,

I have a problem with satip-axe firmware on my TELESTAR DIGIBIT R1, I use a pendrive with last version release satip-axe-201508061051-10, my problem is when I try change for static ip, edit the file /etc/sysconfig/config and change the relative part of networks, change, the line :

IPV4="static"

and uncomment

IPV4="192.168.3.210"
IPV4_NETMASK="255.255.255.0"
IPV4_BROADCAST="192.168.3.255"
IPV4_GATEWAY="192.168.3.50" 

save the file and reboot Telestar

I test with a ping and work.

But when access by web say an 404 error
and in the networks place on my pc does not appear.

but really the problem because after I came back to try DHCP configuration for try

by mistake put the line in

IPV4="DHCP" 

DHCP in upper letter

and save file and reboot

since then it does not respond to ping and nothing (ssh, telnet, etc..)

I try idl4k.cfgresetusb rename to idl4k.scr

With the original firmware (TELESTAR) works fine on dhcp and static.

Help me please.

aleceuta, 2016-01-26:
some method to restore the file /etc/sysconfig/config

aleceuta, 2016-02-01:
No one can tell me what I can do?

perexg, 2016-02-01:
Have you really tried idl4k.cfgresetusb to idl4k.scr and boot the satip-axe firmware again ? Note that you must have satip-axe-201508061051-10.fw file on stick, too!

aleceuta, 2016-02-01:
yes, yes I put the two files idl4k.cfgresetusb rename to idl4k.scr and satip-axe-201508061051-10.fw, any question the file /etc/sysconfig/config is the flash of the telestar?

perexg, 2016-02-01:
Yes, it file in the internal flash, but I tested the reset procedure several times, so I really don't know why you cannot enter it.... The box should be available through DHCP IP as after the first installation. You may check, if there are some DHCP requests when you boot the satip-axe firmware with idl4k.cfgresetusb u-boot configuration.

aleceuta, 2016-02-01:
when I try to reset with the file idl4k.cfgresetusb, after a few minutes the original firmware starts?, even when the stick punctured, that is normal

perexg, 2016-02-01:
no, the satip firmware should be started with the default configuration; something is wrong on your side; you should see the led effect (blinking from left to right and back)  - it means that something is downloaded from USB

note that some USB sticks does not work..

perexg, 2016-02-01:
also check the USB slot - it must be upper one..

aleceuta, 2016-02-01:
yes on the upper when put with file idl4k.cfgresetusb not Knight Rider effect

perexg, 2016-02-01:
show me the directory content for your usb stick, also, I think that the stick must have a parition....

perexg, 2016-02-01:
also, tell me, if you use the satip-axe-201508061051-10.usb as idl4k.scr , do you see the Knight Rider effect ?

aleceuta, 2016-02-01:
yes when use satip-axe-201508061051-10.usb as idl4k.scr. I see Knight Rider effect

perexg, 2016-02-01:
sorry, I see it... the cfgresetusb file is broken.. wait

perexg, 2016-02-01:
ok, download the idl4k.cfgresetusb and follow the section 'Recovery from the wrong configuration:' in https://github.com/perexg/satip-axe/blob/master/dist/README

aleceuta, 2016-02-01:
I see your new file, I put, but I don`t see Knight Rider effect

aleceuta, 2016-02-01:
is necesary special partition, I use fat32,

aleceuta, 2016-02-01:
Now I use the original firmware with static ip, for use  idl4k.cfgresetusb is necesary put dhcp on original firmware.

aleceuta, 2016-02-01:
[image]

perexg, 2016-02-02:
You don't follow updated README. The firmware file must be satip-axe.fw (I tried to make the recovery script universal for all 10+ firmwares (future)).

aleceuta, 2016-02-02:
Ok it works, solved, thanks.

closed 2016-02-02 by perexg