perexg/satip-axe #44: newer oscam needed

bylderup, 2016-02-06:
How can the Oscam version r 11037 be changed to a newer build ?
For mdvbapi 0.16 a newer version is needed - otherwise I get massive PMT Buffer overflow errors.
I use the alternative Firmware on my Digitbit R1 on a USB Stick.

perexg, 2016-02-11:
I'll update oscam in the next release to latest. You may also use any sh4 binary and run it through the /etc/sysconfig/localboot script.

perexg, 2016-02-12:
A beta with oscam 11211:

http://s000.tinyupload.com/?file_id=78443875428972064634

List of changes:

https://github.com/perexg/satip-axe/blob/master/dist/README (upcoming)

Please, give me feedback.

bylderup, 2016-02-12:
Thanks a lot!!
I have just installed it and will test and give feedback

bylderup, 2016-02-12:
ok.... tested today with following equipment :

Digibit R1 beta with new Oscam, Gigabit Network Switches with igmpv3,
Dvbviewer Recording Service 1.32 on Win10 Intel i5, mdvbapi 0.16.
Clients : Pi2 with Kodi, Notebook with Dvbviewer, FireTv with Kodi and PS3
via UpnP Media Server.
4 different HD Channels : Just fine and stable.

Dvbviewer Controller for Android causes Termination of all streams, is
running stable standalone only - but I think thats a problem of dvbviewer..

So far everything seems to run stable and quick.

Thanks for new firmware again !

2016-02-12 12:33 GMT+01:00 Michael Gerke <meikel.gerke@googlemail.com>:

bylderup, 2016-02-13:
@perexg 
Can you mark this as solved and remove my email adress in postings above, please.
I can´t edit the posting..

closed 2016-02-13 by bylderup