perexg/satip-axe #45: Build number 11

perexg, 2016-02-16:
I am tired to update links in multiple issues, so this is alpha/beta version of build 11.

<pre>
    - updated stmmac ethernet driver from latest stlinux kernel
      - linux-sh4-2.6.32.y - _stm24_0217
      - enabled Realtek PHY module (RTL8211E)
      - disable scatter-gather, increase ring descriptors (from 256 to 512)
    - updated dropbear to 2015.71
    - updated nfsutils to 1.3.3
    - updated nano to 2.4.3
    - updated oscam to 11211
    - minisatip accepts -M option to skip first MPEG-TS packets from demuxer
      - the default value is 35 (to skip packets from previous tune)
    - added second minisatip binary (minisatip5 - based on 0.5.26 version)
      - add new lines to /etc/sysconfig/config from /etc/config.default
</pre>

Latest build:

x23piracy, 2016-02-22:
Hi,

will oscam now work via stapi? can the box stream descambled streams?

Regards X23

clandmeter, 2016-02-29:
You are including minisatip5, but what is it used for?

perexg, 2016-02-29:
It's based on minisatip 0.5.30. An updated version which uses threads to control, better diseqc control (support for uncommitted switches, diseqc timing control) and many improvements (status page). But I'm maintaining the older 0.1.84 version, too.

perexg, 2016-02-29:
I created a short document which explains some newly implemented features in minisatip binaries in build 11:

https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md

bylderup, 2016-03-01:
I have two files now in mnt/data/satip-axe/packages : multicast-rtp-1.tar.gz and Python-3.5.1.tar.gz
unpacking these files, too ??  which tar params ? -xf or -xzf -C ?

perexg, 2016-03-01:
No, just reboot the box. The unpacking is automatic on boot (see /etc/init.d/rcS). The root filesystem is in ramdisk, so everything is lost on boot.

bylderup, 2016-03-01:
Latest beta is fine for free TV but I have problems with some paytv channels

m4tt075, 2016-03-05:
A note on the upgrade process:
I downloaded the 11.fw file to /mnt/data and ran "upgrade-fw ./11.fw" from there. The first two flash attempts failed with errors, the third one seemed to work. However, after entering "reboot" following the on-screen instructions, the network connection did not come up again. Powered the box off and on again, no change, network would not come up. Had to flash the firmware via USB (following the dist instructions) to make it work again. It's up and running now. Will test and report back. But just in case, anyone else runs into similar issues...

perexg, 2016-03-05:
@m4tt075 : If you like, create a new issue for this. If the verification succeed, it should work.

perexg, 2016-03-05:
The release 201603051934-11 is available at https://github.com/perexg/satip-axe/tree/master/dist .

* another fixes in minisatip/minisatip5
* added fetch-package script
* upgrade-fw can be run only from / or /root directory

closed 2016-03-05 by perexg

m4tt075, 2016-03-05:
@perexg: Thanks. Not needed anymore. Apparently I ran upgrade-fw from the wrong directory. Just updated again to the release version from /root and no problems whatsoever. Thanks again for all your amazing work on satip-axe as well as TVH!