perexg/satip-axe #46: Errors on 19.2E freq 12721750 H

clandmeter, 2016-02-17:
I reported this already on IRC, but for reference I am adding it here as well.

According to @perexg the errors happen in the hardware path. @perexg has reported this to Inverto, but until now no reply yet.

perexg, 2016-02-17:
I tried again and nothing changed. I'm trying to ping Inverto the second time.

perexg, 2016-02-18:
I have a reply. Inverto thinks that it might be related that this TP has no pilot. They can reproduce it, too.

clandmeter, 2016-02-18:
Nice, this means they will look into a solution or is it a transponder issue which cannot be resolved?

perexg, 2016-02-18:
They only asked me if I see this issue on tuners 3 & 4 (I do - I have 19.2E on third tuner). Appearently, they have continuity errors only on tuner 1 & 2. It looks like a demodular setup issue to me.

perexg, 2016-02-18:
Here is the explanation from Inverto software guys:

"""
A wrong information has been communicated. Actually , this is tuner 1 and 2 that might maybe receive properly the signal, and tuner 3 and 4 that are really on the edge.

The issue on tuner 3 and tuner 4 is likely a RF performance issue. We detected that the performance is slightly lower on tuners 3 and 4 ( because of proximity with other components), however, they are still in the product specification.
So you don't have that much to expect from a miraculous driver fix. because it is not a software problem in the demodulator.

The transponder 12722 H is owned by M7, and is intended for Belgium. 

The characteristic of this transponder is that the pilots are not present. the Pilots is an information that might be added by the broadcaster, in order to increase the robustness of reception (against rain, clouds, slightly badly aimed dish, and finally area of reception, or use of smaller dish size). Depending of the country, the operator might decide to favor robustness or favor bandwidth.  

In this particular case, M7 obviously favors the bandwidth, and also counts on the fact that signal is received from Belgium and surroundings, so from a quite small area which is quite in the focal of the satellite. 

As a consequence, M7 made best use of the transponder, and have taken the risk that signal might be improperly received from other area. 
This is a choice to respect , and there is nothing to do. 

Probably a single tuner decoder might receive properly the signal without macroblocks. 
In our IDL400s , there is an additionnal matrix to route the tuner inputs. It introduces a slight degradation, which is enough to cause some non recoverable packets. You really reach the edge case, where one decoder might work, and a second is not. 

We will try to see if we can optimize the RF performance, but there is unfortunately not to expect. 
"""

clandmeter, 2016-02-23:
I just switched the tuner from nr3 to nr1, and I don't see artifacts now.
So it does look like what Inverto is saying is (somewhat) right.

perexg, 2016-02-23:
I have an updated driver which should fix the "weak" reception issue for tuners 3&4 from Inverto. I'll release a firmware with this fix soon.

perexg, 2016-02-23:
Could you test the latest firmware from issue #45 ?

clandmeter, 2016-02-24:
copying firmware to /mnt/data and exec "upgrade-fw fw" should be ok?

clandmeter, 2016-02-24:
As suggested on IRC, I cd to root and then exec upgrade-fw /mnt/data/fw (had to exec it 7x to get a working upgrade)
Seems the errors have gone, nice job!

Many thanks for this effort!

perexg, 2016-02-24:
Here is the culprit of this 'bug' (quoting the Inverto answer):

"""
The system is using 2 demods, 2 tuners, and 4 VGLNA in a fully symmetric way. Therefore they should be absolutely identical.

However, there is one difference , which is that one system is using a crystal (30MHz) as source, while the second block system uses a loopback of the clock from the first block.

The characteristics of the clock that is different among the 2 systems is enough to cause a slight degradation on the quality of digitization for the second block.

We modified the clock settings (unbuffered, input as output). We are unsure whether this works on your hardware ( because signal might be out of spec, because too much attenuated after 4 cascacdes), but if this works, then in theory you should observe quite close performance among the 4 demodulators blocks, because they will have all the same clock as reference. And hopefully the few discontinuties should be gone.
"""

perexg, 2016-02-29:
Closing. My tests passed and  in issue #45 is rc1 of build11.

closed 2016-02-29 by perexg