perexg/satip-axe #47: Unicable config?

paluseb, 2016-02-17:
Hi.
I have the following problem. I connected just one cable to LNB 1. In the configuration Telestar R1 have set unicable LNB. When using satip 1 and 2 all is well but if I use satip 3 and 4, the image is not acceptable. TvH in the status shows continuity errors. Could this be the wrong configuration Telestar, damage to equipment or something else? I'm using no original power supply. Output current 2,9A instead 3,0A , maybe this is the cause. Do you have any idea?
TvH working on RPi 2. RPi2 and computers are clients TvH. 
I had this configuration:
MINISATIP_OPTS = "- u 0:1-1420,1: 0-1210,2: 2-1680,3: 3-2040"
This is probably not correct?
I should have probably:
MINISATIP_OPTS = "- u 0:1-1420,0: 0-1210,0: 2-1680,0: 3-2040"
MINISATIP_OPTS = "- X 0"
or
MINISATIP_OPTS = "- X 0, u 0: 1-1420,1: 0-1210,2: 2-1680,3: 3-2040"
I do not know, ask for a hint.

Additionally, I had to use Recovery from the wrong configuration, and now all the time creates me config.old. I leave it or remove?

perexg, 2016-02-17:
Recovery - return back the idl4k.scr for the standard firmware. The recovery script should be booted only once otherwise the config will be renamed all times after boot.

The three values for unicable means:

ADAPTER:SLOT-FREQUENCY

ADAPTER=INPUT (on back) (0-3)

The SLOT:FREQUENCY values should be taken from your unicable equipment manual.

perexg, 2016-02-17:
Also see issue #39 for another unicable config .

paluseb, 2016-02-17:
if I understood it after the recovery script I should remove it and add the normal idl4k.scr? Can I replace the files on the version 11 beta?

perexg, 2016-02-17:
Yes for both questions.

paluseb, 2016-02-19:
It looks at the problems of antenna installation. When I connected the 4 cables and changed the settings Unicable TVH shows the signal for all receivers. But still, the image is distorted and additionally tuner connected to the output of legacy LNB loses signal. What should be the configuration for the converter quad?
MINISATIP_OPTS="-Q"

paluseb, 2016-02-20:
There is a new original firmware (1.24.0.156). Many news. Among other things, channel search with preview signal strength and quality, and added support for Unicable II. Maybe it will improve cooperation with Unicable installations and certainly make it easier to diagnose errors of signal - LNB settings. I have to check it out.

CvH, 2016-02-20:
https://www.telestar.de/assetsFS/downloads/assets/medias/common/78/$v4/soft_Digibit_R1_005.zip

paluseb, 2016-02-20:
You are fast ;)

perexg, 2016-02-20:
> What should be the configuration for the converter quad?
> MINISATIP_OPTS="-Q"

No, quad is supported by default (no specific arguments to minisatip). -Q is for quattro (multiswitch LNB which has non-switchable outputs).

perexg, 2016-02-20:
BTW: Unicable II is supported also with minisatip (it's called jess).

perexg, 2016-02-20:
I just checked if Inverto updated drivers in 1.24.0.156 and they're identical like latest satip-axe (except the demux-ts module which just copies data from hw to the application - I don't think that it can improve anything).

You may easy check if the data are wrong when they're received from the demodulator:

<pre>
axe-debug pti vdev 1
</pre>

Replace number (1) with 0-3 which means input. Note that these numbers are swapped in this way:

0 <-> 1 and 2 <-> 3, so number 1 means tuner 0 etc..

There is PktC column (packet counter) and CCErr column (continous error counter). If the continuous error counter increases, something is wrong with your signal quality. It may be too strong (we have some reports that the used tuners have limited reception range.

perexg, 2016-02-23:
Note that Inverto confirmed that the reception on tuners 3&4 might be problematic. I received an updated driver which should fix this issue from Inverto. I'll release new satip-axe firmware with this driver soon.

paluseb, 2016-02-23:
@perexg thanks. Temporarily plugged the quad LNB and during trials with customers on windows such as DVB viewer all four tuners work well. In TVH still I think there is a problem. Of course I am interested in starting lnb unicable. Sorry my english ;)

perexg, 2016-02-23:
Then it looks like a bad network config or something else is wrong. We use TVH with all four tuners without issues you've described. Anyway, I posted new firmware with the updated driver in issue #45 .

paluseb, 2016-02-23:
It seems that TVH is also OK. I need to check with lnb unikable but in a few days. Thanks again. Topic does not close, if possible?

perexg, 2016-05-14:
No more questions, closing.

closed 2016-05-14 by perexg