perexg/satip-axe #48: usb dngle support

rucati, 2016-02-20:
Can USB dongle support (i.e. popular rtl2832 DVBT) be added? 
Or some howto how this could be achieved? I use minisatip now on x86 as a DVB-s+t+c streamer
Thanks!
rucati

perexg, 2016-02-20:
Some support for external tuners might be added, but:

- the patched minisatip must take in account the standard linudvb devices (actually only proprietary DVB interface is used)
- I heard (cannot confirm) that USB slots are not properly electrically isolated - it means that they can be damaged under some circumstances, so be careful to connect anything except the USB sticks there
- compiling the latest linux media drivers on the 2.6.32 kernel might be very tricky

rucati, 2016-03-02:
Anyone willing to put some effort in a generic HOWTO on support for usb dongles?