perexg/satip-axe #50: Recommended TVHeadend settings

Imerwin, 2016-03-07:
Hi,
I have a question about tvheadend with satip-axe.
What are the recommended tv adapters settings for tvheadend?

@perexg
My satip box is running more than 2 weeks without a crash with the version 11 Beta.
Thank you for your quick support and fast bugfix :-)

perexg, 2016-03-07:
Keep everything as default, just enable 'RTSP/TCP (embedded data)' in the device (root) panel.

Imerwin, 2016-03-07:
OK!
Thank you

closed 2016-03-07 by Imerwin

m4tt075, 2016-03-08:
@perexg I'm running a Digibit R1, I think(!) just like you. As I have updated and reconfigured my backend many, many times, would you mind posting a screenshot of your TV adapters -> minisatip parameters from the WebUI, please?

Imerwin, 2016-03-09:
Hi,
here are the default settings from the tvheadend TV-adapters:
http://imerwin.com/tmp/satip/satip_tv-adapters_config_original1.PNG
http://imerwin.com/tmp/satip/satip_tv-adapters_config_original2.PNG

I have installed a new temp VM with tvheadend to view the default settings.

perexg, 2016-03-09:
Yes, these defaults are good, except I strongly recommend to avoid using the UDP transport mode - just enable 'RTSP/TCP (embedded data)' in TVH 4.1 . Basically, UDP packets might be dropped by kernel (receiver) / any active ethernet component in the path between PC with TVH and the SAT>IP box (continuity errors). And yes, this issue exists even for the full gigabit connection.