perexg/satip-axe #51: Issue with original firmware 1.24.0.156

HellGL, 2016-03-11:
Hi,
excuse me that I post this question here as it is not an issue with the satip-axe firmware but with the original firmware version 1.24.0.156:
I am using vtuner/satip daemon to emulate a local dvb device on my system. Since I upgraded my GSS.Box to the new firmware I am not able to tune any channel with vtuner/satip anymore. In the log I can see that the OPTIONS command is answered with "200 OK" a subsequent SETUP request is first answered with "500 Internal Server Error". After some reconnects the server answeres with "453 Not Enought Resources". Does anybody has the same issues using the new firmware?
I also tried an http request with exactly the same data (plus some useful pids) and I get valid data. 

Thanks HellG.

PS I attached a debug log of the satip daemon with the relevant parts.

```
[10286 satip_vtuner.c:416] debug: MSG_SET_FRONTEND freq: 114957 symrate: 22000000 
[10286 satip_vtuner.c:823] debug: ioctl: response ok
[10286 satip_rtsp.c:148] debug: timeout
[10286 satip_rtsp.c:700] debug: connecting...
[10286 satip_rtsp.c:588] debug: connected -> establishing
[10286 satip_rtsp.c:309] debug: >>txbuf:
OPTIONS rtsp://192.168.168.37:554/ RTSP/1.0
CSeq: 1

<<
[10286 satip_rtsp.c:218] debug: rxbuf:
RTSP/1.0 200 OK
CSeq: 1
Date: Fri, 11 Mar 2016 12:35:54 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY
                             

<<
[10286 satip_rtsp.c:606] debug: response complete
[10286 satip_rtsp.c:391] debug: >>txbuf:
SETUP rtsp://192.168.168.37/?src=1&freq=11495.7&pol=h&msys=dvbs2&sr=22000&fec=23&ro=0.35&mtype=8psk&plts=on&pids=none RTSP/1.0
CSeq: 2
Transport: RTP/AVP;unicast;client_port=46181-46182

<<
[10286 satip_rtsp.c:218] debug: rxbuf:
RTSP/1.0 500 Internal Server Error
Date: Fri, 11 Mar 2016 12:35:54 GMT
CSeq: 2
Session: 155C396C

<<
[10286 satip_rtsp.c:600] debug: peer closed
[10286 satip_rtsp.c:137] debug: closing socket
```

perexg, 2016-03-15:
`
Transport: RTP/AVP;unicast;client_port=46181-46182
`

Your client is broken. The first port must be EVEN. See https://en.wikipedia.org/wiki/Real-time_Transport_Protocol#Sessions .

closed 2016-03-15 by perexg

HellGL, 2016-03-15:
Thanks, you are great!