perexg/satip-axe #52: satip-axe doesn't boot

qpirsel, 2016-03-15:
Hi!

I'd like to test this alternative firmware and followed the instructions but I can't get it working. My Digibit R1 has the current firmware V1.24.0.156 (28-01-2016 16:03) installed. I've already tested 3 different USB devices, tried the current alternative firmware and the predecessor and performed a reset on the device.

I always get the same behaviour. After powering it on the power LED illuminates, followed by a ~3s illuminating of all LEDs and then the devices boots up into the original firmware.

Thanks in advance for any advise.
qp

perexg, 2016-03-15:
If input/tuner LEDs do not blink from left-to-right and right-to-left (the knight rider effect), then the uboot (boot-loader) do not see your USB stick. Make sure that:

* it is FAT/VFAT formatted (not NTFS)
* has the partition ! (some USB sticks do not use partitioning)
* the USB stick must be in the upper USB slot

qpirsel, 2016-03-15:
Thank you for the exceptional fast response and advises.

I figured it out. I always tried to download the files with Chrome by "save link to..." and what I've downloaded was always the XML description. ^^ 

Sorry, I've never downloaded files from github before.
Shame on me!

closed 2016-03-17 by perexg