perexg/satip-axe #53: Skip first MPEG-TS packets does not work

prilly, 2016-03-17:
there seems to be some issues with the the skip buffer implemented.

after updating tvheadend and satip-axe to latest stable i have had that problem with buffer not cleaning out with the result that new invalid services are autoadded in tvheadend.

i use the MINISATIP_OPTS="-M 40" in startup parameter for the digibit r1 box and i have sett Skip TS packets (0-200): 40 in tvheadend

even though the skip is presented 2 places it seems that the buffer skip is to low or that some other issues is occuring.

i have tried with the -M 40 option only and the same with skip in tvh without no result.

perexg, 2016-03-17:
Which satellite position?

prilly, 2016-03-17:
> Dato:17.03.2016  12:48  (GMT+01:00)
> Ko: Prilly <prilly@speedbox.me>

1w cmore and 4.8e via

Both have the same behavior

-------- Opprinnelig melding --------

perexg, 2016-05-14:
@prilly : Could you try to use the maximum skip limit ?

prilly, 2016-05-19:
I have given it a new test, deactivated skip TS packets in TVheadend and activated the -M 200 option in the digibit r1 box.
it seems that is working as intended now. tvheadend does not add new services.
"perexg"  skrev 14. mai 2016 kl. 18.39:
	@prilly (https://github.com/prilly) : Could you try to use the maximum skip limit ?

perexg, 2016-05-31:
Tvheadend version v4.1-2097+ tries to be more reliable (it works good even when all these skip initial mpegts packet options are deactivated).

closed 2016-05-31 by perexg