perexg/satip-axe #54: IDL-400s not booting with release 11

PsychoRS, 2016-03-17:
Hi,

My Inverto IDL400s doesn't boot when I install release 11 in flash and change the configuration file.

Changing network address from DHCP to Static IP makes the box unable to boot.
Changing MINISATIP_OPTS from none to anything makes, again, the box unable to boot.

In these cases, I need to use the recovery from bad configuration file to get the box working again.

All of these works great I if use the "boot from USB" file. Installing in flash worked like a charm with release 10.

Got a friend of mine with a Digibit R1 and told me that he is suffering exactly the same... maybe a bug in the .flash file?

perexg, 2016-03-17:
If you install flash version through uboot, try to turn off box for few seconds when LEDs are not blinking. Remove the USB stick and turn on it..

If you have correct config file, the booting is identical for USB / Flash. I boot all my three boxes using flash without any issues. It must be something else :-)

PsychoRS, 2016-03-17:
Yes perexg that with I did always. On the first boot after the flashing (with default config file), the box is booting without any problem. When I ssh'd and changed ip address, minisatip options or simply deactivate oscam, the box will not boot again.

It's an strange issue, thinked that maybe my box was faulty... but some hours ago a friend told me that he is having exactly the same behaviour with a Digibit R1.

Is there any way of getting all flash spaces "oem" and start the process from scratch?

perexg, 2016-03-17:
The config files are shared for USB / flash versions of firmware. They're stored in the jffs2 partition. You should not boot to defaults if you modified the config files in USB firmware.

PsychoRS, 2016-03-17:
Yes, for booting to defaults I need to use the recovery from bad configuration file.

It's really strange, as I say we can only use a modified config if we use the boot from usb method. As I said, never have any troubles booting from flash with release 10.

perexg, 2016-03-17:
If you recover, you should see the backup of the configuration file in /etc/sysconfig . Could you upload it somewhere?

perexg, 2016-03-17:
<pre>
if test "$AXECFG" = "reset"; then
  mv /etc/sysconfig/config.old /etc/sysconfig/config.old1 2> /dev/null
  mv /etc/sysconfig/config /etc/sysconfig/config.old 2> /dev/null
  mv /etc/sysconfig/passwd.old /etc/sysconfig/passwd.old1 2> /dev/null
  mv /etc/sysconfig/passwd /etc/sysconfig/passwd.old 2> /dev/null
fi
</pre>

satip, 2016-03-18:
maybe i can help:
if you are using at static ip configuration, no dns servers are available (missing in config) - so ntpd-client is searching for "0.pool.ntp.org" and maybe this prohibit booting (and ntpd is using 100% cpu)
i have no boot-problems with enabled dhcp client... (firmware 11)

PsychoRS, 2016-03-18:
Damnit, never thought about that, and seems to make sense. Will try tomorrow, thank you all.

pemargo, 2016-05-12:
I have the same problem with the digibit R1, the first time I can boot the custom firm from usb without any problem, but after first reboot with the custom firm Digibit no up the network, i cant say if the only problem is the network, I cant access to sat-ip, no turn on the light of network, neither connect from ssh to device.
I all so tried install the firmware from flash with same results.

I have the same problem with the version 10 and version 11

Any help?? 

Regards

perexg, 2016-05-14:
Everything is described in README - .https://github.com/perexg/satip-axe/blob/master/dist/README . Unless you tried all recovery schemes, we cannot help. Also, try to set the correct DNS as noted in previous comments if you use static IP config.

closed 2016-05-31 by perexg