perexg/satip-axe #55: No input source available for subscription

satip, 2016-03-18:
hi

iam using fw version 201603051934-11 and tvheadend 4.1-1695~g7adc49f
sometimes, tvheadend tells me:

Mar 18 15:55:10 nas tvheadend[5859]: subscription: 00D2: service instance is bad, reason: No input detected
Mar 18 15:55:10 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:12 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:12 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:12 nas tvheadend[5859]: dvr: Recording unable to start: "Scorpion": No input detected
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:14 nas tvheadend[5859]: mpegts: 11273.25H in SAT - tuning on SAT>IP DVB-S Tuner #1 (192.168.240.22)
Mar 18 15:55:16 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:16 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:18 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:18 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:18 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:18 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:20 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:20 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:22 nas tvheadend[5859]: subscription: 00D2: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
Mar 18 15:55:22 nas tvheadend[5859]: subscription: 00D1: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"

this was seen only in "newer" tvheadend-versions (> 4.1-1575)

after restarting satip-box (via putty/ssh),  tvheadend is able to tune and all is working well

at this time, i have also a ticket on https://tvheadend.org/issues/3611 because of stability problems/crashes with tvheadend (see case vor more infos)

perexg, 2016-03-18:
Specify which minisatip binary do you use and the config.

satip, 2016-03-18:
/mnt/data/satip-axe/sysconfig/sysconfig
http://pastebin.com/JduME3K5

satip, 2016-03-18:
sorry

satip, 2016-03-18:
ps exaf | grep minisatip
 3187 root       7:05 minisatip -f -g -ll -d 38
 3285 root       0:00 grep minisatip

satip, 2016-03-22:
again (3 times) - see log from last night:
http://pastebin.com/8HKrStaS

satip, 2016-03-23:
again

output satip´s "dmesg":
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
[dms-ts] stream stop ok

tvheadend:

2016-03-23 06:21:38.820 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:40.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:42.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:44.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:46.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:48.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:50.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:52.821 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:54.822 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:56.822 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
2016-03-23 06:21:58.822 [ NOTICE] subscription: 0178: No input source available for subscription "DVR: Scorpion" to channel "MYCHANNEL1"
[Thread 0x7fff9a7f4700 (LWP 11723) exited]
2016-03-23 06:22:00.003 [   INFO] subscription: 0178: "DVR: Scorpion" unsubscribing from "MYCHANNEL1"
2016-03-23 06:22:00.003 [   INFO] dvr: "Scorpion" on "MYCHANNEL1": End of program: Time missed
2016-03-23 06:22:00.007 [   INFO] dvr: delete entry 9868d397a0d9a1e9b4b88952476a5c53 "Scorpion" on "MYCHANNEL1" start time 1458741900, scheduled for recording by "s6app" (duplicate with 9868d397a0d9a1e9b4b88952476a5c53)

my android satip-app "tivizen" cannot find any satip-devices

after rebooting satip-device all is working fine (with tvheadend and tivizen)

should i start whireshark?

satip, 2016-03-23:
updating to https://github.com/perexg/satip-axe/issues
changing to minisatip5 
using default config file from
https://raw.githubusercontent.com/perexg/satip-axe/3d93c74f8afcb349499ac27fd6b045efce38fcbd/fs-add/etc/config.default

changing default to:
TELNETD="no"

MINISATIP="no"
MINISATIP_OPTS=""

MINISATIP5="yes"
MINISATIP5_OPTS=""

OSCAMD="no"

satip, 2016-03-28:
no problems until now
closing case

closed 2016-03-28 by satip