perexg/satip-axe #57: minisatip updates for build 12

perexg, 2016-03-23:
*minisatip-201603231127-11.tar.gz* : http://s000.tinyupload.com/index.php?file_id=09275279209718883188

* some input allocation fixes
* fixed -L option for minisatip5
* added -A option for minisatip5 (free input mode)

m4tt075, 2016-04-07:
@perexg: I am running201603301537-11 and am pretty certain that minisatip5 significantly increases continuity errors in my environment vs. minisatip. I have two Digibit boxes on two different sat positions (19.2E and 28.2E) with identical quattro tuners and have seen the effect on both systems.
Here what I tested:
* Minisatip5: -Q and RTSP/TCP enabled or disabled -> Plenty of cont errors
* Minisatip5: -Q -T and RTSP/TCP enabled or disabled -> Plenty of cont errors
* Minisatip: -Q and RTSP/TCP disabled -> Very few cont errors
Just an observation for the time being but if you want me to try certain settings or debug somehow, please just let me know...

m4tt075, 2016-04-09:
@perexg: Ran another test over the last 24h:

* Minisatip: -Q and RTSP/TCP enabled -> Very few cont errors

No idea why but points at Minisatip5.

m4tt075, 2016-04-11:
@perexg: The old Minisatip version with the last set of settings with -Q and RTSP/TCP enabled is by far the best one. Almost no continuity errors since about 4 days.

m4tt075, 2016-05-14:
@perexg: Seen you have been active here again after quite some time. Any idea about the above?

perexg, 2016-05-14:
@m4tt075 : Have you tried to add '-P 1', too ?

m4tt075, 2016-05-15:
@perexg: No I hadn't but switched to minisatip5 with "-Q -P1" args this morning. Significant number of continuity errors since (see attached screenshot). For comparison, with the old minisatip version I usually get 0-3 continuity errors per tuner in about 24 hours... 
[image]

perexg, 2016-05-15:
@m4tt075 : The CC errors at the start of stream should be ignored. Do you observe any picture / sound distortions ? Or does CC errors increase when the stream is active ?

m4tt075, 2016-05-15:
@perexg : Hmh, can't tell to be honest. I simply observed all those errors and thought the setup producing fewest of those would be best. What I realized with earlier tests is that recordings contained data errors though. I will observe and report back. Might take some time...

m4tt075, 2016-05-16:
I have watched a couple of progams in the meantime. So far I have not observed any CC errors being produced while watching Live TV. The high number of errors are apparently produced through EPG grabbing. I have updated to your latest minisatip package now. Will continue testing...

paluseb, 2016-05-16:
"added packet counter (hardware) / CC error counter (hardware) / coax cable colums to status.html"

How to see the status of html?
I tried minisatip5 -w 192.168.1.1:8080
but I do not see?
minisatip (5) probably does not have a web interface?

perexg, 2016-05-17:
URL: http://satip1:8080/

[image]

The -w options if for different purpose. Just keep defaults - the minisatip HTTP server is on port 8080.

paluseb, 2016-05-17:
I updated (I think) the last official version is that of 15.05.2016 but that I certainly did it?
And if I use minisatip or minusatip5? I assume the log, but that version minisatip5?

satip-axe login: root
Password:

Welcome to SATIP-AXE server 201603051934-11

~ # minisatip -h
[16/05 12:56:11.281]: minisatip [-f] [-r remote_rtp_host] [-d device_id] [-w http_server[:port]] [-p public_host] [-s rtp_port] [-a no] [-m mac] [-l] [-a X:Y:Z] [-e X-Y,Z]
                -f foreground, otherwise run in background
                -r remote_rtp_host: send remote rtp to remote_rtp_host
                -d specify the device id (in case there are multiple SAT>IP servers in the network)
                -w http_server[:port]: specify the host and the port where the xml file can be downloaded from
                -x port: port for listening on http
                -s force to get signal from the DVB hardware every 200ms (use with care, only when needed)
                -a x:y:z simulate x DVB-S2, y DVB-T2 and z DVB-C adapters on this box (0 means auto-detect)
                -m xx: simulate xx as local mac address, generates UUID based on mac
                -e list_of_enabled adapters: enable only specified adapters, example 0-2,5,7 (no spaces between parameters)
                -c X: bandwidth capping for the output to the network (default: unlimited)
                -b X: set the DVR buffer to X KB (default: 5640KB)
                -l increases the verbosity (you can use multiple -l), logging to stdout in foreground mode or in /tmp/log when a daemon
                -g use syslog instead stdout for logging, multiple -g - print to stderr as well
                -p url: specify playlist url using X_SATIPM3U header
                -u unicable_string: defines the unicable adapters (A) and their slot (S) and frequency (F):
                        The format is: A1:S1-F1[,A2:S2-F2[,...]]
                        When * character is used before frequency, force 13V for setup
                -j jess_string: same format as unicable_string
                -L link adapters (identical src,lo/hi,h/v), the format is M:S (master:slave)
                -Q quattro LNB config (H/H,H/V,L/H,L/V)
                -Z X: enable loband (0) or hiband (1) only for quattro LNB
                -X X[,Y]: AXE unicable/jess input (0-3) for all tuners
                -M X: skip initial MPEG-TS packets for AXE demuxer (default 35)
                -P X: power to all inputs (0 = only active inputs, 1 = all inputs)

~ # minisatip5 -h
minisatip version 0.5.49-axe102, compiled with s2api version: 0501 (AXE)
[16/05 12:56:26.144]: Built without dvbcsa
[16/05 12:56:26.145]: Built without CI
[16/05 12:56:26.145]: Built without dvbapi
[16/05 12:56:26.145]: Built with AES (OpenSSL)
[16/05 12:56:26.145]: Built without tables processing
[16/05 12:56:26.145]: Built with satip client
[16/05 12:56:26.145]: Built with linux dvb client
[16/05 12:56:26.145]: Built with backtrace
[16/05 12:56:26.145]: Built without netceiver

        ./minisatip [-[fgltzQ]] [-a x:y:z] [-b X:Y] [-c X] [-d A:C-U ] [-D device_id] [-e X-Y,Z] [-i prio]
                [-[uj] A1:S1-F1[-PIN]] [-m mac][-p public_host] [-r remote_rtp_host]
                [-R document_root] [-s [DELSYS:]host[:port] [-u A1:S1-F1[-PIN]] [-w http_server[:port]]
                [-x http_port] [-X xml_path] [-y rtsp_port] [-L M1:S1[,M2:S2]] [-U unicable_adapter]
                [-M mpegts_packets] [-A SRC1:INP1:DISEQC1[,SRC2:INP2:DISEQC2]]

How to run a Web interface? How can I check the settings or change the port?

perexg, 2016-05-17:
Web interface is available by default on tcp port 8080 - see my example URL.

paluseb, 2016-05-17:
I am using version minisatip or minisatip 5?
and that is the last patch released because the release date in the log is not correct?

perexg, 2016-05-17:
*minisatip version 0.5.49-axe102* is the last version, the date before is the log date, not the release date.

paluseb, 2016-05-17:
the date before is the log date, not the release date

I now bat my question is this date:
Welcome to SATIP-AXE server 201603051934-11

or just change
MINISATIP = "yes"
on
MINISATIP = "yes"?

trujulu, 2016-05-17:
Hello, sorry for the interruption but Perexg could you update the lenguages from Transifex? Thanks.

perexg, 2016-05-17:
> Welcome to SATIP-AXE server 201603051934-11

This is the base build number, it has nothing to do with the minisatip build / version.

> MINISATIP = "yes"

You should have two sections in the /etc/sysconfig/config file, one MINISATIP and the other one MINISATIP5 . If you don't have MINISATIP5 section - your config is based on the older satip-axe release - copy it from /etc/config.default .

paluseb, 2016-05-17:
>  If you don't have MINISATIP5 section - your config is based on the older satip-axe release - copy it from /etc/config.default .

> 

Just copy the file or in addition to edit the settings?
And the configuration to edit config. or config.default?

m4tt075, 2016-05-18:
@perexg: Nope, having actual visible screen artifacts with minisatip5, too. I started watching the UEFA cup final about 60 min ago and got blocks and stuttering from time-to-time. 

TVH WebUI -> Status -> Stream reports: 34 corrected blocks and 43 CC errors.
Sat>IP box reports (via your new web interface on 8080): 20 CC errors 

TVH log excerpt:

2016-05-18 22:32:21.939 [  DEBUG]:capmt: OSCAM: ECM_INFO: adapter=0 sid=5505
caid=1843(nagra) pid=19AF provid=000000 ecmtime=428 hops=0 reader=hdplus02 from=local protocol=mouse
2016-05-18 22:32:25.826 [WARNING]:TS: AstraGER/10802.75H/SPORT1 HD: H264 @ #1535 Continuity counter error (total 3)
2016-05-18 22:32:25.826 [WARNING]:TS: AstraGER/10802.75H/SPORT1 HD: AC3 @ #1539 Continuity counter error (total 3)
2016-05-18 22:32:25.826 [WARNING]:TS: AstraGER/10802.75H/SPORT1 HD: TELETEXT @ #37 Continuity counter error (total 4)
2016-05-18 22:32:25.837 [  ERROR]:parser: transport stream TELETEXT, DTS discontinuity. DTS = 196898400, last = 196632000
2016-05-18 22:32:25.840 [  ERROR]:parser: transport stream H264, DTS discontinuity. DTS = 197028515, last = 196762115
2016-05-18 22:32:25.857 [  DEBUG]:ecm: 10802.75H in AstraGER: PID 19AF CC error 5 != 9
2016-05-18 22:32:25.878 [  ERROR]:parser: transport stream AC3, DTS discontinuity. DTS = 196977600, last = 196701120
2016-05-18 22:32:25.933 [  DEBUG]:eit: 10802.75H in AstraGER: PID 0012 CC error 11 != 12
2016-05-18 22:32:25.958 [  DEBUG]:pmt: 10802.75H in AstraGER: PID 0065 CC error 8 != 0
2016-05-18 22:32:25.958 [  DEBUG]:cat: 10802.75H in AstraGER: PID 0001 CC error 11 != 14
2016-05-18 22:32:25.958 [  DEBUG]:pat: 10802.75H in AstraGER: PID 0000 CC error 7 != 10
2016-05-18 22:32:25.958 [  DEBUG]:emm: 10802.75H in AstraGER: PID 1005 CC error 1 != 14
2016-05-18 22:32:26.445 [  DEBUG]:nit: 10802.75H in AstraGER: PID 0010 CC error 4 != 11
2016-05-18 22:32:26.840 [  DEBUG]:bat: 10802.75H in AstraGER: PID 0011 CC error 1 != 13
2016-05-18 22:32:26.840 [  DEBUG]:sdt: 10802.75H in AstraGER: PID 0011 CC error 1 != 13
2016-05-18 22:32:28.938 [  DEBUG]:capmt: OSCAM, CA_SET_DESCR adapter 0 par 0 idx 0 110c36533c65dc7d
2016-05-18 22:32:28.938 [  DEBUG]:descrambler: info - service='SPORT1 HD' caid=1843(nagra) provid=000000 ecmtime=420 hops=0 reader='hdplus02' from='local' protocol='mouse'

Please let me know if you want me to test anything else and I will try, but I'm tempted to revert to stable minisatip again as this is my production / family system...

perexg, 2016-05-19:
@m4tt075 : No idea. You may try to determine if other tuner / input might cause the stream interruption. Also minisatip5 uses a bit different DiseqC timing. The CC errors in the web interface shows the hardware CC errors (in data before they go to minisatip).

m4tt075, 2016-05-19:
@perexg: Can you guide me on how to determine whether other tuners / inputs cause stream errors?

In the meantime, some more observations, which might or might not be helpful:
1) The vast majority of CC errors observed on the Digibit (via 8080) occur when tuning begins
2) Somehow I cannot make sense of how CC errors are counted at different places though. Example: - The CC errors on the Digibit (via 8080) for a single stream increased by 1
- Around that time I observed a MPEG2VID/AUDio DTS discontinuity in the TVH log on my TVH server
- However, the continuity error counter in the TVH WebUI (under Status -> Stream) stayed at 0
- I actually did not notice a screen distortion, but not sure I just missed it because I was looking at logs... ;-)
In any case, does that make sense to you?
3) Yesterday, I have observed picture distortions in combination with CC errors and uncorrected blocks in the TVH WebUI while the CC errors on the Digibit (via 8080) stayed constant, which might indicate (temporary) network issues...

paluseb, 2016-05-19:
I tried to change the config as @satip:
https://github.com/perexg/satip-axe/issues/55
but without success. Do you have any suggest this?

JV16Bar, 2016-05-31:
Hello,  can you please update minisatip to the latest version? It fixes an important issue: using the same tuner, if 2 clients are asking 2 channels from the same frrequency. 
https://github.com/catalinii/minisatip/issues/300
thanks for the great work!

perexg, 2016-05-31:
Added all to build 12 (including minisatip5 upgrade to 0.5.50). Don't forget to remove the old packages from /mnt/data/satip-axe/packages when upgraded.

closed 2016-05-31 by perexg