perexg/satip-axe #61: Unicable II LNB

niceborn, 2016-05-13:
Hello,

I have a possible issue with the unicable II support. I have an Inverto IDLU-32UL40-UNBOO-OPP 32 channel Unicable 1/2 LNB. This LNB can support up to 32 tuners. I have no problem to use this LNB with the original Digibit firmware (V156) but with SATIP-AXE only a view channels could be used. Even some unicable 1 channels refused to work.

This are the possible channels of the LNB:

CH1 1210MHz (EN50494+EN50607)
 	CH2 1420MHz (EN50494+EN50607)
 	CH3 1680MHz (EN50494+EN50607)
 	CH4 2040MHz (EN50494+EN50607)
 	CH5 984MHz (EN50494+EN50607)
 	CH6 1020MHz (EN50494+EN50607)
 	CH7 1056MHz (EN50494+EN50607)
 	CH8 1092MHz (EN50494+EN50607)
 	CH9 1128MHz (EN50607)
 	CH10 1164MHz (EN50607)
 	CH11 1256MHz (EN50607)
 	CH12 1292MHz (EN50607)
 	CH13 1328MHz (EN50607)
 	CH14 1364MHz (EN50607)
 	CH15 1458MHz (EN50607)
 	CH16 1494MHz (EN50607)
 	CH17 1530MHz (EN50607)
 	CH18 1566MHz (EN50607)
 	CH19 1602MHz (EN50607)
 	CH20 1638MHz (EN50607)
 	CH21 1716MHz (EN50607)
 	CH22 1752MHz (EN50607)
 	CH23 1788MHz (EN50607)
 	CH24 1824MHz (EN50607)
 	CH25 1860MHz (EN50607)
 	CH26 1896MHz (EN50607)
 	CH27 1932MHz (EN50607)
 	CH28 1968MHz (EN50607)
 	CH29 2004MHz (EN50607)
 	CH30 2076MHz (EN50607)
 	CH31 2112MHz (EN50607)
 	CH32 2148MHz (EN50607)

Is there a restriction of the possible channel and frequence or how is it possible to use e.g. channel 9 to 12?

perexg, 2016-05-14:
Have you tried minisatip5 ? Also, show the minuisatip parameters you're trying to use..

niceborn, 2016-05-14:
I tried both, minisatip and minisatip5. The parameter was 
MINISATIP_OPTS="-u 0:8-1128,1:9-1164,2:10-1256,3:11-1292" or
MINISATIP5_OPTS="-u 0:8-1128,1:9-1164,2:10-1256,3:11-1292"

to use the following channels, that works well with the original firmware:

CH9 1128MHz (EN50607)
CH10 1164MHz (EN50607)
CH11 1256MHz (EN50607)
CH12 1292MHz (EN50607)

trujulu, 2016-05-14:
I have also problems when trying to use 2 frecuencias at the same time. Tvhead says busy device all times.

niceborn, 2016-05-14:
Before I used an INVERTO BLACK IDLP-USS200-CUO10-8PP Unicable switch without any problems with 4 parallel channels with this settings

MINISATIP/MINISATIP5_OPTS="-u 0:4-1484,1:5-1586,2:6-1688,3:7-1790"

The only issue (only with minisatip) was, that after a time without a tuning the first request was unsuccessful and leads to a switch to the second tuner in TVHeadend. After that the first tuner was working again.

perexg, 2016-05-14:
@niceborn: Unicable II is jess in minisatip. You should use -j parameter.
@trujulu: It looks like a wrong minisatip or tvh config.

niceborn, 2016-05-14:
Thanks perexg, I will try later and report. Perhaps you should update the minisat.md documentation or where do i find a complete list of parameters and settings to use?

perexg, 2016-05-14:
minisatip -h ? but appearently, nobody uses that :-)

niceborn, 2016-05-15:
Now I tried both, minisatip and minisatip5. The parameters used were
MINISATIP_OPTS="-j 0:8-1128,1:9-1164,2:10-1256,3:11-1292" or
MINISATIP5_OPTS="-j 0:8-1128,1:9-1164,2:10-1256,3:11-1292" 
but I don't get a tuning. TVH showed shortly a strength of 100% and a SNR of 0%. The Tuner LED of the Digibit was shortly on and then off.

I try to use your firmware, because TVH shows a lot of uncorrected errors with the original firmware.

perexg, 2016-05-15:
Provide log, use '-llll' options for minisatip/5. The best way is to create a file /tmp/nosatip, kill the minisatip and run minisatip from ssh on cmd line with debug options (and all options as in config file).

niceborn, 2016-05-15:
Thanks, I will try tomorrow, because of WAF (Woman acceptance factor) :-)

perexg, 2016-05-15:
Could you test this minisatip5 binary http://s000.tinyupload.com/?file_id=05432057424516225260 ?

I think that https://github.com/catalinii/minisatip/blob/master/dvb.c#L527 line is wrong (it sets the frequency), it should be:

<pre>
  cmd.msg[1] |= ((t >> 8) & 0x07);
</pre>

niceborn, 2016-05-15:
Could you tell me, what to do with this file, please?

perexg, 2016-05-15:
Same way as for '-lll' - kill the minisatip and prevent it to be restarted and run the new uploaded binary.

niceborn, 2016-05-15:
I'm sorry, but my Linux knowledge is restricted. How do I copy the file to the Digibit to run it? Thanks for your help!

perexg, 2016-05-15:
The package with updated minisatip5 is available in #57 .

niceborn, 2016-05-16:
Thanks for the update, but how do I copy the .tar.gz from my windows computer to the digibit?

perexg, 2016-05-16:
Through your ssh client, like putty for windows or anything else which support ssh protocol...

niceborn, 2016-05-16:
Yes, but I don't know the comand to copy the file within ssh client. Could you help, please?

niceborn, 2016-05-17:
Hi perexg,

I found a solution myself by copy the gz to the usb stick and then move the file to the packages folder. 

The updated minisatip5 is working without any problems with my LNB with the above settings. Thanks for the update! Do I have to delete the file in the packages folder now?

perexg, 2016-05-17:
Keep the package file - it overwrites the old version on boot. The filesystem is in RAM - thus it is not persistent.

perexg, 2016-05-17:
I also should fix the diseqc command in old minisatip...

perexg, 2016-05-31:
Fixed in build 12 for both minisatip / minisatip5. Don't forget to remove the old packages from /mnt/data/satip-axe/packages when upgraded.

closed 2016-05-31 by perexg