perexg/satip-axe #62: satip: Cannot get http://192.168.1.30:8080/0: Success

paluseb, 2016-05-22:
Do you use minisatip5 and on what version of TVH?
You can post your config?
I ask because I have the same problem as you advance.
More here:
https://github.com/perexg/satip-axe/issues/57

perexg, 2016-05-31:
Could you give output from 'ps', 'logread' and 'netstat -an' ?

But it looks like a network configuration issue rather a bug in the satip-axe .

paluseb, 2016-06-06:
Logs give later. I'm surprised such action client satip Panasonic tv. Unencrypted channels (even those test UHD) work properly if I use the original software Telestar R1 as well minisatip. When the factory firmware if the decryption is made in the TV through the CI module also everything works properly. But when using minisatip and CI module for video noise appears. The same situation is where descrambling takes place in a stream tvheadend.

perexg, 2016-06-06:
I doubt about the last sentence. Note that almost all SAT>IP clients use UDP transport which may be intolerant to ethernet packet drops. I know that the original firmware does some packet scheduling to eliminate this issue (but it's not 100%). TVH with RTP/AVP/TCP enabled (embedded RTSP data mode) should be fine.

paluseb, 2016-06-06:
satip-axe login: root
Password:

Welcome to SATIP-AXE server 201605311610-12

~ # ps
PID   USER     TIME   COMMAND
    1 root       0:00 /bin/busybox init
    2 root       0:00 [kthreadd]
    3 root       0:00 [ksoftirqd/0]
    4 root       0:00 [events/0]
    5 root       0:00 [khelper]
   13 root       0:00 [async/mgr]
   14 root       0:00 [pm]
  174 root       0:00 [sync_supers]
  176 root       0:00 [bdi-default]
  178 root       0:00 [kblockd/0]
  186 root       0:00 [ksuspend_usbd]
  191 root       0:00 [khubd]
  194 root       0:00 [kseriod]
  209 root       0:00 [rpciod/0]
  215 root       0:00 [kswapd0]
  216 root       0:00 [aio/0]
  217 root       0:00 [nfsiod]
  218 root       0:00 [kslowd000]
  219 root       0:00 [kslowd001]
  221 root       0:00 [crypto/0]
  309 root       0:00 [mtdblockd]
  378 root       0:00 [scsi_eh_0]
  379 root       0:00 [usb-storage]
  461 root       0:00 syslogd -C256
  464 root       0:00 [jffs2_gcd_mtd4]
  473 root       0:00 [STTBX_CircBuf_O]
  474 root       0:00 [STFDMA_ClbckMgr]
  475 root       0:00 [STFDMA_ClbckMgr]
  476 root       0:00 [STFDMA_ClbckMgr]
  485 root       0:00 [kfe-0]
  492 root       0:00 [kdmxts_64]
  493 root       0:00 [kdmxts_64]
  494 root       0:00 [kdmxts_64]
  495 root       0:00 [kdmxts_64]
  496 root       0:00 [kdmxts_64]
  497 root       0:00 [kdmxts_64]
  498 root       0:00 [kdmxts_64]
  499 root       0:00 [kdmxts_64]
  500 root       0:00 [kdmxts_64]
  501 root       0:00 [kdmxts_64]
  502 root       0:00 [kdmxts_64]
  503 root       0:00 [kdmxts_64]
  504 root       0:00 [kdmxts_64]
  505 root       0:00 [kdmxts_64]
  506 root       0:00 [kdmxts_64]
  507 root       0:00 [kdmxts_64]
  508 root       0:00 [kdmxts_64]
  509 root       0:00 [kdmxts_64]
  510 root       0:00 [kdmxts_64]
  511 root       0:00 [kdmxts_64]
  512 root       0:00 [kdmxts_64]
  513 root       0:00 [kdmxts_64]
  514 root       0:00 [kdmxts_64]
  515 root       0:00 [kdmxts_64]
  516 root       0:00 [kdmxts_64]
  517 root       0:00 [kdmxts_64]
  518 root       0:00 [kdmxts_64]
  519 root       0:00 [kdmxts_64]
  520 root       0:00 [kdmxts_64]
  521 root       0:00 [kdmxts_64]
  522 root       0:00 [kdmxts_64]
  523 root       0:00 [kdmxts_64]
  524 root       0:00 [kdmxts_64]
[... 214 lines cut ...]
udp        0      0 0.0.0.0:34744           0.0.0.0:*
udp        0      0 0.0.0.0:952             0.0.0.0:*
udp        0      0 0.0.0.0:50240           0.0.0.0:*
udp        0      0 239.255.255.250:1900    0.0.0.0:*
udp        0      0 0.0.0.0:1900            0.0.0.0:*
udp        0      0 0.0.0.0:111             0.0.0.0:*
udp        0      0 192.168.1.30:5500       192.168.1.187:3456      ESTABLISHED
udp        0      0 192.168.1.30:5501       192.168.1.187:3457      ESTABLISHED
udp        0      0 :::59609                :::*
udp        0      0 :::123                  :::*
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node Path
unix  2      [ ACC ]     STREAM     LISTENING       2591 /var/run/rpcbind.sock
unix  9      [ ]         DGRAM                      1006 /dev/log
unix  3      [ ]         STREAM     CONNECTED       4227
unix  3      [ ]         STREAM     CONNECTED       4226
unix  2      [ ]         DGRAM                      4225
unix  2      [ ]         DGRAM                      4213
unix  2      [ ]         DGRAM                      2794
unix  2      [ ]         DGRAM                      2656
unix  2      [ ]         DGRAM                      2646
unix  2      [ ]         DGRAM                      2599
unix  2      [ ]         DGRAM                      2589
~ #
192.168.1.187 it"s Panasonic TV

perexg, 2016-06-07:
If TVH with RTP/AVP/TCP enabled (embedded RTSP data mode) works, the problem is in the ethernet packet loss as I wrote before.

9000h, 2016-06-07:
but "autoneg off" you should take in account that you have to make sure you need to set the same parameters on the R1 and the switch

paluseb, 2016-06-08:
How can I check if I turned off auto negotiation network connection?

paluseb, 2016-06-19:
@perexg help me. What I can do? Why TVH don't shows minisatip5? minisatip workd fine.

perexg, 2016-06-19:
Replace '-X 0' with '-U 0' for minisatip5. This parameters is different ('-X' option is for the XML file - which explains /0 in URL for minisatip5).

closed 2016-06-19 by perexg