perexg/satip-axe #63: Heavy artefacts while playing rtsp stream

HellGL, 2016-06-14:
Hi,
yesterday I tried the new V12 firmware for my GSS.Box with unicable and did some tests. The rtsp stream played with VLC under Linux (GSS.Box -> GBit-Switch -> Fritz.Box (Full Power Lan) -> WLAN) shows heavy artefacts and is hardly visible (same on the RPi2). The same stream over http plays fine however. I checked other issues here and tried to switch the ethernet speed to 100M. This helped for the rtsp stream and distortions are only visible every 10 to 20 seconds. minisatip5 with "-O" did not help. I have a dlink 1GB "Green" switch which is already quite old. Do you think it would help to by a new switch or what buffers could I tweak to improve the situation?

perexg, 2016-06-16:
Perhaps you can try to force RTP over RTSP (TCP) in VLC:

https://www.wowza.com/forums/content.php?64-How-to-configure-VLC-media-player-for-RTSP-RTP-playback-%28RTSP-RTP-interleaved-and-tuning%29&s=3d20ee8465f7e079a029c77c57eb1b0a

HellGL, 2016-06-16:
Thanks for the links. I will try that with VLC and see if it helps. As it is an option for the SETUP command I can try to implement it in the satip client and see if it also works on the Raspberry Pi. 
If I remember correctly the artefacts are similar to the errors I got while I tried the V1.24.0.156 firmware. Yesterday I tried the new Inverto firmware 1.25.0.157 and the picture is quite perfect. However I have an other error: while I can start everything up and watch "Das Erste HD" it is not possible to switch to "ZDF HD". Switching further to other HD or SD channels work but ZDF does not work. This is consistent with different clients (Raspberry Pi with vdr-plugin-satip, vtuner/satip, iOS App, HTTP stream with VLC). Do you think it would help to update your firmware to 1.25.0.157 toolchain? I can see on https://github.com/jollyjinx/idl4k/ that there are updates in the kernel network drivers.

perexg, 2016-06-16:
The original firmware does some UDP packet scheduling - it means that it tries to send UDP/RTP packets with the continuous timing. Minisatip uses standard linux network buffering, so the packets can be send in chunks which may be rejected by any component on the path (an ethernet switch {limited buffers}, a receiving ethernet adapter etc.).

If you use the TCP protocol, then the information is repeated again when some packets are lost. Of course, it might saturate the network more, but usually, you'll get more good results if there's enough bandwidth in your ethernet network.

timehlers, 2016-06-16:
Hi,

I just checked the differences of the kernel modules from the new V1.25.0.157 and what we had before. Indeed there are differences on 2 kernel modules:

axe_dmxts.ko
axe_fe.ko

I'm using these two new modules now and I have the feeling that distortions in an rtsp stream to my raspberry pi VDR (with satip plugin) are gone now (or at least much more seldom). I had some artifacts from time to time before.

I mean this is just an impression now, maybe I'm wrong. But release notes at least say "General improvements of stability and performance", maybe it is worth to try the modules.

Best

Tim

perexg, 2016-06-16:
I am testing the new kernel modules from idl4k-1.25.0.157 fw, too. But I've not recognized any differences. Test build of satip-axe firmware is in issue #65 .

HellGL, 2016-06-16:
RTSP with TCP did the trick with VLC and a first test with vtuner/satip on my Intel NAS does not show distortions. I have to check the logs regarding stream errors but so far switching to TCP might do the trick! But first I will try the new firmware and see if this already helps with my RPis.

timehlers, 2016-06-17:
Maybe a bit increasing the buffers in the RPI does also help, these are my current addings to sysctl.conf:

`net.ipv4.tcp_no_metrics_save = 1`
`net.ipv4.tcp_window_scaling = 1`
`net.ipv4.tcp_syncookies = 0`
`net.core.rmem_max = 33554432`
`net.core.wmem_max = 33554432`
`net.ipv4.tcp_rmem = 2621440 3276800 39321600`
`net.ipv4.tcp_wmem = 2621440 3276800 39321600`
`net.ipv4.udp_rmem_min = 2621440`
`net.ipv4.udp_wmem_min = 2621440`
`net.ipv4.udp_mem = 2621440 3276800 39321600`
`net.core.rmem_default = 2048000`
`net.core.wmem_default = 2048000`
`vm.dirty_expire_centisecs = 1000`
`net.core.netdev_max_backlog = 6000`

Best

Tim

timehlers, 2016-06-22:
Using the new kernel modules now for some days, I have to correct myself, the distortions are the same like before (that's very seldom, so it takes to value). But I found another optimization: I have enabled "flow control" in my central HP Switch and I had no distortions until now!

Maybe this helps some people...

CvH, 2016-06-22:
@timehlers vdr satip could now use rtsp/tcp :)

timehlers, 2016-06-22:
Wow, thanks CvH. The world is moving... :) I'll test this tomorrow at latest.

timehlers, 2016-06-30:
Hi,

when I switch on "use RTP-over-TCP mode" in vdr-plugin-satip, I get really bad switching timings. Often I only get a black screen forever after changing the channel. Is this a known issue (do you have that too?), or does this only happen on my setting?

Best Tim

CvH, 2016-06-30:
@timehlers I have no VDR in usage (Tvh ..) :( This sounds like an bigger bug in the plugin - normally it has 0,00 impact of the usage if you activate RTP/TCP.

perexg, 2017-06-07:
This is old bug and I think that it's mostly problem with the network config or client issue. Please, open a new one, if you can reproduce this with minisatip7 (latest build 14).

closed 2017-06-07 by perexg