perexg/satip-axe #64: DIGIBIT R1 irregular looses network connection

qpirsel, 2016-06-16:
Hi,

I've noticed that my DIGIBIT R1 irregular looses the network connection (network status LED off). Satip-axe is booting from usb.

I can solve it only due a power reset of the device. The strange is that it can happen after a few days of running but also within 5 minutes after a power reset.

I'm pretty sure that it's not a problem of satip-axe because it also happens with the original firmware running in different connection environments (dhcp,static,switch,directly connected, etc.). But maybe there are tweaks possible with satip-axe that can solve this (known?) issue.

Thanks in advance for any advise
qpirsel

HellGL, 2016-06-16:
I think without a serial connection to debug it would be very difficult to track down this error.

asterix555, 2016-06-16:
Did you replace cable and switch already?
Might be a issue related to auto-negotiation caused by defective cable or switch - or even the digibit R1,..

qpirsel, 2016-06-16:
I've already changed the cable and the kind of connection (switch / directly to PC). Anyway, shouldn't the network come up again after e.g. replug the network cable if it is related to negotiation?

I read that it could be useful to force 100Mbit with ethtool. I will give it a try. If the error appears again I guess I really must add a serial connector to track this issue.

perexg, 2016-06-16:
It looks like a hardware fault of R1 - the box seems dead, but you cannot confirm that without a serial console (thus warranty void). Perhaps, it might be better to return the original firmware and apply the warranty, if this issue is reproducible with the original fw, before you open the box.

qpirsel, 2016-06-16:
The seal is already broken (I MUST open everything :) so modifing the box is no problem at all.

qpirsel, 2016-08-13:
Ok, I've found the time to start researching. I think I did the connection correct but when I connect the box with my PC it does not indicate any boot signals (Knight Rider effect).

Just to verify the steps:
- upper usb port with dist release
- Serial connection J3 to PC (2-5 | 3-2 | 4-3)
- Serial setting 115200 8N1 (COM1 and Putty properties)
- Power on the box and press ENTER within 1 second

What I'm doing wrong? :)

perexg, 2016-08-14:
The ENTER is required for the uboot prompt. Just don't press it and you should see the linux serial console / linux boot messages, if the wiring is correct. Also, disable handshake control (RTS/CTS) in your terminal program.

qpirsel, 2016-08-14:
Thank you for the clarification of the behaviour.

qpirsel, 2016-08-16:
K, I got it. I got a TTL USB converter from a colleague that works. I'll start logging within the next days. A current boot log was added at http://pastebin.com/TJkvDQBb

qpirsel, 2016-10-25:
Ok, it tooks a while, because after my last post the box was running with the original firmware without any issues for weeks. Then it happened again, so I started the box with satip-axe and began to log. This was 2 weeks ago and this night the issue appeared again.

These are tha last few lines before the box **freezes totally**. Only a cold restart solves this state:
[i2c] read(0x15,1) 0x2C
[i2c] t1_w(0xD0, 2) 0xF3.99
[i2c] t1_r(0xD1, 3) 0x00.00.00
[i2c] t1_w(0xD0, 2) 0xF2.0E
[i2c] t1_r(0xD1, 2) 0x93.BF
[i2c] t1_w(0xD0, 2) 0xF2.84
[i2c] t1_r(0xD1, 2) 0x0B.72

closed 2016-11-22 by qpirsel