perexg/satip-axe #65: Alpha/Beta releases for build 14

perexg, 2016-06-16:


m4tt075, 2016-06-16:
What happened to build 13???  :)

m4tt075, 2016-07-08:
Just a quick note to whom it may concern:
Upgraded from build 11 to 201607042257-12 via ssh and upgrade-fw on two different Digibit R1. After rebooting the network connections were gone. I couldn't even ssh into the boxes again. Flashing the same (new) fw again via USB stick solved the issue on both boxes...

estimadarocha, 2016-11-16:
Hello @perexg...

sorry to be annoying you but could you please try to make a new beta build with new commit from minisatip? @catalinii make some changes to improve the buffer issue on wan.

thanks

flotux2, 2017-04-25:
Hi @perexg 

thanks for the new build. It uses minisatip5 as far as I can see from a ps -ef call:

3367 root       0:01 minisatip5 -f -g -P 1

How can enable the minisatip7 version to be started at boot time? I found the config.default file in /etc that looks like:

# minisatip 0.5
#
#MINISATIP5="yes"
#MINISATIP5_OPTS=""

#
# minisatip 0.7
#
MINISATIP7="yes"
MINISATIP7_OPTS=""

Can I change it to minisatip7 also when booting from USB stick and not flashing the firmware to the internal flash storage?

Please support a newbie. :-)

Thanks
Florian

perexg, 2017-04-26:
Just merge the /etc/config.default with your /etc/sysconfig/config . The configuration file is not replaced when you upgrade to the new version. You should do it yourself (select new minisatip version and update arguments).

arrobar, 2017-04-26:
To see some channels in Spain I need to change the LO1 / LO2 frequency to 9750/9750 on my DIGIBIT r1.
In the github of catalinii indicates that the modifier to be used is -L:

-L --lnb specifies the adapter and LNB parameters (low, high and switch frequency)

eg: -L *:9750-10600-11700 - sets all the adapters to use Universal LNB parameters (default)
eg: -L *:10750-10750-10750 - sets the parameters for Sky NZ LNB using 10750 Mhz
eg: -L 0:10750-10750-10750,1:9750-10600-11700 - adapter 0 has a SKY NZ LNB, adapter 1 has an Universal LNB

Which modifier should I use with minisat7

Thanks
Enrique

perexg, 2017-04-26:
It's same source code. Check 'minisatip7 --help'. So -L like you described.

estimadarocha, 2017-04-26:
@perexg first of all thanks for the new build.... remember the old continuity problems with minisatip5? i am still getting those... i will try this night with -B flag to see if i have improvements any other sugestions will be welcome.

arrobar, 2017-04-27:
Works perfect @perexg !! 
Many Thanks

estimadarocha, 2017-04-27:
@perexg 

tried this night with -B 10000 and -B 20000 and still have:

2017-04-27 12:02:37.455 TS: DVB-S 30/12245.34H/Sport TV 1 HD: H264 @ #6433 Continuity counter error (total 1)
2017-04-27 12:02:37.455 TS: DVB-S 30/12245.34H/Sport TV 1 HD: MPEG2AUDIO @ #6434 Continuity counter error (total 1)
2017-04-27 12:02:37.499 tsfix: transport stream H264, DTS discontinuity. DTS = 1128400, last = 854800
2017-04-27 12:02:37.733 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 1082152, last = 794872

don't know if you remember but this issues como from:

https://github.com/catalinii/minisatip/issues/326
https://tvheadend.org/issues/3997

perexg, 2017-04-27:
Try to resolve this with @catalinii in https://github.com/catalinii/minisatip/issues/326 . He wrote that the code is not much tested. I think that he will require some logs to check where is the problem . Assuming, that you don't have this problem when client/server is on the local network.

estimadarocha, 2017-04-27:
@perexg thanks i will ask him to have a look on this and if we have problems i get back to you

m4tt075, 2017-04-27:
@estimadarocha, @perexg: Just for reference, I have had issues with 0.5 producing cc errors as well. Reported them [here](https://github.com/perexg/satip-axe/issues/57) at the time. I have never been able to resolve them. Running axe-fw with minisatip 0.3 version ever since. Anyways, if you should need additional logs or tests, please just let me know...

estimadarocha, 2017-04-27:
@m4tt075 did you try the 0.7?

@perexg i know you split your time on different projects (great job on tvheadend) but i really hope you have a few left for this.

today i make some more tests that i think that will help us getting closer to the problem...

Like i said before and at least for me the most part of the cc errors are when i try to send streams from external network (wan). So until the new build with minisatip 0.7 i was not able to do something like this:

satipaxe ----> minisatip ----> tvheadend (on dedicated server)
(             internal            )

with this setup the cc errors almost disappear, so i believe (and after a lot of information exchange with catalinii) that real problem is somehow related with kernel and hardware itself on Inverto, digibit, grundig.

catalinii think that hardware it self have a limitation of about 4-5MB/s on lan because tcp use of resources and memory, saying that he also believes that sending a stream over wan to a destination 40-50ms away that this limit decreases to 1MB/s on tcp. This don't happen with udp however he also find issues with packet drops without kernel reporting... old kernel????

By the way if i make something like this:

minisatip (with local dvb card) ----> tvheadend (on dedicated server) also 0 cc errors, so even i am a newbie on this i really believe we are looking to the wrong place, because if minisatip standard work the problem can't be own minisatip, right?

catalinii, 2017-04-27:
Hi,

if I remember right, I managed also to get around 7MB (for all 4 tuners) in lan (udp) by running: make DVBAPI=0 DVBCSA=0 DVBAES=0  DVBCA=0 . This disables tables and enable the codepath from https://github.com/catalinii/minisatip/blob/0.7/stream.c#L908 which basically just dumps what is received form the DVR device to the network.

Few other notes:

1) I did not find any way of reporting packet loss if udp is being used. If you want to use tcp, when the processor is being used you might not see any other effects than just CC errors.
2) minisatip 0.5 have a small issue closing the streams which is visible on axe, which takes about 3-4 seconds (fixed in 0.7).
3) minisatip 0.5 have also all the fixes previously included by @perexg (basically a better support of the hardware), in 0.7 just unicable is implemented (ofc normal diseqc or simple setup should be fine too).
4) About the faster code path theoretically it can be implemented even if the dvbapi is compiled inside of minisatip but only if it is disabled at runtime.

Thanks

perexg, 2017-04-28:
> with this setup the cc errors almost disappear, so i believe (and after a lot of information exchange with catalinii) that real problem is somehow related with kernel and hardware itself on Inverto, digibit, grundig.

Do you see any 'CC Errors' increase in the webui (http://axe-ip:8080) ? If the CPU usage (see top in command-line - satip-axe box) is bellow 90%, then everything should be ok.

> minisatip 0.5 have a small issue closing the streams which is visible on axe, which takes about 3-4 seconds (fixed in 0.7).

I don't know about this. The 0.5 was heavily tested with tvheadend which opens many transpoders in the scan phase - no noticable delays betwen tuning

>   About the faster code path theoretically it can be implemented even if the dvbapi is compiled inside of minisatip but only if it is disabled at runtime.

I'm using the TCP data mode in 0.7 and I'm able to stream 2 full muxes (around 12MB) through the local lan now. I need to verify 0.7 code more - I found another issue yesterday and looking to the code, there are other things which should be re-added (axe_skippkt code was removed for example).

catalinii, 2017-04-28:
About axe_skippkt, this will be fixed in 0.7.5.
Basically I am allowing the data to pass thru just after the new PAT is
received. This is a system wide solution as DVB is also affected by it.

Just to confirm, are you able to use 12MB on rtsp over tcp  with the binary
compiled in the latest axe image ?

Thank you

perexg, 2017-04-28:
> Basically I am allowing the data to pass thru just after the new PAT is received. This is a system wide solution as DVB is also affected by it.

> OK. Just to confirm, are you able to use 12MB on rtsp over tcp with the binary compiled in the latest axe image ?

Yes. My fixes / changes are in https://github.com/perexg/minisatip/commits/axe7 .

I'm looking for the hight CPU usage for the full mux streaming and it seems that the culprit is small writev() calls for the TCP socket.

catalinii, 2017-04-28:
> minisatip/commits/axe7 .

Does it work better if you compile without tables (the make I have wrote in
my earlier post ) ?

perexg, 2017-04-28:
The binary is already without this junk :-) 

<pre>
configure --enable-axe --disable-dvbca --disable-dvbcsa --disable-dvbaes --disable-netceiver
</pre>

perexg, 2017-04-28:
Ohh. It seems that --disable-dvbapi is missing. Retesting..

catalinii, 2017-04-28:
Feel free to push the changes once you are done.

Thanks

perexg, 2017-04-28:
OK. New version is available (only minisatip7 fixes/changes) for download. I can stream 3 full muxes almost without errors through RTP/TCP data transfer mode. If you use only RTP/TCP data transfer mode, you can set buffering to these values **-b 4042752 -B 10000** .

estimadarocha, 2017-04-28:
@perexg 

i did put the -b 4042752 -B 10000 on MINISATIP7_OPTS and issues persist:

tell me if you need some logs

2017-04-28 17:19:20.578 TS: DVB-S 30/12322H/RTP 2: MPEG2VIDEO @ #5152 Continuity counter error (total 1)
2017-04-28 17:19:23.900 TS: DVB-S 30/12245.34H/Sport TV 1 HD: MPEG2AUDIO @ #6434 Continuity counter error (total 10)
2017-04-28 17:19:23.900 tbl-base: bat: 12245.34H in DVB-S 30: invalid checksum (len 646, errors 1)
2017-04-28 17:19:23.900 tbl-base: sdt: 12245.34H in DVB-S 30: invalid checksum (len 646, errors 1)
2017-04-28 17:19:23.986 tsfix: transport stream H264, DTS discontinuity. DTS = 10044000, last = 9392400
2017-04-28 17:19:24.208 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 10012872, last = 9349752
2017-04-28 17:19:24.298 TS: DVB-S 30/12322H/RTP 1: TELETEXT @ #5138 Continuity counter error (total 1)
2017-04-28 17:19:24.298 TS: DVB-S 30/12322H/RTP 2: TELETEXT @ #5138 Continuity counter error (total 1)
2017-04-28 17:19:30.443 TS: DVB-S 30/12322H/RTP 1: MPEG2VIDEO @ #5136 Continuity counter error (total 1)
2017-04-28 17:19:35.892 TS: DVB-S 30/12245.34H/Sport TV 1 HD: H264 @ #6433 Continuity counter error (total 11)
2017-04-28 17:19:35.892 TS: DVB-S 30/12245.34H/Sport TV 1 HD: MPEG2AUDIO @ #6434 Continuity counter error (total 11)
2017-04-28 17:19:35.979 tsfix: transport stream H264, DTS discontinuity. DTS = 10945800, last = 10332000
2017-04-28 17:19:36.198 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 10894152, last = 10269912
2017-04-28 17:19:38.429 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5137 Continuity counter error (total 1)
2017-04-28 17:19:38.429 TS: DVB-S 30/12322H/RTP 1: TELETEXT @ #5138 Continuity counter error (total 2)
2017-04-28 17:19:38.430 TS: DVB-S 30/12322H/RTP 2: MPEG2VIDEO @ #5152 Continuity counter error (total 6)
2017-04-28 17:19:38.430 TS: DVB-S 30/12322H/RTP 2: MPEG2AUDIO @ #5153 Continuity counter error (total 1)
2017-04-28 17:19:38.430 TS: DVB-S 30/12322H/RTP 2: TELETEXT @ #5138 Continuity counter error (total 3)

perexg, 2017-04-28:
Provide log with '-llllllllllllllll', I'm using 'ssh -t root@satipbox minisatip ARGS | tee out.txt' . Also, kill the minisatip in box before any tests and create /tmp/nosatip file (touch /tmp/nosatip) to avoid minisatip respawn. Also, do all tests only with one service (channel).

estimadarocha, 2017-04-28:
@perexg 

i did exactly as you say:

ssh -t root@satipbox minisatip7 -llllllllllllllll -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 4042752 -B 10000 | tee out.txt

it creates the out.txt with this:

[28/04 15:50:03.009]: adapter.c:66 allocation_wrapper malloc allocated 3652 bytes at 0x438008
[28/04 15:50:03.010]: map_int: s=>NULL, v=(nil), NULL NULL
[28/04 15:50:03.010]: Setting unicable adapter 0 slot 1 freq 1420
[28/04 15:50:03.010]: adapter.c:66 allocation_wrapper malloc allocated 3652 bytes at 0x438e68
[28/04 15:50:03.010]: map_int: s=>NULL, v=(nil), NULL NULL
[28/04 15:50:03.010]: Setting unicable adapter 1 slot 0 freq 1210
[28/04 15:50:03.010]: adapter.c:66 allocation_wrapper malloc allocated 3652 bytes at 0x439cb0
[28/04 15:50:03.011]: map_int: s=>NULL, v=(nil), NULL NULL
[28/04 15:50:03.011]: Setting unicable adapter 2 slot 2 freq 1680
[28/04 15:50:03.011]: adapter.c:66 allocation_wrapper malloc allocated 3652 bytes at 0x43aaf8
[28/04 15:50:03.012]: map_int: s=>NULL, v=(nil), NULL NULL
[28/04 15:50:03.012]: Setting unicable adapter 3 slot 3 freq 2040

and them i get on ssh Connection to satipbox closed.

perexg, 2017-04-28:
Try to run the command in the ssh session without tee to see why minisatip is not started.

estimadarocha, 2017-04-28:
@perexg maybe a stupid question aren't we missing -f?

estimadarocha, 2017-04-28:
@perexg don't know if it helps or not because as soon as i connect the satipaxe to tvheadend it pulls more than one channel... but the problems start immediately here's the link for the out.txt:

https://www.dropbox.com/s/x9dcjrpd1vn7luc/out.txt?dl=0

perexg, 2017-04-29:
@catalinii : It seems that the TCP socket is in the blocking mode (O_NONBLOCK is not set - verified when I tested with the bellow patch):

<pre>
diff --git a/socketworks.c b/socketworks.c
index b4ba864..5117ac2 100755
--- a/socketworks.c
+++ b/socketworks.c
@@ -1108,6 +1108,7 @@ int my_writev(sockets *s, const struct iovec *iov, int iiov)
                return -1;
 
        LOGL(log_level + 1, "start writev handle %d, iiov %d", s->sock, iiov);
+       LOG("blocking %d", fcntl(s->sock, F_GETFL) & O_NONBLOCK);
        if (opts.log == log_level)
                stime = getTick();
        rv = writev(s->sock, iov, iiov);
</pre>

Thus these big time gaps during write in the log:

<pre>
[28/04 23:47:45.601]: start writev handle 17, iiov 2
[28/04 23:47:45.725]: writev returned 26336 handle 17, iiov 2
</pre>

It means that the whole buffering code is NEVER activated.

perexg, 2017-04-29:
@estimadarocha : Could you test http://s000.tinyupload.com/index.php?file_id=28439880167745932685 ? It's minisatip7 with this dirty patch:

<pre>
diff --git a/socketworks.c b/socketworks.c
index b4ba864..d6d26cf 100755
--- a/socketworks.c
+++ b/socketworks.c
@@ -401,6 +401,10 @@ int sockets_accept(int socket, void *buf, int len, sockets *ss)
                ss->action(s[ni]);
        }
        set_linux_socket_timeout(new_sock);
+       {
+       int flags = fcntl(new_sock, F_GETFL, 0);
+       fcntl(new_sock, F_SETFL, flags | O_NONBLOCK);
+       }
 
        return 1;
 }
</pre>

estimadarocha, 2017-04-29:
@perexg sure... i am here to help!!!

and if you need access to something to check just say me... it's lunch time in one hour i will have feedback for you!

estimadarocha, 2017-04-29:
@perexg how do i know if minisatip running its the one i put on /tmp/ or the sbin?

i downloaded binary uploaded to temp... get into the dir /temp and them execute the normal command, right?

if i am right the problems still exist...

https://www.dropbox.com/s/fe00j0d97t5r5at/minisatip2.log?dl=0

catalinii, 2017-04-29:
@perexg 

in tcp_connect the timeout is set to 100ms, basically if the write lasts more than that, it will start buffering.

If I disable that, in a lot of sceniarios it buffers right away especially in high latency writes, and is not really working as desired.

In the logs from ricardo I did not see any overflow, which means all the writes are successful, right ?

perexg, 2017-04-29:
@estimadarocha : You must run '/tmp/minisatip7' (full path), otherwise the binary from /sbin is used.

@catalinii : It's not the 100ms threshold. The problem is that the socket is in the blocking mode, so if writev() is called and kernel is not able to buffer the new data, it's stalled (kernel waits for the room in it's buffer). But we need the socket in the non-blocking mode to return from the partial write without any wait operation.

perexg, 2017-04-29:
@estimadarocha : Analyzing the second log and it's correct output. I see the partial writes now:

<pre>
[29/04 12:46:44.320]: send_rtp: sent 26320 bytes for stream 0, handle 8, sock_id 6, seq 3048 => 62.210.116.189:41526
[29/04 12:46:44.320]: start writev handle 8, iiov 2
[29/04 12:46:44.360]: writev returned 15056 handle 8, iiov 2 (took 0 ms)
</pre>

perexg, 2017-04-29:
OK, I think I found another issue (sorry - debugging the @estimadarocha  's code). Here's the patch:

<pre>
diff --git a/socketworks.c b/socketworks.c
index b4ba864..57e9330 100755
--- a/socketworks.c
+++ b/socketworks.c
@@ -401,6 +401,10 @@ int sockets_accept(int socket, void *buf, int len, sockets *ss)
                ss->action(s[ni]);
        }
        set_linux_socket_timeout(new_sock);
+       {
+       int flags = fcntl(new_sock, F_GETFL, 0);
+       fcntl(new_sock, F_SETFL, flags | O_NONBLOCK);
+       }
 
        return 1;
 }
@@ -1107,6 +1111,10 @@ int my_writev(sockets *s, const struct iovec *iov, int iiov)
        if (s->timeout_ms == 1)
                return -1;
 
+       len = 0;
+       for (i=0; i<iiov; i++)
+               len += iov[i].iov_len;
+
        LOGL(log_level + 1, "start writev handle %d, iiov %d", s->sock, iiov);
        if (opts.log == log_level)
                stime = getTick();
@@ -1114,11 +1122,6 @@ int my_writev(sockets *s, const struct iovec *iov, int iiov)
        if (opts.log == log_level)
                stime = getTick() - stime;
 
-       len = 0;
-       for (i=0; i<iiov; i++)
-               len += iov[i].iov_len;
-
-
        if(rv > 0)
        {
                bw += rv;
</pre>

@estimadarocha  : Try this binary (it should identify with -axe202-t2 version suffix): http://s000.tinyupload.com/index.php?file_id=08316236044136555534

estimadarocha, 2017-04-29:
@perexg 

/ # /tmp/minisatip7 -llllllllllllllll -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 4042752 -B 10000
-sh: /tmp/minisatip7: Permission denied

if i make first cd /tmp/
and them just  minisatip7 -llllllllllllllll -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 -b 4042752 -B 10000

the log says [29/04 15:59:35.160]: minisatip version 0.7.4-axe202, compiled with s2api version: 0501

i don't see the -axe202-t2

perexg, 2017-04-29:
@estimadarocha : Correct permission with 'chmod 755 /tmp/minisatip7'. I have this output:

<pre>
~ # /tmp/minisatip --help | head -1
minisatip version 0.7.4-axe202-t2, compiled with s2api version: 0501
</pre>

estimadarocha, 2017-04-29:
Ok give me some mins

estimadarocha, 2017-04-29:
@perexg now minisatip dies few seconds after a stream start
check log

https://www.dropbox.com/s/m0v78iyw2rjmai9/minisatip3.log?dl=0

perexg, 2017-04-29:
Things are going to be complicated: https://github.com/perexg/minisatip/commit/83a4ff7535eee6c2f230031836de12d38de2ea86 .

@estimadarocha : Try -t3 : http://s000.tinyupload.com/index.php?file_id=76651583633185769710

estimadarocha, 2017-04-29:
@perexg new log

https://www.dropbox.com/s/lyur469tfshs85f/minisatip4.log?dl=0

2017-04-29 18:39:07.778 mpegts: 12245.34H in DVB-S - tuning on SAT>IP DVB-S Tuner #4 (satip03.ddns.net)
2017-04-29 18:39:07.778 capmt: CA client 1: Starting CAPMT server for service "SPORT TV 1 HD" on adapter 1
2017-04-29 18:39:07.779 subscription: 0062: "HTTP" subscribing to service "DVB-S/12245.34H/SPORT TV 1 HD", weight: 100, adapter: "SAT>IP DVB-S Tuner #4 (satip03.ddns.net)", network: "DVB-S", mux: "12245.34H", profile="pass", hostname="192.168.1.109", client="VLC/2.2.4 LibVLC/2.2.4"
2017-04-29 18:39:38.388 satip: SAT>IP DVB-S Tuner #4 (satip03.ddns.net) - RTSP error -90 (Message too long) [5-0]
2017-04-29 18:39:39.780 subscription: 0062: service instance is bad, reason: Tuning failed
2017-04-29 18:39:41.780 subscription: 0062: No input source available for subscription "HTTP" to service "DVB-S/12245.34H/SPORT TV 1 HD" in mux "12245.34H in DVB-S"
2017-04-29 18:39:41.780 webui: Couldn't start streaming /stream/service/ec1460c18d5b3cee527ce06901582b62?ticket=2B95DE0BAA9678E28ABE2600A4E0ED482BFA77BF, Tuning failed
2017-04-29 18:39:41.780 subscription: 0062: "HTTP" unsubscribing, hostname="192.168.1.109", client="VLC/2.2.4 LibVLC/2.2.4"

perexg, 2017-04-29:
@estimadarocha : Try -t4 : http://s000.tinyupload.com/index.php?file_id=04815066942490634300 .

estimadarocha, 2017-04-29:
@perexg 

https://www.dropbox.com/s/wbtnki3pv1eut46/minisatip5.log?dl=0

perexg, 2017-04-29:
@estimadarocha : The last log is perfect. If you see CC errors, then it's not caused by the network.

estimadarocha, 2017-04-29:
@perexg the channels dont open... could this be related...?

perexg, 2017-04-29:
@estimadarocha : Could you test some FTA channel?

estimadarocha, 2017-04-29:
What do you prefer @perexg? Fta or full mux cause I don't have any fta with with bitrate and hd

perexg, 2017-04-29:
I'm a bit confused. If the channel works with the minisatip7 from /sbin (no test version) with CC errors, then it should work with the test version, too.

estimadarocha, 2017-04-29:
I will confirm again give me a few minutes... but this last test was like the streams don't make em requests

perexg, 2017-04-29:
-t5 : http://s000.tinyupload.com/index.php?file_id=97451163452113074058

This version should fix the RTSP error -90 (reported broken stream from tvheadend).

@catalinii : A ton of fixes. The http_response() code is a quick hack - the replies may be probably inserted to the write queue before the next RTP data. My latest code is at: https://github.com/perexg/minisatip/commits/axe7-test

estimadarocha, 2017-04-29:
@perexg webinterface its broken http://prntscr.com/f2bxhs

the unscramble was my fault...

log from last -t5

https://www.dropbox.com/s/qdpigufelr1x3ao/minisatip6.log?dl=0

do you think its stable enought to test pulling more than 1 stream?

catalinii, 2017-04-30:
Hi @perexg ,

I saw the interface issue before and I have replaced it in my debug version (which is not comited yet)

About the blocking, it happens for 100ms, right ?

Thanks

perexg, 2017-04-30:
Added new build (see first comment). The minisatip7 with the buffering changes should be stable again.

@estimadarocha : Try it.

@catalinii : I pushed all my changes to https://github.com/catalinii/minisatip/pull/350 . Thanks for your comments in the test branch. And no, it's not about the 100ms interval - when the socket is in the blocking mode (default), the writev() call can stall (wait for network or anything) for some delay. It might be few ms (as it was visible in the @estimadarocha  's logs when the network was busy for a reason). The only way is to set the socket to the non-blocking mode and buffer data inside the app.

perexg, 2017-04-30:
The new version is able to transfer three full muxes without any errors (RTP/AVP/TCP) and four muxes with 'small' error count and almost full bandwidth . It's just because of the CPU power. But I think that this usage is rare, so we're fine.

estimadarocha, 2017-04-30:
@perexg testing now will give feedback still today

estimadarocha, 2017-04-30:
@perexg is it normal web interface don't work?

perexg, 2017-04-30:
It is fixed in -axe203 (satip-axe-201704301818-14):

[image]

<pre>
$ ssh root@satip1

Welcome to SATIP-AXE server 201704301818-14

~ # minisatip7 --help | head -1
minisatip version 0.7.4-axe203, compiled with s2api version: 0501

~ # ps ax | grep minisatip7
  756 root      19:45 minisatip7 -f -g -d 38 -M 0 -W 0 -q *:5-15-5-5-5-0 -d *:1-0 -A 0:0:0,0:1:0,1:0:1,1:1:1,2:2:0,3:2:1,4:3:0,5:3:1,6:3:2,7:3:3 -b 4042752 -B 10000
  786 root       0:00 grep minisatip7
</pre>

estimadarocha, 2017-04-30:
Now I am the one who's confused... I thought I've downloaded that file already and it's the version I am running to make the tests. But now that you said that I checked and it's not 203... can you confirm the link it's really the 203?

[48 more replies]

closed 2017-06-07 by perexg