perexg/satip-axe #67: Unable to negotiate with 192.168.1.22 port 22: no matching host key type found. Their offer: ssh-dss

dreamcat4, 2016-06-19:
Hi,
Running a slightly older version of satip firmware here. Just looking at the date its probably r10 `release satip-axe-201508061051-10`.

Can't login over ssh. It just says:

```sh
$ ssh satip
Unable to negotiate with 192.168.1.22 port 22: no matching host key type found. Their offer: ssh-dss
```
Ever since May 23 2016. Yet last backup seems to show the folder `/etc/sysconfig/sysconfig/` had `authorized_keys` file containing my ssh rsa pubkey.

I also see in same backup folder a file named `dropbear_dss_host_key`... it seems something created by satip-axe firmware. But its got a passphrase and I don't know it. So I cannot regenerate a pubkey for logging in over ssh. And it won't accept password method of authentication, or rsa, anything except `dss`.

I still can login over FTP with root password. But the `/media` folder which ftpd points to as its root folder is empty, without any files in it. My telnet and nfs services are disabled.

@perexg Can you please let me know what to do? Many thanks.

asterix555, 2016-06-19:
Hi,
the hostkey is a DSA key, recent openssh versions deprecated DSA keys. You can add an exemption (replace <IP ADDRESS> by your satip-axe IP address:

# echo -e "Host <IP ADRESS> \n HostkeyAlgorithms +ssh-dss"  >> ~/.ssh/config

http://stackoverflow.com/questions/34208495/unable-to-negotiate-with-xx-xxx-xx-xx-no-matching-host-key-type-found-their-of

regards

dreamcat4, 2016-06-19:
Hey... It worked! Thanks no end @perexg. I really appreciate this.

Maybe I should upgrade my firmware now. Is it still OK to with the cmdline / online updater program `upgrade-fw`? I am on version 10 at the minute.

closed 2016-06-19 by dreamcat4

perexg, 2016-06-20:
Remove /etc/sysconfig/dropbear_dss_host_key for the latest builds. The latest builds generates only RSA keys.

dagelf, 2017-02-20:
You can also try:

    ssh -o KexAlgorithms=diffie-hellman-group14-sha1 -oHostKeyAlgorithms=+ssh-dss my.host.com

Add -v if you want to see what's happening, and -o HostKeyAlgorithms=ssh-dss if it still doesn't work:

    ssh -v -o HostKeyAlgorithms=ssh-dss -o KexAlgorithms=diffie-hellman-group14-sha1 my.host.com

You can also, of course, edit /etc/ssh/ssh_config or ~/.ssh/ssh_config, and add:

    Host my.host.com *.myinsecure.net 192.168.1.* 192.168.2.*
        HostKeyAlgorithms ssh-dss
        KexAlgorithms diffie-hellman-group1-sha1

kulongwangzhi85, 2017-08-30:
/etc/ssh/ssh_host_key file exist?